EverythingSmartHome/everything-presence-lite #1: Assembly Instructions - Photo Error

sembeeuk, 2023-10-23:
Minor error - but might cause some people some problems if not concentrating. 

In the build instructions, the first photo after "Board and Sensors" ( https://everythingsmarthome.github.io/everything-presence-lite/images/assembly-insert-epl-board-1.jpg ) shows the case in the wrong orientation compared to the board. 

On the case I received, the two holes in the back are at the bottom with the hole for the USB port. Whereas in your photo, you have the case shown which has those at the top. If you put the board in following those exact instructions, then the USB port is on the wrong side. 

A very minor thing for a first issue, but thought I would point it out.

EverythingSmartHome, 2023-10-23:
Thanks, the image was more intended to be a "here are the parts you need" thing rather than an orientation thing but I can see how users would think that. I'll try update it at some point next time I am getting images.

maxelkins, 2023-10-25:
[image]

In case you want a quick edit :)

maxelkins, 2023-10-25:
Couple PRs for this:

Fix: https://github.com/EverythingSmartHome/everything-presence-lite/pull/13
Additional: https://github.com/EverythingSmartHome/everything-presence-lite/pull/14

Split them up incase you're not bothered about reducing the image sizes, but it shouldn't affect quality in any noticable way :)

closed 2023-10-25 by EverythingSmartHome