EverythingSmartHome/everything-presence-lite #4: Added to HA fine but doesn't appear in the ESPHome Add On

swampylee, 2023-10-24:
Maybe I'm doing something wrong or not doing something I should but I'm pretty sure my EP1 appeared in ESPHome addon after adding it to HA but the Lite doesn't.

davidrule1969, 2023-10-24:
I added them manually using this .yaml:

```
substitutions:
  name: everything-presence-lite-xxx
  friendly_name: Everything Presence Lite
packages:
  Everything_Smart_Technology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: "your-encyption-key="

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

sumitkm, 2023-10-25:
@davidrule1969 this should be part of the documentation. It is really handy for _weird_ folks like me who don't have Chrome on their laptops can can't use Web USB/Web Serial.

EverythingSmartHome, 2023-10-25:
> Maybe I'm doing something wrong or not doing something I should but I'm pretty sure my EP1 appeared in ESPHome addon after adding it to HA but the Lite doesn't.

I'm not sure why that's happening as the dashboard_import line is there and correct as far as I can see. I did try changing some letters to lower case in the most recent firmware, you could try updating to that using the user guide and see if it works - I don't have high hopes though. I'll look more into it tomorrow

JRascagneres, 2023-10-25:
Reported on Discord but just +1 that I'm also having this issue

TripelJ, 2023-10-30:
I don't have this problem. It was discovered under integrations and afterwards went to esphome addon where I could hit adopt as well. Looking forward to tinker with them.

uvjim, 2023-10-31:
Mine was missing from the add-on as well. I'd upgraded to 1.0.1 firmware by following the user guide and using the browser to install. That didn't seem to help. However, there was an update to the Home Assistant ESPHome add-on. I applied that and it was detected. I could then adopt and get it into the console.

jeleniain, 2023-11-02:
Hey guys, just a head-sup in case someone made the same mistake, running ESPHome in docker. I had the "USE_PING" environment variable enabled due to some issues with other boards being offline often. After reverting that I could adopt the sensors as expected.

swampylee, 2023-11-02:
> Mine was missing from the add-on as well. I'd upgraded to 1.0.1 firmware by following the user guide and using the browser to install. That didn't seem to help. However, there was an update to the [...]

Yeah mine popped in too after I updated the Lite firmware and the ESPHome Add-On so this can probably be closed now Lewis.

EverythingSmartHome, 2023-11-02:
Thanks, sounds like it's starting to work now!

closed 2023-11-02 by EverythingSmartHome