EverythingSmartHome/everything-presence-lite #6: Speed up detection

uvjim, 2023-10-24:
Is there any way to speed up detection of movement?

Entering the room takes a few seconds to turn the light on. I have this combined with lux in a Home Assistant automation to turn on only if "dark". It just takes a fraction too long to come on. There is already a PIR sensor in the room that I can add into the automation but was hoping to be able to do away with that.

Leaving the room is detected quickly and I have that set with a 2 minute delay before turning lights off. That works well.

P.S. the distance feature is great as so far it has stopped things being detected outside (unless that's a fluke).

EverythingSmartHome, 2023-10-24:
Are you sure it's the sensor changing that is taking long? From my experience it shouldn't be more than 1 second so wonder if it's your automation or something else that is adding delay?

uvjim, 2023-10-24:
Well, if I ignore the automation and just look at the entity state it takes a few seconds to detect.

I am walking into the room from behind the sensor.

TheManoeuvre, 2023-10-24:
If you approach directly is this the same? If not then the devices may have different degrees of visibility impacting how quickly they see you when approaching from behind.

uvjim, 2023-10-25:
I may have managed to sort this out with a bit of tweaking on the positioning of the sensor.  I'll keep an eye on it mind you.

@TheManoeuvre - I did try approaching directly but that didn't seem to help at the time (may have been something environmental).

I'll be in and out of the office today so should be able to determine if the response time is better and also if how it is positioned now keeps the occupancy sensor at `detected`.

I have spotted that the target changes when I move but I guess that may be something to do with speed of movement and detection in a different location.  This isn't an issue for me as I really don't care about multiple targets in this room.

I should hopefully be able to close this out later today.

uvjim, 2023-10-25:
The new orientation seems to have done the trick. Now I just need to run the power cable properly.

closed 2023-10-25 by uvjim