EverythingSmartHome/everything-presence-lite #7: User Guide Refers to EP1

TheManoeuvre, 2023-10-24:
The [Tuning Page](https://everythingsmarthome.github.io/everything-presence-lite/tuning.html) of the EP Lite makes several references to the EP1 & EP1 functions such as PIR.
It also suggest adjusting sensitivity if you are seeing periods of "not detected" while people are present (which I am) but there is no clear direction how to adjust this value.

EverythingSmartHome, 2023-10-24:
Yes that page hasn't been touched since copying over from EP1, it's a work in progress along side some of the other pages.

The Lite doesn't have configurable sensitivity unfortunately

TheManoeuvre, 2023-10-24:
Thanks, I just found the merge in #3 
Relating to the sensitivity issue, would the only option to extend the latency or ensure automations had a “for x period” condition?

EverythingSmartHome, 2023-10-24:
Hmm, I don't follow how sensitivity relates to latency, perhaps you could explain the issue you are running into and I could suggest something?

TheManoeuvre, 2023-10-24:
Sure - I have the ESL currently sat on the lounge fireplace throughout the evening the sensor would show “clear” while one or two people were in the room on different seats. This is typically for a 1-3 second period and then it detects again.
I had thought that extending the delay off may cover these gaps (though admittedly impact legitimate responsiveness).

EverythingSmartHome, 2023-10-24:
Ah yeah, extending the delay off is a good solution to that and probably recommended. For most people, they don't really need less than 30-60 seconds.

What do you mean by impact legitimate responsiveness?

TheManoeuvre, 2023-10-25:
I mean it clearing when it should - one planned use case was pausing the TV when people have left the room and the remaining runtime is over x minutes. Building in off latency means this won’t pause as quick as I’d like.

I think we’re good to close this but I do think a good enhancement would be a zone clearing latency control to help cover the same potential issue in zones.

closed 2023-10-31 by EverythingSmartHome