EverythingSmartHome/everything-presence-lite #20: No response from LD2450 sensor

stewartwlewis, 2023-10-31:
I have flashed the device and added it to ESPHome.  This all went OK but I do not get any presence values.  Illuminance is fine. Occupancy shows "Detected" and Target 1 to 3 are all shown as "On" but all other entries show Unknown.
Looking at the logs, there are no presence responses, only one from the Illuminance sensor every 2 seconds.  During the initialisation there doesn't seem to be anything for the LD2450 though I don't know what I am looking for.
I have made sure the sensor is seated properly into the socket and all the pins are correctly aligned and have distance entries in all of the controls section.
I have attached part of the device log. and screen shots of settings and output. 
Can you help please. 

[ESPHome EPL log 31 Oct 23 .txt](https://github.com/EverythingSmartHome/everything-presence-lite/files/13213745/ESPHome.EPL.log.31.Oct.23.txt)
[image]
[image]

EverythingSmartHome, 2023-10-31:
Hello! Interesting issue, it looks like the device isn't outputting anything via UART. Are you able to confirm that you have the sensor seated correctly in the socket and not offset (it's an easy thing to do, I've done it myself!) by one pin to the left or right?

Are you able to download the HiLink app on iPhone or Android and see if you can discover the device and update the firmware? You should be able to find it by searching for "HLKRadarTool" by hlktech.

EverythingSmartHome, 2023-10-31:
The Lite doesn't expect any software version in particular, might just have been a bug with the version that was loaded on the LD2450 from factory I guess - glad its working.

X is the distance left or right of the sensor, Y is the distance forward or back from the sensor. The start and end values allow you to define zones where you want to detect people in

stewartwlewis, 2023-10-31:
Great thanks, that makes sense.  I think the "dual people" come from reflections.  Should be able to minimise that by setting the zones.

EverythingSmartHome, 2023-10-31:
It's possible when doing larger movements like walking that it can detect more than 1 target, because for example when your hand/arm are moving in a different motion to your torso for example, it can be difficult for the sensor to detect if this is one person or two - hopefully that makes sense. This is typically not an issue when standing or sitting

stewartwlewis, 2023-10-31:
Sorry but there still seems to be something wrong.  The device has again stopped sending any information via ESPHome and there is no presence output in the logs.  Connecting  via the HiLink app shows the LD2450 is working and tracking presence.  Powering off/Rebooting didn't help.  I dave done nothing in between, was just doing other things and came back to this to find it not responding again.

stewartwlewis, 2023-10-31:
As the ESP32 device is working as is the LD2450 through BLE it seems likely there is an issue with the 8 pin connector so I put a meter across each pin and the matching pad on the LD2450 board.  The DM pin is floating and not properly soldered to the board.
What would be best? Do you want to replace it or are you happy with me having a go at soldering it on?  My soldering is OK but it is a surface mount component and probably done with lead free solder so can't promise to succeed.

EverythingSmartHome, 2023-11-01:
Are you able to attach some images of that? By DM, which pin are you referring to? Not all pins are attached on the socket as not all are used.

stewartwlewis, 2023-11-03:
Sorry I am away at the moment but will upload a picture when I get back. It could be that more than one of the pins on the connector is not making proper contact,  If I squeeze the  the connector with my fingers the device reports the settings through your board to home assistant.  Take the pressure away and it doesn't though it still responds to the app through Blue Tooth.  The issue is the surface mount 8 pin connector on the LD2450..

stewartwlewis, 2023-11-13:
[image]

This is the connector on the LD2450.  The 8 pins also have 8 corresponding pads on the circuit board.  Checking the continuity between the pins and the pads, the DM pin and the RX pin are open circuit to their respective pads.  
It just looks like the soldering of the connector was not done effectively.  I assume it's the open circuit RX pin thats causing the real problem.  As I said above I am happy to get the soldering iron out but if you want to examine it I will return if you want to replace it.

stewartwlewis, 2023-11-13:
I re-soldered all the pins on the connector and the sensor seems to be communicating properly with the ESP32.

stewartwlewis, 2023-11-14:
Re-soldering the sensor connector seems to be working fine.

closed 2023-11-14 by stewartwlewis