EverythingSmartHome/everything-presence-lite #22: High data usage

jamesgreeley, 2023-10-31:
Hi, is this normal 24hour data usage for the sensor? 

[image]

openPhiL, 2023-10-31:
I can confirm that it is "high". 
[image]

Not sure if that is a problem, but compared to "the others", that is really high. 
I wonder if that is related to the issue that I have (but not yet raised) with Wifi-dropping, just tooo much?

EverythingSmartHome, 2023-11-01:
I hadn't actually noted the data usage so thanks for bringing it up. I am seeing around 250mb's like @openPhiL screenshot above, but yours is almost double at 500. Do you happen to have it connected to 2 Home Assistant instances at the same time?

[image]

The only thing I can think this would be related to is Bluetooth Proxy as it would constantly be sending scan results back to HA, something that's not yet enabled in the EP1 firmware.

CDAutomate, 2023-11-17:
For what its worth mine are about ~250mb usage as well:

[image]

openPhiL, 2023-11-17:
I tinkered a bit with the config and removed the information that I don't need: 
[image]

that boils down the data submitted to around 7MB / day: 
[image]

So if that is an issue for you, you may use this to fix, just as an inspiration. 

[image]

Which leads me to think, if you have lots of sensors, you will have lots of data. hence, this is not "an issue" to fix and this can be closed, what do you think?

EverythingSmartHome, 2024-02-20:
Thanks for sharing the findings!

closed 2024-02-20 by EverythingSmartHome