EverythingSmartHome/everything-presence-lite #23: Occupancy Sensor - Slow to Respond or Incorrect Reading

skrilla0, 2023-11-01:
I have my EP Lite setup in my office next to the EP1 for the initial setup, and having issues with the Occupancy Sensor providing the correct reading: sometime's it responds, other times there is a delay, or no response at all. 

I've tried to re-seat the sensor, re-flash the board, and restart the device multiple times, but the Occupancy Sensor has not been accurate. What's odd is that the Target 1 Distance Resolution, and Speed sensors are responsive and functional. 

[image]

I've also downloaded the HIlink app, and the Sensor was detected, and proceeded to update to the latest firmware.

TripelJ, 2023-11-01:
I have a side question if I may. :) 
Do you actually recommend firmware upgrading the sensor with latest firmware using the app? Or is it only if we experience an issue. 

Also could you please link to the app on iOS if exists. Thanks.

EverythingSmartHome, 2023-11-01:
> I have my EP Lite setup in my office next to the EP1 for the initial setup, and having issues with the Occupancy Sensor providing the correct reading: sometime's it responds, other times there is a [...]

That sounds more like a bug in the logic on the ESP32 side than the mmWave sensor firmware - I'll take a look see what I can see

skrilla0, 2023-11-02:
> That sounds more like a bug in the logic on the ESP32 side than the mmWave sensor firmware - I'll take a look see what I can see

Thank you. Not sure if related but also noticed in the logbook that the Occupancy Sensor goes Unavailable:

[image]

EverythingSmartHome, 2023-11-02:
Do any other entities go unavailable?

skrilla0, 2023-11-02:
Target 1, 2, & 3 Active Entities also go unavailable. Also Zone 1 Occupancy.

EverythingSmartHome, 2023-11-02:
It sounds more like your having connectivity issues in that case, which is causing the status of the occupancy to be reset.

How familiar are you with ESPHome? It would be great if you could capture some logs of this happening

skrilla0, 2023-11-02:
Thanks, I am seeing this recurring error in the logs:

```[14:19:56][D][sensor:094]: 'Zone 1 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[14:19:56][D][sensor:094]: 'Zone 2 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[14:19:56][D][sensor:094]: 'Zone 3 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[14:19:56][D][sensor:094]: 'Zone 4 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[14:19:56][W][component:204]: Component uart took a long time for an operation (0.17 s).
[14:19:56][W][component:205]: Components should block for at most 20-30ms.```

EverythingSmartHome, 2023-11-02:
That is "normal" as of a couple releases ago. It would be more like disconnect lines or resets

skrilla0, 2023-11-02:
> That is "normal" as of a couple releases ago. It would be more like disconnect lines or resets

Ah ok got it. I did notice this:

```WARNING everything-presence-lite-24ea20.local: Connection error occurred: EOF received
INFO Processing unexpected disconnect from ESPHome API for everything-presence-lite-24ea20.local
WARNING Disconnected from API
INFO Successfully connected to everything-presence-lite-24ea20.local
[14:54:25][D][sensor:094]: 'Target 1 X': Sending state 31.00000 mm with 0 decimals of accuracy

skrilla0, 2023-11-03:
Not exactly. I'm far from an ESPHome expert, & could not locate how to disable in the ESPHome documentation. Is this accomplishable via YAML?

jjgadget, 2023-11-04:
I was seeing the same unavailable cycling on my EPLs. I have disabled BT Proxy and esp32_improv but as still getting the same cycling.

Littlecheez17, 2023-11-05:
> I added power_save_mode: none to my wifi section and looks to have stopped cycling.
I have tried to add this to my yaml config but when validating the code I get the error 'power_save_mode NONE is incompatible with esp32_ble.', can anyone help me?

skrilla0, 2023-11-08:
Disabled BT proxy & esp32_improv, and that looks like it solved the issue.

closed 2023-11-08 by skrilla0