EverythingSmartHome/everything-presence-lite #26: Unable to detect motion with LD2450

kostda, 2023-11-02:
Hello two days ago I recieved your sensor, but I'm unable to make it work, since that day.
Light sensor is working, but occupancy is not.

What i tried:
1) make sure to have sensor properly connected
2) update ESPHome in HA
3) update ESPHome in Everything presense lite
4) install firmware multiple times at https://everythingsmarthome.github.io/everything-presence-lite/updating.html
5) tried multiple cables and power bricks

This image shows how its connected:
[image]

Here is preview of settings inside HA:
[image]

Here is log from ESPHome:
[logs_everything-presence-lite-24e71c_logs.txt](https://github.com/EverythingSmartHome/everything-presence-lite/files/13245009/logs_everything-presence-lite-24e71c_logs.txt)

Thanks for any help

EverythingSmartHome, 2023-11-02:
Hello, it looks like from your logs that the sensor is working correctly but I notice in your screenshot that you have the distance set to a maximum of 10cm which would explain why you aren't seeing anything?

kostda, 2023-11-02:
Thanks for quick response!

Sorry for that. As I said, I was experimenting with settings to try if anything change. Even with maximum value its not working.

When device is restarted it shows as motion is detected but after few seconds it shows no motion and Im unable to detect it.
[image]

EverythingSmartHome, 2023-11-02:
Can you post some updated logs with the distance set correctly?

kostda, 2023-11-03:
Here it is:
[logs_everything-presence-lite-24e71c_logs (2).txt](https://github.com/EverythingSmartHome/everything-presence-lite/files/13247222/logs_everything-presence-lite-24e71c_logs.2.txt)

EverythingSmartHome, 2023-11-04:
Strange as it seems like it is updating. What you can try next is download the "HLKRadarTool" app on iOS or Android and try to update the firmware of the mmWave sensor. Once you open the app it will discover it over bluetooth and give you the option to update firmware

kostda, 2023-11-04:
It looks like I have latest formal version.

Should I try to install some of Trial versions? It's something like beta releases, right?  I dont know, if it can be downgraded, so I want to make sure to not to mess anything.

[image]

EverythingSmartHome, 2023-11-04:
Yeah you can try and download a beta, downgrading is possible

kostda, 2023-11-04:
I upgraded to 2.04. I had some few miner problem with upgrading (had to start upgrade multiple times). But at the end it ended well.
I even get new detail info page in app. But still doesnt show any motion :(
[image]

kostda, 2023-11-04:
Now I also tried v2.02 and no change

kostda, 2023-11-14:
Hi, do you have any other options? Sensor is still not working.

Thanks

EverythingSmartHome, 2023-11-15:
Sorry for the delay, somehow I missed it!

If you can please get in touch with me using the contact page on the shop, then we can get you sorted out with a replacement: https://shop.everythingsmart.io/pages/contact

kostda, 2023-11-19:
Hi, I ordered new sensor LD2450 and after replacing it started working. 

Thanks for trying to help me :)

closed 2023-11-19 by kostda