EverythingSmartHome/everything-presence-lite #27: Issue with order of x and y location updates

crhammock, 2023-11-02:
I have done quite a bit of testing and I think I have got to the bottom of an issue although this is all based on home assistant data rather than the logs from esphome.  I think it is simpler if I explain the scenario.

My sensor is located about 1m from a wall on its right (the wall has windows along it).  Because of this and bushes outside the window I have reduced all the "zone begin x" to be -1000mm.  I keep seeing targets tracking at beyond -1000mm (-1500mm for example), this is a bush moving in the wind.  Obviously at this point there is no occupancy in the zone.

The problem comes when it stops detecting motion, I think what is happening is the x axis is updating to 0 before the y axis updates.  This means for a split second the target location is x = 0 and y = 500mm (for example).  Obviously this puts a target in the zone which triggers the occupancy.  I assume that occupancy is based on a target not equal to x = 0 and y = 0.  However in HA the update times for both x and y are the same to the nearest second (but that maybe rounded based on ESP communicating them).  What I know for sure is every time the target counts drops to 0 the occupancy briefly changes to occupied.

If my assumption is correct, I am not sure how to fix the issue.  If you were to change the update order so y is updated before x then I guess this would fix my issue but cause an issue with a target moving between zones at different distances from the sensor.

I will try and take some logs from the esp and see if it proves my theory

crhammock, 2023-11-03:
I have attached the below logs from the ESP Log.  At 08:17:09 the target x and y is outside of zone 1, so the zones target count is 0.  However at 08:17:10 there is no target information yet the target count in the zone is 1 and the occupancy state is changed to on.  Maybe it isn't doing what I thought but it certainly is the case it is changing the target count and occupancy state when there is no reported target data. 

> [08:17:09][D][sensor:094]: 'Target 1 X': Sending state -1425.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 1 Y': Sending state 2264.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 1 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 1 Resolution': Sending state 360.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 1 Distance': Sending state 2675.13013 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 1 Angle': Sending state -32.18701 ° with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 2 X': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 2 Y': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 2 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 2 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 2 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 2 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 3 X': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 3 Y': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 3 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 3 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 3 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Target 3 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Zone 1 Target Count': Sending state 0.00000  with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Zone 2 Target Count': Sending state 0.00000  with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Zone 3 Target Count': Sending state 0.00000  with 0 decimals of accuracy
> [08:17:09][D][sensor:094]: 'Zone 4 Target Count': Sending state 0.00000  with 0 decimals of accuracy
> [08:17:09][W][component:214]: Component uart took a long time for an operation (0.17 s).
> [08:17:09][W][component:215]: Components should block for at most 20-30ms.
> [08:17:10][D][sensor:094]: 'Target 1 X': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 1 Y': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 1 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 1 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 1 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 1 Angle': Sending state -90.00000 ° with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 2 X': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 2 Y': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 2 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 2 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 2 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 2 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 3 X': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 3 Y': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 3 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 3 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 3 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Target 3 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
> [08:17:10][D][binary_sensor:036]: 'Zone 1 Occupancy': Sending state ON
> [08:17:10][D][sensor:094]: 'Zone 1 Target Count': Sending state 1.00000  with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Zone 2 Target Count': Sending state 0.00000  with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Zone 3 Target Count': Sending state 0.00000  with 0 decimals of accuracy
> [08:17:10][D][sensor:094]: 'Zone 4 Target Count': Sending state 0.00000  with 0 decimals of accuracy

Let me know if any further information or testing is useful.

crhammock, 2023-11-25:
anyone?

kloppy, 2023-11-29:
Looks like a bug to me. Zone count and occupancy are checked via the state of the "target active" template sensors. I don't know much about the inner workings of ESPhome but I'd assume those do not get updated while the UART lambda is running.

That means the check whether a target is in a zone is only performed if the target was active **_one second ago_** (last time the UART lambda ran).

An easy fix would be to check against any of the local variables instead of the sensor state.

E.g. for target 1 instead of: 

```c
if (id(target1_active).state == true )
```
use
```c
if (p1_distance > 0)
```

vilhalmer, 2024-01-02:
I believe I'm seeing this as well, I recently moved my EPL to a different wall and reconfigured it such that I had a zone which started at x=0 for simplicity. The lights controlled by this zone's occupancy were regularly blinking on for just a moment. After watching the sensor states for a while I've moved the beginning of the zone above 0mm and this seems to have bandaided it.

EverythingSmartHome, 2024-10-08:
Is anyone able to confirm if this is an issue still on the latest version?

crhammock, 2024-10-10:
Is there something specific changed to the code to fix this?  I am currently running Kloppy's custom code but can switch back and try the latest offical code.  If I recall its a bit of effort as the measurements are all different so I need to setup zones etc again but can give it a go at the weekend.

crhammock, 2024-10-24:
I have upgraded to the latest offical firmware and the issue apears to be resolved.

Thanks to Lewis for implementing a fix.  Special thanks to @kloppy for the interim firmware fix and troubleshooting

closed 2024-10-24 by crhammock