EverythingSmartHome/everything-presence-lite #29: EPL doc are not actual

v00g100skr, 2023-11-04:
https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/home-assistant-entities.html

Home Assistant Entities and all below is copypasted from EP1 and not valid for EPL
There is pir, temp and humidity etc. and no docs for targets setup

and no explanation about "4 Additional GPIO pins for expansion"
what gpio numbers of esp pins are they?

EverythingSmartHome, 2023-11-04:
Yep, hence the big note at the top of the page describing that the docs are under construction: 
![image](https://github.com/EverythingSmartHome/everything-presence-lite/assets/53482654/00fb8da0-88d8-4fa2-adbb-19ca20941e47

Here is the pinout for the GPIO:

[image]

v00g100skr, 2023-11-04:
thanks)
wiil wait to actual docs)

closed 2023-11-04 by v00g100skr