EverythingSmartHome/everything-presence-lite #32: Keeps going offline

Lampyridae1984, 2023-11-05:
The EPL keeps going offline and reconnecting at random intervals. Anyone have any idea what the issue could be?

```
[16:43:51][D][sensor:094]: 'Target 2 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 2 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 2 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 3 X': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 3 Y': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 3 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 3 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 3 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Target 3 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Zone 1 Target Count': Sending state 1.00000  with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Zone 2 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Zone 3 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[16:43:51][D][sensor:094]: 'Zone 4 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[16:43:51][W][component:214]: Component uart took a long time for an operation (0.17 s).
[16:43:51][W][component:215]: Components should block for at most 20-30ms.
[16:43:51][W][api.connection:126]: Home Assistant 2023.11.1 (192.168.50.182) didn't respond to ping request in time. Disconnecting...
[16:43:51][D][api:102]: Accepted 192.168.50.182
[16:43:51][W][component:214]: Component api took a long time for an operation (0.06 s).
[16:43:51][W][component:215]: Components should block for at most 20-30ms.
[16:43:51][D][api.connection:1050]: Home Assistant 2023.11.1 (192.168.50.182): Connected successfully
[16:43:52][D][sensor:094]: 'Target 1 X': Sending state 1189.00000 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 1 Y': Sending state 1010.00000 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 1 Speed': Sending state 0.08000 m/s with 2 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 1 Resolution': Sending state 360.00000 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 1 Distance': Sending state 1560.07080 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 1 Angle': Sending state 49.65366 ° with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 2 X': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 2 Y': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 2 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 2 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
[16:43:52][D][sensor:094]: 'Target 2 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
```

Littlecheez17, 2023-11-05:
I am also having this issue

jjgadget, 2023-11-06:
I added power_save_mode: none to my wifi section and it has been rock solid.

TripelJ, 2023-11-06:
> I added power_save_mode: none to my wifi section and it has been rock solid.

Cool, is that something you support Lewis? Maybe it should be a part of main config then. 

I also saw people having issues with proxy and improv enabled. Would it make sense to have two configs perhaps that you merge changes to.? So people can choose if they want to/ need proxy but still get all your tweaks and changes you commit.

EverythingSmartHome, 2023-11-06:
Sure if it works then go for it - I'm hesitant to change this by default because it's not clear from the docs what potential downsides there might be. It also seems to be an issue impacting a smaller number of users too, so changing the default might cause issues for a larger group of people possibly?

But if it works for you then perfect, no issues with that.

So it seems there are some possible fixes for anyone reading this:

1. If you don't use Bluetooth Proxy then disable ble_tracker and bluetooth_proxy components
2. Try setting WiFi power save mode to none
3. Try setting the log level to error or none
4. Set a static IP for the node
5. Lock the device to a certain wireless AP. I've read some comments indicating this may happen if the node is roaming from one  AP to another which seems plausible

brofeelgood, 2023-11-06:
No joy with **power_save_mode: none**. I can't flash it because the only values accepted here are "light" (default) and "high".

[image]

How did you guys bypass this?

Also... is disabling esp32_ble_tracker and bluetooth_proxy as simple as setting **active: false**?

brofeelgood, 2023-11-06:
Thanks Lewis, but I'm still a little confused. Doesn't that mean default values (**active: true**) would be used instead? I commented out those lines in the yaml file and ran a validation, and the log is showing what's pasted below.

It'd be great if you could give it a once-over and suggest other areas and settings that I could fine-tune, thanks in advance!

```
INFO ESPHome 2023.10.6
INFO Reading configuration /config/esphome/everything-presence-lite-88ab5c.yaml...
substitutions:
  name: everything-presence-lite-88ab5c
  friendly_name: Everything Presence Lite 88ab5c
  illuminance_update_interval: 2s
  hidden_ssid: 'false'
esphome:
  name: everything-presence-lite-88ab5c
  comment: Everything Presence Lite
  friendly_name: Everything Presence Lite 88ab5c
  name_add_mac_suffix: false
  project:
    name: EverythingSmartTechnology.Everything_Presence_Lite
    version: 1.0.1
  build_path: build/everything-presence-lite-88ab5c
  platformio_options: {}
  includes: []
  libraries: []
  min_version: 2023.10.6
esp32:
  board: esp32dev
  framework:
    version: 4.4.5
    sdkconfig_options: {}
    advanced:
      ignore_efuse_mac_crc: false
    components: []
    source: ~3.40405.0
    platform_version: platformio/espressif32@5.4.0
    type: esp-idf
  variant: ESP32
logger:
  level: INFO
  baud_rate: 115200
  tx_buffer_size: 512
  deassert_rts_dtr: false
  hardware_uart: UART0
  logs: {}
api:
  encryption:
    key: ----
  port: 6053
  password: ''
  reboot_timeout: 15min
ota:
  safe_mode: true
  port: 3232
  reboot_timeout: 5min
  num_attempts: 10
wifi:
  ap:
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: LIGHT
  fast_connect: false
  enable_btm: false
  enable_rrm: false
  passive_scan: false
  enable_on_boot: true
  networks:
  - ssid: ----
    password: ----
    priority: 0.0
  use_address: everything-presence-lite-88ab5c.local
dashboard_import:
  package_import_url: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
  import_full_config: false
improv_serial: {}
esp32_ble_tracker:
  scan_parameters:
    active: true
    duration: 5min
    interval: 320ms
    window: 30ms
    continuous: true
bluetooth_proxy:
  active: true
  cache_services: true
  connections:
  - {}
  - {}
  - {}
globals:
- id: mmwave_update_time
  type: unsigned long
  restore_value: false
  initial_value: '0'
i2c:
- sda: 21
  scl: 22
  scan: true
  sda_pullup_enabled: true
  scl_pullup_enabled: true
  frequency: 50000.0
light:
- platform: status_led
  name: ESP32 LED
  pin:
    number: 14
    mode:
      output: true
      input: false
      open_drain: false
      pullup: false
      pulldown: false
    drive_strength: 20.0
    inverted: false
  internal: false
  restore_mode: ALWAYS_OFF
  disabled_by_default: false

*** truncated ***

button:
- platform: restart
  icon: mdi:power-cycle
  name: ESP Reboot
  disabled_by_default: false
  entity_category: config
  device_class: restart
output: []

INFO Configuration is valid!
```

TB-SE, 2023-11-07:
@EverythingSmartHome

I've changed the log level to ERROR and added a static IP for the node, but how do I lock it to a wireless AP? I'm utilizing a mesh network.

TB-SE, 2023-11-08:
> Sure if it works then go for it - I'm hesitant to change this by default because it's not clear from the docs what potential downsides there might be. It also seems to be an issue impacting a [...]

I've tried all five suggestions, but non work. But I did find this error in the HA system log, any idea what it could be?

```
Logger: aioesphomeapi.connection
Source: runner.py:188
First occurred: November 4, 2023 at 23:30:52 (232 occurrences)
Last logged: 01:04:19

ep1-guestroom @ 192.168.50.90: Connection error occurred: ep1-guestroom @ 192.168.50.90: EOF received
esp32-bt-proxy-hallway @ 192.168.50.193: Connection error occurred: [Errno 104] Connection reset by peer
ep-lite-office @ 192.168.50.91: Connection error occurred: ep-lite-office @ 192.168.50.91: EOF received
ep-lite-kitchen @ 192.168.50.92: Connection error occurred: ep-lite-kitchen @ 192.168.50.92: EOF received
ep1-hallway @ 192.168.50.89: Connection error occurred: [Errno 104] Connection reset by peer
```

TB-SE, 2023-11-11:
@EverythingSmartHome I finally managed to achieve a stable connection. After a lot of trial and error the first option (disabling ble_tracker and bluetooth_proxy components) finally seemed to fix the issue.

But now I have a new issue. The mmWave sensor randomly looses its target/targets and reports 'clear' for a couple of seconds. Is there a way to adjust the sensitivity within the code? I also tried to replace the senor with the DFRobot SEN0395, but that didn't work at all. Majority of the EP Lite sensors within Home Assistant became unavailable.

EverythingSmartHome, 2023-11-12:
You can try to extend the occupancy off delay for a while longer to keep the status as detected. 

What config did you use for the SEN0395? It's completely different

TB-SE, 2023-11-12:
> You can try to extend the occupancy off delay for a while longer to keep the status as detected.
> 
> What config did you use for the SEN0395? It's completely different

Ah, ok ok. I thought there was some sort of intercompatibility within the code, since I could not find any clear reference to the LD2450 within the [config](https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/everything-presence-lite-ha.yaml) file on Github (then again, I might be blind). I apologise for any confusion, I'm not very good at this. 😳 I'll keep experimenting with the off delay and placement of the EP Lite. Thank you for clearing this up!

TripelJ, 2023-11-12:
@EverythingSmartHome 
Just a suggestion. Maybe remove the Bluetooth settings from the main file, and then people can opt in if they want to try and play around with the proxy/BLE if they need it on the same device. Then you can still mention it as a feature to run on the EPL but then it's not on as default if most people buy it for the presence just to work :)

CDAutomate, 2023-11-13:
Also having similar issues. I have 4 different Everything Presence Lites, all 4 disconnect and reconnect at seemingly random times.

TripelJ, 2023-11-13:
Can only recommend turning off ble & Bluetooth proxy that helps. At least for me and some others. 
The default config when you flash the EPL comes with the configuration where it's enabled. 
A different approach could be to ask during flash of the firmware if you will use it only for presence or presence + Bluetooth Proxy and refer to two different configs.

CDAutomate, 2023-11-13:
@TripelJ  Appreciate the reply. I don't see any of the config mentioned above. 

Here is what I have:

> substitutions:
  name: basement-presence-sensor-24ea30
  friendly_name: Basement Presence Sensor
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key:  REDACTED

>  wifi:
> 
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password

TripelJ, 2023-11-13:
Hi, 
Everything that makes the EPL tick comes from the included package. 

packages:
EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main

CDAutomate, 2023-11-17:
@TripelJ  Thanks. Sorry these were the first devices where I pulled a whole package vs just copying code and modifying from there.

I believe it was a combination of all 3 of the following that fixed my issues:

- Set a Static IP
- Set Power Save Mode to None
`wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  manual_ip:
    static_ip: ###.###.###.###
    gateway: ###.###.###.###
    subnet: ###.###.###.###`

-Commented out the BLE Tracker
`#esp32_ble_tracker:
#  scan_parameters:
#    active: false

#bluetooth_proxy:
#  active: false`

EverythingSmartHome, 2023-11-19:
FYI I added a non bluetooth proxy option to the firmware install page here: https://everythingsmarthome.github.io/everything-presence-lite/updating.html

TripelJ, 2023-11-20:
Don't know if it's changes in the config or latest esphome updates or something completely different. Mine was more stable, not entirely good after I removed ble and Bluetooth proxy manually. Now I switched to the new config without ble that was just added. 
But recently it's just now more unavailable to such extend that it does not work. When it turn it off/on it works one time. And then it's like it just hangs and goes in out unavailable. 
I have 8 other esphome devices/sensors that works great. I have also tried save mode : none without effect. Only missing to try the static ip. But it's not because it's changing nor roaming and the signal them is good.
Anyone else just have theirs working now with these changes or still experience the unavailability despite all the recommendations in place. I'm running out of ideas.

TripelJ, 2023-11-20:
> FYI I added a non bluetooth proxy option to the firmware install page here: https://everythingsmarthome.github.io/everything-presence-lite/updating.html
FYI:
The new file includes esp32_improv: which the original file with bluetooth does not have. 
So BLE is still active and you for instance cannot use power_save_mode: none because of ble still active.

brofeelgood, 2023-11-21:
> > The new file includes esp32_improv: which the original file with bluetooth does not have.
> > So BLE is still active and you for instance cannot use power_save_mode: none because of ble still active.

@EverythingSmartHome: any chance of removing the **esp32_improv** section in the no-ble package as well?

TripelJ, 2023-11-21:
> @EverythingSmartHome: any chance of removing the **esp32_improv** section in the no-ble package as well?

I think the best thing we can do atm is to download the configuration locally and test it out and tweak where needed. Once we have something stable for a while we can inform Lewis about what config might work for the majority.

EverythingSmartHome, 2023-11-21:
It was my understanding that esp32_improv would only be active on initial setup but could also be a bug in that I guess - try again now it's removed

rev010, 2023-11-23:
> FYI I added a non bluetooth proxy option to the firmware install page here: https://everythingsmarthome.github.io/everything-presence-lite/updating.html

Thank, i'm struggling how to disable bluetooth exactly etc. Glad you added an extra option, but I get the following error using the non-blutooth option:

[image]

Tried a different browser and on another computer/laptop. The standard option works without a problem.

dylan09, 2023-11-27:
Had the same behaviour. Constantly  offline in HA. 
Tried everything mentioned in this thread: disable ble, disable improv, ...

Finally I changed the code, so that `publish_state` is only called when state of sensor has really changed.
Don't know why that is working? The `sensor` code also does this check. But for me it is working for days without error.

Done this change first only in the non-ble configuration. Maybe there are others, that can confirm its also working for them.
Then the changes could added to the ble version.

EverythingSmartHome, 2023-11-28:
@dylan09 Interesting idea, have you tried re-enabling bluetooth proxy with this change to see if that causes timeouts again?

dylan09, 2023-11-28:
> @dylan09 Interesting idea, have you tried re-enabling bluetooth proxy with this change to see if that causes timeouts again?

Will try the next days. But I think it will also work with ble enabled.
And I couldn't find a reason why my change should have such an impact. There is a specialized class "Deduplicator" which prevents sending sensor values if they don't change.

dylan09, 2023-12-02:
Have enabled ble a few days ago. Installed one device in our livingroom. There is a lot of traffic. And for test I have defined two different zones.
Only one or two times error `EOF received` in log files. Seems nearly stable.

The second is installed in stairway hall. Not as much traffic as in our livingroom. No more errors in log files.

Would like to test a few other rooms. But need more devices. 😀 Unfortunately just sold out 😫😱 

Also digged a little bit inside the ESPHome code. And could not find a valid reason why this change has an impact to this error? I am a little bit busy the next days. If I get a little bit of time I will change back the code and do another test and have a look if the error is back again than.

EverythingSmartHome, 2024-10-08:
I believe this to be resolved but let me know if not

dylan09, 2024-10-08:
Yes. The fix publishing state only when changed makes it a lot better. The sometimes occurring disconnects may also related to WiFi or something else. But for me it is now working.