EverythingSmartHome/everything-presence-lite #33: Wrong Firmware installed for EP Lite

actran, 2023-11-05:
I bought EP Lite and tried to install.    

I am on Lite documentation site, troubleshooting page.  When I clicked on Updating button on this URL:  https://everythingsmarthome.github.io/everything-presence-lite/troubleshooting.html, it actually installed the EP One firmware.

Where/how can I get EP Lite firmware?

EverythingSmartHome, 2023-11-05:
Hi, you can find it here: https://everythingsmarthome.github.io/everything-presence-lite/updating.html

actran, 2023-11-05:
Thanks Lewis.  You can close this issue.   You may want to update the "Updating" button at https://everythingsmarthome.github.io/everything-presence-lite/troubleshooting.html to correct URL.

closed 2023-11-19 by EverythingSmartHome