EverythingSmartHome/everything-presence-lite #35: Disappearing presence for two targets with different speeds

4D4M-Github, 2023-11-07:
Hi Lewis,

I would like to share my next initial experience. If you have two targets one at relative rest, the other moving and moving out of max range, it will clear presence for 1 second and then return to detection. I'm not sure, but I think they created a solution for the same phenomenon here:
[https://community.home-assistant.io/t/hlk-ld2450-initial-experiments-to-connect-to-homeassistant/578878/115?u=4d4m](url)

Maybe we could implement this solution if we have the same problem?

Thanks in advance!

EverythingSmartHome, 2023-11-07:
Sure, this was already implemented a while ago for the main occupancy sensor and in EP1 firmware - I've added it now for the zones if you'd like to update to the latest :)

4D4M-Github, 2023-11-08:
Thank you for your quick response. I really appreciate it. 
I will start testing it as soon as my 2 very fast moving miniature targets wake up (:

kloppy, 2023-11-29:
Most likely this is related to #27. The issue is that the zone occupancy is checked against the target active state from one second ago, instead of the current state.

closed 2024-09-08 by 4D4M-Github