EverythingSmartHome/everything-presence-lite #36: Presence Lite Pinout

chksome, 2023-11-09:
Hey ESH,

May we get a graphic or similar showing what pins of the ESP32 go to each of the pins on the Presence Lite? PJRC kills it at this with the [Teensy 4.0 pinout cards](https://www.pjrc.com/teensy/card10a_rev2_web.pdf), for example. I've seen the documentation mention four additional pins for other things we might want to hook up, and am sure they go to the block next to presumably the boot or reset button, but which are they? The block only has 3.3v silk screened. Ideally it'd be nice to know where _all_ of the pin headers, including those that connect to mmWave sensors, go.

Thanks!

EverythingSmartHome, 2023-11-09:
Yes it's definitely planned when I get some time for sure!

pieterdevriesch, 2023-12-26:
Just to get you started here's my findings from some probing, no fancy graphics:

```
DFRobot Connector (top)
[1][2][3][4][5][6]

1:	5V
2:	GND
3:	GPIO19 / pin38
4:	GPIO18 / pin35  
5:	GPIO17 / pin27  < in use by LD2450  |  UART TX
6:	GPIO16 / pin25  < in use by LD2450  |  UART RX

LD2410 connector (bottom)
[1][2][3][4][5]

1:	GPIO16 / pin25
2:	GPIO17 / pin27
3:	GPIO18 / pin35
4:	GND
5:	5V

Debug connector (next to 5V connector)
[1][2][3]
[4][5][6]

1:	GPIO32 / pin12
2:	GPIO26 / pin15
3:	3.3V
4:	GPIO25 / pin14
5:	GPIO27 / pin16
6:	GND

```

So there's quite a few usable pins broken out even without any soldering. GPIO18,19,25,26,27,32. 

In fact from reading the configuration the default only uses 16 and 17 to talk to the mmwave sensor and 14 for controlling the board LED and everything else is pretty much free. So far I've added a PIR sensor to GPIO32, and I'm going to add an environmental and light sensor next. (Just need to print the next version of the cover)

BertrumUK, 2024-07-06:
@pieterdevriesch Thanks for the Pinout details at the bottom of your entry. I would like to add an RGB light to the EPL Lite I received the other day. Did you just add your sensor and GPIO details in the visible code in ESPHome as the main code gets it info back from github ?

pieterdevriesch, 2024-07-06:
Yes, the EPL part is pulled from this Git repo via the packages bit, and since we're just adding and not conflicting you can just add your extra stuff at the bottom. Actually I think the way esphome works the explicit config you add in the yaml will override anything linked but I'm not 100% sure on that. Either way for this it doesn't matter.

BertrumUK, 2024-07-06:
Thank you @pieterdevriesch 

I have just added the code to the bottom of the config using platform: ledc and your provided GPIO pins - not showing any errors and it validates ok, but it fails when I try to upload it so I am guessing that there is a conflict with the main code. Will need to check when I finish work and get access to my PC, not so easy using an ipad :wink: -

[image]
[image]

BertrumUK, 2024-07-06:
Fixed the above by running the "Clean Build Files" command before installing :grinning:

Thanks again for your work and help on this.

pieterdevriesch, 2024-07-06:
Glad to hear! Yes seeing the location of the error that usually means it's nothing to do with your yaml as it's finished compiling already. But ESPHome sure could work on some error catching to make things clearer :)

So from your config am I to read that you've got a presence sensor at your bathtub that tells you when the water's a nice temperature to get in? :)

BertrumUK, 2024-07-06:
`So from your config am I to read that you've got a presence sensor at your bathtub that tells you when the water's a nice temperature to get in? :)`

No, I have a hot water tank with a temperature probe connected to it. I used to have a Pi4 with an LED cut into the case to show if the water was green or red ( hot or boiling) for a shower or bath, and blue was cold. 

Since I moved over to a mini PC running Proxmox, I was after a way to have another LED to show the water temperature in the living room as we watched TV. With my new EPL positioned under the TV the GPIO pins allow me a new way to have an LED to show the water temperature in the tank ( I only heat the water up in the morning or the evening to save gas :wink: )

pieterdevriesch, 2024-07-06:
Oh that's brilliant. Hang on while I add an RGB led to my EPL as well :)
Time to go print another case I guess.

BertrumUK, 2024-07-07:
Did a quick and dirty LED mod after work last night. Still need to work on the LED placement, but its good enough for now :wink:

[image]

pieterdevriesch, 2024-07-07:
Nice. If you're interested, here's my modified cover:
[https://makerworld.com/en/models/530068](https://makerworld.com/en/models/530068)

BertrumUK, 2024-07-08:
Neat, is that the latest update with the RGB led using similar code to mine?

pieterdevriesch, 2024-07-08:
Actually this is an addressable LED so it only uses one GPIO. Also less soldering, just +/-/Din. But once it's added in ESPHome everything after that will be the same.

BertrumUK, 2024-07-08:
I have a bunch of 4 pin leds so this is a way of using another one up. Going to modify some 8 pin jumpers to a 3x3 jumper and add the led to that instead of using single leads per LED. The spare gpio pins would be ready for future use then.

chiebert, 2024-07-08:
Since the OP is asking for some docs re. pinouts, it would also be useful to include some details about the following, for those of us planning using some of these extra GPIO pins :)

1. The 3.3v pin in the breakout header (the 3x2 block on the lower left): I see it comes from the little AP2112 3v3 regulator that (I assume) is also powering the ESP. What's the max current available from that pin without causing problems? The AP2112 is rated at 600mA, and the ESP32-WROOM-32E wants 500mA, so perhaps when the documentation catches up, that pin should be listed as 100mA max?
2. The 5v+ pin in the 2-pin header next to the USB-C connector: is this coming straight off the USB power bus? Docs should give some indication what the power limit might be.
3. What are the two buttons on the board? (bottom edge between the 3.3v and 5v+ pins, and right edge)

jamiewhitham, 2024-10-29:
Trying to get the CO2 module to work from the picture it looks like it is only connected to pins 2&3 of the 6 pin header. But I'm guessing it should really be 2,3,5&6?

Also struggling to find how to update firmware as went through auto ble mode first of all now I can't see how I change it just says 1.2.0 upto date in HA.
[image]