EverythingSmartHome/everything-presence-lite #37: Target 2 not considered Occupied?

Zeoic, 2023-11-11:
Hello,

I just got the Lite in today and have been loving it! It is so much quicker and more accurate than I expected!
I have run into one issue, however. If I sit very still in a room I will suddenly be Target 2 instead of Target 1, or two people are in a room and Target 1 leaves the room, there are some instances where the only remaining Target is Target 2. Target 1 will be all 0s while Target 2 is still detected and updating just fine. 

This situation seems to be considered Clear rather than Occupied though. I wonder if making the Occupancy sensor be marked as Occupied when any Target is detected would be possible? That or make Target 2 move down to Target 1 when Target 1 is no longer being detected?

EverythingSmartHome, 2023-11-11:
Glad to hear it!

Unfortunately the target tracking is handled on the mmWave sensor firmware and not on the esp32 firmware which we don't have access to.

The thing where target 2 doesn't have occupancy as detected though shouldn't be the case, that sounds like a bug that I'll need to look into

Want100Cookies, 2023-11-11:
I experienced this several times. Perhaps just wait on hareeshmu's PR [#5674](https://github.com/esphome/esphome/pull/5674) to be released and use that component instead of manually serializing?

Zeoic, 2023-11-11:
Let me know if there is anything I can test to help out! I have found a certain position in the room that reliably makes me Target 2 after about a minute of sitting still, so I could pretty easily test anything.

In the meantime, I have set a 10-minute buffer on how long it needs to be clear before turning off the lights in hopes that it will smooth out the no Target 1 stuff. So not the end of the world.

kloppy, 2023-11-12:
It is due to to this check only considering target1:

https://github.com/EverythingSmartHome/everything-presence-lite/blob/261c2c0303e8a8a65d243e783daeaa3521839398/everything-presence-lite-ha.yaml#L63C1-L76C8

This possible alternative lambda works for all targets 1 .. 3:

```c
return (id(target1_distance).state != 0
  || id(target2_distance).state != 0
  || id(target3_distance).state != 0);
```

4D4M-Github, 2023-11-13:
I think this issue is the same as [the one I opened](https://github.com/EverythingSmartHome/everything-presence-lite/issues/35) and I fear that the problem will be in the LD2450 firmware.
In the end I avoided it by dynamically changing the delay time with another sensor.
@EverythingSmartHome (Lewis), if you wish, feel free to merge the two issues or close mine.

EverythingSmartHome, 2023-11-13:
> It is due to to this check only considering target1: [...]

Good spot, if you have the time to do a PR for this that would be hugely appreciated, otherwise I can tackle soon

kloppy, 2023-11-13:
I am very inexperienced with git(hub), but it is a good opportunity to change that a little.

I'll try!

EverythingSmartHome, 2023-11-14:
If you would like to try and update to the latest version and give it a try that would be great!

Thanks @kloppy

EverythingSmartHome, 2024-10-08:
I believe this to be resolved but let me know if not

closed 2024-10-08 by EverythingSmartHome