EverythingSmartHome/everything-presence-lite #39: Signs and Offsets of Target Coordinates and Angles

kloppy, 2023-11-12:
I did some digging after getting confusing / inconsistent data for the target coordinates and angles.

The way the coordinates are extracted from the UART data is a bit complicated. First the data is read, and then the following effectively flips the sign twice (comments by me):

```c
int16_t p1_x = (uint16_t((bytes[5] << 8) | bytes[4] ));
if ((bytes[5] & 0x80) >> 7){ // if the sign bit is set (it is a negative number, but we read this as a number > 32768)
  p1_x -= pow(2, 15);        // then subtract 32768 => we end up with a positive number
}else{                       // if the sign bit is NOT set (it is a positive number)
  p1_x = 0 - p1_x;           // we negate, ending up with a negative number
}
p1_x = -p1_x;                // negate, second sign flip
```

The last line makes this correct (for now), but this line is missing for target 2 and 3. In the end, the reported coordinate has the wrong sign though because it is flipped a third time here:

```c
id(target1_x).publish_state(p1_x * -1); // third sign flip
```

The angle of target 1 is calculated while the x coordinate is correct (before the last flip), but then it is also modified during publish, other than angles for target 2 and 3.

```c
id(target1_angle).publish_state(p1_angle - 90); // 90 degrees are subtracted
```

The resulting angle is inconsistent to targets 2 and three. You can see that when the targets switch and the angle suddenly jumps (e.g. it was 10 ° as target 1 and then it is 80 ° as target 2). Also its direction is opposite to the angles of targets 2 and 3. Convention is that an increasing angle corresponds to counter-clockwise rotation. Target 1 angle follows this, the other two do not. 

A simpler approach would be the following, given for target 1 but the same for the other two:

```c
uint16_t twobytes = (bytes[5] << 8) + bytes[4];
int16_t p1_x = (data & 0x7FFF) * ((data >> 15) ? -1 : 1);
twobytes = (bytes[7] << 8) + bytes[6];
int16_t p1_y = (data & 0x7FFF) * ((data >> 15) ? -1 : 1) * -1; // here we flip delibeately to work with positive y coordinates
twobytes = (bytes[9] << 8) + bytes[8];
int16_t p1_speed = (data & 0x7FFF) * ((data >> 15) ? -1 : 1); // I think this is cm/s, can be converted for publishing 
twobytes = (bytes[11] << 8) + bytes[10];
int16_t p1_distance_resolution = (data & 0x7FFF) * ((data >> 15) ? -1 : 1); 
```

Once the math works, you can for example add rotated zones more easily, which are really useful if the EPL is in a corner of the room and nothing aligns with the axes of the device.

Hope it helps!

kloppy, 2023-11-29:
I have uploaded a package f1799d1781a315ed78824890588df905f5a4dec1 and example e9d1971fad5545ef15785587fb998f00c9cb0b2e .

This package fixes #27, #28, #35, #37 and #39.

Otherwise it adds
* zone rotation (non-axis-aligned zones)
* x flip switch (to mount the EPL upside down)
* oversampling for illuminance sensor

it changes
* zone units to cm
* global occupancy only diagnostic (no max distance, no off delay)

and it removes
* target resolution (always 360 mm anyway)
* target speed
* wifi power save
* bluetooth
* improv

Feel free to use the code. Or if you want PRs for specifics, tell me for what parts.

crhammock, 2024-02-07:
sorry, I am a bit of a github and esphome noob.  How do I get these updates in my EPLite to test?  Can I replace the EverythingSmartTechnology.Everything_Presence_Lite package with the yaml linked above?  Not sure how to convert the https:// protocol syntax into github://

kloppy, 2024-02-11:
You can download the two files to ESPHome, change EPL_example_device.yaml to your liking and flash onto your EPL. The package is then included locally.

crhammock, 2024-03-04:
Thanks @kloppy I have installed your package and it seems to be working.  So I guess once your fix is committed to the main branch a lot of open issues can be closed.  I did notice that the controls are all measured in cm and the targets are measured in mm.  Not sure if this is something you changed or I just didn't notice in the main branch but I think it is clearer if both measured the same.  Centimetres makes sense based on the application however it is called a Millimetre Wave sensor so maybe mm is more appropriate.

Thanks again

kloppy, 2024-04-11:
Glad it works for you too! I've been using it since November and not found any more issues either.

Indeed I changed the units to cm, as mm is too accurate for both the application and the sensor resolution, you then have to work with longer numbers with a lot of non-significant zeroes. "dm" would have been even better, but it is quite uncommon.

I just wish I could buy more EPLs, I am always too late when there is new stock. :)

EverythingSmartHome, 2024-04-11:
Thanks for reminding me I still need to look through this with the changes that were done recently!

More Lites will be in stock in around 6-7 weeks - a while longer to wait, but we've made some nice refinements to it that will make it worth the wait (IMO) and it's by far the biggest batch we've ever ordered so hoping stock will last this time!