EverythingSmartHome/everything-presence-lite #40: Consistent labeling - to help with sort order

stevearmitage, 2023-11-13:
There are some properties labeled: Target N [PROPERTY] and then some labeled Zone N [PROPERTY]

Could they all be labelled Target N or Zone N (i prefer the 2nd) so that they appear together in the list - or a hybrid as below:

Zone 1 Target Active (Presence?)
Zone 1 Target Angle 
Zone 1 Target Distance
Zone 1 Target Resolution
Zone 1 Target Speed
Zone 1 Target X
Zone 1 Target Y
Zone 1 Occupancy
Zone 1 Target Count

EverythingSmartHome, 2023-11-13:
Hmm sorry I don't follow, target is different to a zone - target is the person moving (persons coordinates, speed, angle etc) while a zone is a static area you define to track a target inside of, i.e is a person inside a zone (area)

How would a zone have an angle and speed when it's a static area?

stevearmitage, 2023-11-13:
Ah, so have i misunderstood how the properties work

Is it that:
1. There are zones, which identify if one or more targets are present in the zone
2. There are targets which are tracked independently of the zones

If so, then please ignore me :o)

EverythingSmartHome, 2023-11-13:
That is indeed correct!

stevearmitage, 2023-11-13:
Perfect - thanks Lewis. Appreciate the correction.

closed 2023-11-13 by stevearmitage