EverythingSmartHome/everything-presence-lite #42: Occupancy Off Delay per zone

crhammock, 2023-11-15:
Not sure if this is an issue or a feature request.  The occupancy off delay only seems to effect the main sensor occupancy state not the per zone occupancy state.

Is it possible for this value to apply to all zones occupancy or to add additional occupancy off delay entities to each zone?

EverythingSmartHome, 2023-11-15:
Have you updated in the last couple of days? This should have already been added earlier this week but I haven't yet tested it

crhammock, 2023-11-15:
I checked for updates on Monday and nothing.  I see now there is an update.  I am performing the update now and will report back and close the issue if all good.

Thanks for all your efforts

TripelJ, 2023-11-15:
Maybe I didn't notice. But where do you see the update available. ?
I only see release 1.0.1 from 21 days ago.

EverythingSmartHome, 2023-11-15:
There have been some bug fixes that haven't yet went into a "release" yet but are available if you press update or use the install from the user guide.

Still figuring out the whole GitHub thing

crhammock, 2023-11-15:
seems to be working great, thanks again

closed 2023-11-15 by crhammock