EverythingSmartHome/everything-presence-lite #53: Unable to update Firmware EP Lite

adam50hhh, 2023-11-23:
I followed the instructions on https://everythingsmarthome.github.io/everything-presence-lite/updating.html. When I try No Bluetooth Proxy, fails to download manifest. So I click on the regular Home Assistant one, Connect and when I try Install Everything Presence Lite HA it just continues to circle on "Preparing Installation". For over an hour. I have tried several times with Chrome and Edge. I decided to configure the WIFI and got it to configure and connect to HA and works great! Viewing videos on YouTube, it appears there is a newer firmware version with additional options. This is my first ESPHome device, any assistance would be appreciated. 
Firmware 2023.7.7 (Sep 1 2023, 21:54:15)
Hardware 0.1
Using ESPHome.

ThiSaadeh, 2023-11-28:
No bluetooth proxy doesn't work me either, same error.

dougoberman, 2023-11-29:
The same issue exists for me. The "No Bluetooth Proxy" firmware fails, and presents the "Failed to download manifest" error every time

rev010, 2023-11-29:
Same issue here. I mentioned the "Failed to download manifest" in the "Keeps going offline" thread, but strangely enough no resposne form Lewis on that. But he commented after that on maybe another solution presented by someone else.... Hope this issue gets fixed.

Vinl0c, 2023-11-30:
Same for me, no response from Lewis yet?

adam50hhh, 2023-12-01:
Agreed, either well deserved vacation or working behind the scenes. 
The issue is I did try the web flasher (Chrome and Edge) for the BLE version, it just keeps spinning for almost an hour saying preparing for installation.

EverythingSmartHome, 2023-12-01:
Sorry everyone, been super busy, I'll try to fix this weekend. Meantime if you want to try it, create a manual config in ESPHome add-on and use the code from here: https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/everything-presence-lite-ha-no-ble.yaml

rev010, 2023-12-01:
> Maybe Lewis is taken a well deserved vacation. :) I don't know how familiar you are with ESPhome. But if to get started you need to flash using the web flasher, just use the BLE version then after device is added in HA and ESPhome you can change the config url to the non BLE or do the config locally and remove the Bluetooth proxy, ble etc.

@TripelJ Yeah Lewis sure deserves it! First time using ESPHome, so it was just replacing the packages line in the config in ESPHome for the device for the no-ble yaml. The @main at the end got me off guard. Thank you!!

StayPirate, 2023-12-10:
Hi all, same problem here. Any update?

adam50hhh, 2024-01-10:
No updates yet. 
I was not able to manually install the manual config in ESPHome using the HA add-on.
I tried using web.esphome.io in Chrome from HA, it gets stuck on "Connecting". I have tried a 3.0 usb port and a 2.0 usb port. Different ports continue to produce the same result, it gets stuck on the "Connecting" screen where I can't install or Prepare For First Use. Computer can see the COM paired.

EverythingSmartHome, 2024-01-11:
Sorry for the delay, I resolved this over the holiday period if you would like to try and use the non bluetooth firmware from the user guide.

If it's getting stuck on connecting with your computer, you can try holding the boot button whilst hitting connect to see if that helps

adam50hhh, 2024-01-13:
Holding the boot button while hitting connect worked for me! Much appreciated!

closed 2024-01-13 by adam50hhh

Taomyn, 2024-02-05:
I wanted to add my experience to this as I encountered similar problems when I got my two Lites a few weeks ago. Same symptoms but nothing would work, until while trying different USB ports (onboard and hub) I got to my USB-2 ones, and immediately I could update. I'd been trying all the free USB-3 ports on my PC, so I disconnected something from USB-2, connext the Lite to it and that was successful.