EverythingSmartHome/everything-presence-lite #56: Occupancy gets stuck on 'Clear' state

TB-SE, 2023-12-21:
Has anyone else had the issue that the occupancy after a "longer" period goes from detected to clear and refuses to change unless you leave the room. Same issue with Zone occupancy. The weird part is that the sensor still counts a person/persons in the zone and it definitely seems to register my/our movements. The issues also happens when someone is sitting still, about 50 cm away from the sensor. Did I miss something in the settings?

dancwhite, 2023-12-28:
I have the exact same problem

rmccondach, 2024-01-27:
I have this issue, but the other way round. Last night the one in my Livingroom got stuck on as detected, well after the timeout I had set. During that time there was nothing re-triggering it as none of the targets and zone occupancy were always set as occupied but with no target.

carmicp, 2024-02-02:
My EPL is located in a very small bathroom. I have occupancy stuck to detected and target 1 stuck to on with no targets or movement. Often I get occupancy detected and target 1 off. Then there's  occupancy clear,  target 1 on with the target moving through the room. All I can do is reboot then it works for a little while until it freezes in one state or another. I also have had the occupancy clear,  target 1 off but the target is registering and moving. Help!!

pointful, 2024-02-08:
I see the same issue regularly of detection stuck on, mainly that Zone 1 will be stuck on, even though there are no active targets and general occupancy is clear. Here is an example of an EPL in that state:

[image]

RichyA, 2024-02-12:
I also have this problem (**Occupancy** stuck as **Clear** even though **Target 1 Active** is **On**).

This appears to resolve after **Target 1 Active** remains **Clear** for the duration of **Occupancy Off Delay**. (Actually, on inspecting the code, it appears to require the **Target 1 Distance** to be **0cm**; however, as **Target 1 Active** is based on **Target 1 Distance**, these are effectively the same thing.)

This can be forced without leaving the room by temporarily setting the **Occupancy Off Delay** to **0s** and remaining very still for a few moments.

The **Occupancy Off Delay** logic appears to use the **delayed_off** filter which is a built in feature of ESPHome that does exactly what you expect it to do. It delays sending the **Off/Clear** status of **Occupancy** to Home Assistant until the **Occupancy Off Delay** has passed. If **Occupancy** reverts to **On/Detected** before the timeout, the **Off/Clear** status should never be sent to HA in the first place.

That's starting to feel like a bug in ESPHome, where it's sending the **Off/Clear** status but doesn't think it has, so never sends the **On/Detected**.

I'm going to see if HA's logbook has a record of what set the **Occupancy** to **Clear**.

RichyA, 2024-02-12:
I've found an instance of this happening:
[image]

What stands out to me here is that **Target 1 Distance** went non-zero _very shortly_ (14:38:29) after **Occupancy** went Clear (14:38:27). It's also notable that **Target 1 Active**, which is directly based on **Target 1 Distance**, _also_ didn't update correctly at this point. It didn't update to **On** until **Distance** dropped to zero for a moment.

Based on my understanding of the YAML code, there's nothing that the EPL or its code is doing wrong here. I do think it's an ESPHome bug.

Hmm, alternatively... The ESP is having to recalculate **Occupancy** _and_ **Target 1 Active** _every_ time **Target 1 Distance** updates, which is pretty frequently. Is it possible there's just a bit too much going on here and the ESP is getting overloaded or hitting some kind of weird race condition?

pointful, 2024-03-06:
I updated several EPLs to ESPHome 2024.2.0 a couple of weeks ago and have not had this problem since. Anyone having the same experience?

TB-SE, 2024-03-12:
> I updated several EPLs to ESPHome 2024.2.0 a couple of weeks ago and have not had this problem since. Anyone having the same experience?

Agree, mine work great too. Anyone else? If not, I'll close this issue at the end of the week.

rmccondach, 2024-03-12:
I've not had an issue with occupancy sticking "on" either recently..

closed 2024-03-19 by TB-SE