EverythingSmartHome/everything-presence-lite #68: Can't connect to ESP. Please make sure your YAML file contains an 'api:' line.

ryanandrewbaker, 2024-02-02:
Installing either firmware (proxy or no proxy) I get the following error when trying to connect the EPL to home assistant manually. 

Can't connect to ESP. Please make sure your YAML file contains an 'api:' line.

The EPL is not autodiscovered by home assistant, although I can see it gets an IP on my router. So attempting to add it as an ESP32 device manually gets the above error.

EverythingSmartHome, 2024-02-02:
If your getting that error then generally means you are using the incorrect IP address or the IP isn't accessible from Home Assistant. If auto discovery isn't working, is it on a different vlan and do any firewall rules apply?

ryanandrewbaker, 2024-02-02:
Wow, immediate response, cheers!

I don't believe either of those is the case. The IP is what's reported by my google wifi in the home app, so I can't see how it'd be different. And the firewall in google wifi is also not configurable as far as i'm aware so there are no custom rules, it just... is. Hmmm. If it's my network, I may need to wait until I move and stop using google wifi then unless there are any other suggestions?

EverythingSmartHome, 2024-02-03:
Is your Home Assistant server connected via the same router? From another machine, can you try running a ping to the Lites IP?

ryanandrewbaker, 2024-02-03:
HA is wired, but same subnet as the Lite. Can def ping the EPL. Very strange. 

I'm switching to a full omada setup which will allow me better control and troubleshooting in a few weeks, I may just put the lite on the shelf until then and revisit. Thanks for getting back to me so quickly. Google wifi is... not great.

Smiley1244, 2024-06-28:
I've just flashed my new lite via the website and i'm getting the same. I can confirm I'm manually trying to connect to the correct IP address and it is pingable from other machines on the network.

Smiley1244, 2024-06-28:
I've reflashed again via the site and it's now working :)

chondro878, 2024-08-20:
Please help as im seeing the same issue with 5 EPL devices. I feel people are seeing this and just "by luck" are able to resolve the issue!

EverythingSmartHome, 2024-08-21:
Cross posting for others searching:

Chondro has Home Assistant on a different subnet to SSID they have connected their devices too. We have confirmed over email that Chondro's Home Assistant cannot ping devices on that SSID either because of a network misconfiguration or because of firewalls rules blocking access

Home Assistant needs to be able to communicate with devices across the network or the integration won't work

closed 2024-10-08 by EverythingSmartHome

HarvsG, 2025-05-27:
@EverythingSmartHome 
I am getting this issue even though I can ping the EPL from HA
[image]
[image]
[image]