EverythingSmartHome/everything-presence-lite #73: u

mark-mcdougall, 2024-02-09:
I have 2 everything presence lites - unfortunately neither of them seem to be working correctly. 

EPL1: Running firmware 2023.12.8
EPL2: Running firmare 2023.12.9

What works on both devices:
Ocupancy
Target 1: Active, Angle, Distance, Resolution, Speed, 1 X, 1 Y (I haven't tested with multiple targets)

On each I set up 2 zones and set the XY values of the top left and both right corners as per this video https://www.youtube.com/watch?v=W8vOWSbXzKY. The zones I have set appear correct (I took them from the device Target 1 X and Target 1 Y values as instructed in this video https://www.youtube.com/watch?v=wCwMNSeLRcw

On EPL1:
Zone 1 occupancy shows as 1 but zone 2 does never changes from 0. However looking at the Logbook it appears that both zone 1 and zone 2 occupancy sensors are being turned on and off but I can't see this from the device's screen in HA

On EPL2:
On the device screen neither zones' occupancy shows as on. Via the Logbook zone 1 shows as turning on and off but zone 2 just shows as going on and then to unavailable

As I was having so much trouble I decided to download smarthomejunkie's blueprint to see if I could make any progress.
On EPL1, in the blueprint I set up the same 2 zones and set zone 1 to turn on/off light 1 and zone 2 to turn on/off light 2.
Weirdly the zone 2 blueprint automation worked but zone 1 did not. Even more strangley was when I made a normal HA automation based off of the zone occupancy instead of the XY values, zone 1 worked but zone 2 didn't. 

This leads me to believe that for EPL1 there is an issue with the zone 1 XY values and the zone 2 occupancy sensor. I have set the XY values multiple times for both zones and seen no improvement. Using the blueprint for zone 2, automation for zone 1 kind of works but it is messy and the response time using the zone occupancy value is very slow compared to the XY values.

On EPL2 neither the blueprint or automations work for either zone.

I should note that for the blueprint I set the XY coordinates for the other 2 unused zones to zeroes and the triggered entities to None so that I didn't get an error when I tried to save the blueprint.

Has anyone experienced similar issues? 

Any suggestions what I can try to get these up and running?

jonwork88, 2024-02-18:
Also not able to get the Zones to work.  
Device info
Everything_Presence_Lite
Firmware: 2023.12.9 
Hardware: 1.0.2 

Set up 2 zones.  Followed the instructions from the same videos listed above.  Input the coordinates from Top L and Bottom R for each Zone.    

The sensors read me, the main occupancy binary sensor reads me, but I don't register in any of the zones.

I Have tweaked my x,y's several times, and also switched where I take my readings (ie:  did from top R to Bottom L as well), doesn't appear to make a difference.  Rebooted the sensor after making changes... rebooted HA after making changes. 
 Not sure what else I can do.

Nothing in my EPL yaml about Zones (not sure if there is supposed to be).  

Are Zones not working right now?  Any suggestions?

Appreciate any help you can provide.

Thnx.  Jon

EverythingSmartHome, 2024-02-19:
Hi guys (sorry I missed this when it was created)

Can you share some images of the coordinates and settings you are using?

EverythingSmartHome, 2024-02-25:
Did you try switching the beginning and end X coordinates around?

jonwork88, 2024-02-26:
only the x coordinates?  I changed both the x&y....  but not "just" the X

jonwork88, 2024-02-26:
I flipped the coordinates and I have it "sometimes" registering, and how its registering brought up a thought.  

Will the "Zones" that are created off of the 2 sets of coordinates always be a perfect square/rectangle aligned with the orientation of the EPL?  If so, that might be part of my problem as I have it coming off a side wall at an angle, it its probably not able to create the zone I am looking for with only 2 sets of coordinates....  

Below are a couple of images that show the placement and ~ angle of the EPL and the type of rectangle I want, vs what I think "might" be happening...  Does that make sense?

[image]

[image]

Thanks!  Jon

EverythingSmartHome, 2024-02-26:
Yes the second image is correct with the coordinate system

kloppy, 2024-04-14:
I have posted a modified package in #39 which includes rotated zones like in the first picture @jonwork88 posted, as I have also put my EPLs into corners. You could give it a try.

neoKushan, 2024-06-25:
Hi,

I'm really struggling to understand how zones work and the coordinates behind them. I was hoping the EPL documentation would be updated to explain them but so far nothing.

Does anyone have a simple _written_ guide on how to set up and define zones?

poker86, 2024-06-30:
Hello,
same here,  I guess I'm just not completely understanding this.  I feel like based on the plot map that I should have zone 1 and zone 2 with active occupancy. but I am not. 

Map:
[image]
Sensors:
[image]

EverythingSmartHome, 2024-06-30:
What is the actual zone settings here as they aren't shown? 

Do you have the start and end X values the right way around with regards to negative and positive numbers?

poker86, 2024-06-30:
Controls
[image]

EverythingSmartHome, 2024-06-30:
Try swapping the start X and end X values around as mentioned

kloppy, 2024-06-30:
@poker86 how did you setup that neat map showing zone and target?

poker86, 2024-06-30:
it was the Pull Resquest post by @obbers

https://github.com/EverythingSmartHome/everything-presence-lite/pull/57

poker86, 2024-06-30:
@EverythingSmartHome, and thank you for helping me with zones. that indeed worked. and also making a great product

bikeymouse, 2024-07-14:
> @poker86 how did you setup that neat map showing zone and target?

I have also used this map representation, but found out that [it has mirrored the x-axis](https://github.com/EverythingSmartHome/everything-presence-lite/pull/57#issuecomment-2227444755), so that could explain your issues.

neoKushan, 2024-08-01:
> Hi,
> 
> I'm really struggling to understand how zones work and the coordinates behind them. I was hoping the EPL documentation would be updated to explain them but so far nothing.
> 
> Does anyone have a simple _written_ guide on how to set up and define zones?

So uhh...is there no written documentation anywhere on this feature? I still haven't got it to work and I'm struggling with where to even begin.

klaneos, 2024-08-29:
I have this same zone problem with the latest EPL. Will keep an eye on this issue in case there are any more developments on the fix. Preferably EPL update fix

EverythingSmartHome, 2024-08-29:
> So uhh...is there no written documentation anywhere on this feature? I still haven't got it to work and I'm struggling with where to even begin.

Sorry didn't see this, can you share the settings you have and where you are running into issues?

EverythingSmartHome, 2024-08-29:
> I have this same zone problem with the latest EPL. Will keep an eye on this issue in case there are any more developments on the fix. Preferably EPL update fix

What's the issue? Can you share the zone configs?

neoKushan, 2024-09-13:
> Sorry didn't see this, can you share the settings you have and where you are running into issues?

Hi, thanks for the response. I was hoping for some documentation on how zones are meant to be set up, ideally with some examples because I'm failing at the first hurdle.

I'll explain what I am trying to achieve, because I think (hope) that what I am trying to do is actually quite straightforward but I'm not clear at all how the Zone X/Y Begin/End values are meant to be applied.

Here's a diagram of the room I am trying to zone:

[image]

This is taken from the floor plans so the outer dimensions are relatively accurate. The Everything Presence Lite is mounted high up at the top of the image, it's facing down into the room at about a 45 degree angle.

The problem I am trying to solve is that when sat at my desk, I'm slightly too warm but when sat on the sofa I am slightly too cold - thanks to the aircon being positioned in the corner as shown. All I want to do is adjust the temperature by a degree if someone is on the sofa. Or in other words, detect if someone is in the Red zone (Call that Zone 1) or the yellow Zone (Call that zone 2) or both. The automation part I can handle easily enough, but as for which values I have to use to achieve this, I am not even sure where to begin as there's no way that I can tell to "visualise" what the sensor is seeing.

I'm aware that the sensor is going to see in a sort of "arc" pattern but I'm not entirely clear on the size of that arc, or where the zone coordinates fall in the arc.

The only setting for the EPL I'm relatively certain of is "Max Distance", which I have set to 400cm as that's the length of the room, but where I go next I am not sure.

I hope the above makes sense?

EverythingSmartHome, 2024-10-08:
Please see the new docs for the zone and the new visual zone configurator too which I believe will help with this: https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/creating-zones.html

closed 2024-10-08 by EverythingSmartHome