EverythingSmartHome/everything-presence-lite #74: Flashing the EPL with BLE functionality

jeleniain, 2024-02-19:
Hi all,
I am currently trying to figure out how to enable the Bluetooth/ ESPresense functionality on my EPL sensors.
In order to add the to my ESPHome container I had to flash them directly through my own ESPHome instance. For that I used the following yaml code:
```
substitutions:
  name: everything-presence-lite
  friendly_name: Everything Presence Lite
packages:
  Everything_Smart_Technology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```
My question is, what would I need to change to add the BLE functionality?
If I look at the package 'https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/everything-presence-lite-ha.yaml' it seems to already contain the Bluetooth configurations:

```
packages:
  device_base: !include common/everything-presence-lite-base.yaml
  ld2450_base: !include common/ld2450-base.yaml
  bluetooth_base: !include common/bluetooth-base.yaml
```
But than again I do not see ESPresense to be auto discovered in HA, any idea what I am doing wrong/ my above yaml needs to be adjusted?
Any help is very appreciated!

EverythingSmartHome, 2024-02-19:
Hello,

Did you try just using the pre-built option in the user guide instead? Then you can adopt it in ESPHome after and it's all completed for you without modifications

jeleniain, 2024-02-19:
Hi @EverythingSmartHome :) 
sadly that wont work in my case. I am running ESPHome in a separate docker and the only way (as far as I can tell) to list any ESP board in there is to flash it at least once directly from that ESPHome instance. 
If I use the pre-build/ preflashed  option, HA will discover the device but I wont be able to access the device in my ESPHome container and there is no adopt option in that case. This is not unique to the EPL but any board that I added to HA via ESPHome.

EverythingSmartHome, 2024-02-19:
It's totally possible to get it working in Docker AFAIK, you just have to make sure mDNS is working properly on your network and passes through to your containers properly. But that's a different issue.

Regardless the config you have is correct, what are the symptoms your seeing of Bluetooth proxy not working? Can you attach the ESPHome diagnostics file from Settings > Devices > ESPHome > Everything Presence Lite XXXXX > 3 dots (top left) > ESPHome diagnostics

jeleniain, 2024-02-19:
@EverythingSmartHome thanks for looking into this.
Regarding the symptoms, I thought that as soon as a ESPresense device is detected it will be shown in HA and can than be configured but that not happening in my case. 
Following my my diagnostic that I got from the HA ESPHome integration, hope its the right thing. I redacted a few lines in the beginning: 
[config_entry-esphome-d70544cc31396324fb01ade5d54c00b7.txt](https://github.com/EverythingSmartHome/everything-presence-lite/files/14334408/config_entry-esphome-d70544cc31396324fb01ade5d54c00b7.txt)

EverythingSmartHome, 2024-02-19:
That's not what this firmware enables, this firmware enables bluetooth proxy: https://esphome.io/components/bluetooth_proxy.html

Bluetooth Proxy extends the range of Home Assistant's bluetooth integration. It doesn't track ESPresence devices (I'm not sure I understand what you mean here because ESPresence is the tracker and so is ESPHome, do you mean you want to track a bluetooth device like a watch?). Regardless you would need the bluetooth tracker integration for that, which cannot be enabled by default in the firmware because you need to provide unique details: https://esphome.io/components/esp32_ble_tracker.html

I can confirm from the log you provided that bluetooth proxy is enabled correctly.

In short, the bluetooth proxy option is enabled correctly on your device and I think you perhaps misunderstood what it's used for?

Hope that helps

jeleniain, 2024-02-19:
Awesome thanks! I think I misunderstood it indeed.
I would like to to be able to measure the approximate distance from a e.g. my phone (with bluetooth beacon enabled in HA companion) to the closest EPL. I will read up more on it and see if I can figure it out.
Thanks a bunch for your help!

closed 2024-02-19 by EverythingSmartHome

TripelJ, 2024-02-19:
Hi I also wanted to give you some input as well. What you are trying to achieve is unfortunately not possible atm. Many wish that ESPrescense will be a part of ESPhome so you can benefit from both the proxy and tracking presence. For now you have to use one device for Bluetooth proxy and one with ESPrescense firmware. It cannot be both. 🙂 perhaps in the future.

jeleniain, 2024-02-19:
@TripelJ thank you very much! I did not know and assumed it can do both. Fingers crossed it can do it in the future. Appreciate your input, you saved my quite a headache :)