EverythingSmartHome/everything-presence-lite #76: Firmware update?

auanasgheps, 2024-03-07:
Hi,
This is not a proper issue but there's no "discussion" section, so I'm posting it here

This is my Everything Presence Lite:
[image]

On GitHub I see a 1.0.2 release. Am I supposed to see this update in HA or do it manually?

EverythingSmartHome, 2024-03-07:
No unfortunately releases from GitHub won't show in Home Assistant. The only update you will see is an ESPHome update whichever will show in Home Assistant, but this is notifying you of a new ESPHome version, not necessarily a new Lite versison

auanasgheps, 2024-03-07:
So "Hardware" version is in fact the EP Lite version?

I have to update manually connecting the sensor to a PC?

auanasgheps, 2024-03-07:
Thank you, it worked! I was able to update it, although now shows 1.0.3 which I can't see anywhere!

EverythingSmartHome, 2024-03-07:
Nice, glad it worked!

closed 2024-03-07 by EverythingSmartHome