EverythingSmartHome/everything-presence-lite #82: SCD40 CO2 sensor communication fails

danielcecil, 2024-05-28:
Hi, I cannot seem to get the SCD40 CO2 sensor to work by adding the scd4x component to ESPHome.

https://github.com/EverythingSmartHome/everything-presence-lite/compare/main...danielcecil:everything-presence-lite:main

I tried adding a separate i2c bus to the bh1750 as mentioned by @kloppy in a previous issue but this didn't seem to have any effect. Since the SCD40 uses a different address I'd prefer to have them both on the same bus anyway.

Sensor is connected to the GPIO pins correctly.

Log:
```
[09:54:29][C][scd4x:094]: scd4x:
[09:54:29][C][scd4x:095]:   Address: 0x62
[09:54:29][W][scd4x:099]: Communication failed! Is the sensor connected?
[09:54:29][C][scd4x:112]:   Automatic self calibration: ON
[09:54:29][C][scd4x:121]:   Ambient pressure compensation disabled
[09:54:29][C][scd4x:122]:   Altitude compensation: 0m
[09:54:29][C][scd4x:127]:   Measurement mode: periodic (5s)
[09:54:29][C][scd4x:139]:   Temperature offset: 4.00 °C
[09:54:29][C][scd4x:140]:   Update Interval: 60.0s
[09:54:29][C][scd4x:141]:   CO2 'CO2'
[09:54:29][C][scd4x:141]:     Device Class: 'carbon_dioxide'
[09:54:29][C][scd4x:141]:     State Class: 'measurement'
[09:54:29][C][scd4x:141]:     Unit of Measurement: 'ppm'
[09:54:29][C][scd4x:141]:     Accuracy Decimals: 0
[09:54:29][C][scd4x:141]:     Icon: 'mdi:molecule-co2'
[09:54:29][E][component:082]:   Component scd4x.sensor is marked FAILED
```

danielcecil, 2024-05-28:
Doh! I should have tinkered more.

FYI just add this to the bottom of the ESPHome config file. Thanks @kloppy 

```
# override the i2c definition in everything-presence-lite-ha.yaml
# to include an id for the bus, also add a second i2c bus which 
# does not break the bh1750
i2c:
  - id: bus_epl
    sda: 21
    scl: 22
    scan: true
  - id: bus_custom
    sda: 26
    scl: 27
    scan: true

# extend the definition for the bh1750 illuminance sensor 
# to reference the i2c bus by id, also add a bme280 
# environment sensor
sensor:
  - id: !extend illuminance_sensor
    i2c_id: bus_epl
  - id: co2_sensor
    i2c_id: bus_custom
    platform: scd4x
    address: 0x62
    update_interval: 30s
    co2:
      name: "CO2"
```

closed 2024-05-28 by danielcecil

EverythingSmartHome, 2024-05-28:
Yes that is the correct way for now, in the process of adding it to the docs and hopefully shorten the code a little bit too!