EverythingSmartHome/everything-presence-lite #90: Failing to boot

StayPirate, 2024-06-20:
In HA, if I flash the build coming from the following template, then the device goes in boot loop and doesn't work anymore.

```
substitutions:
  name: everything-presence-lite-285334
  friendly_name: mmWave Bagno in Camera
packages:
  EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha.yaml@main
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: ***********************************************
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
```

If I instead try to build/flash the following template, then the device boot fine.

```
esphome:
  name: test
  friendly_name: test

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "Ocdc7S3RWj0P/UcYCCXwExJz2yZw1aS8uj+3kZMnUcQ="

ota:
  - platform: esphome
    password: "fe295e52c1235cf0aa567c53b7ed947a"

wifi:
  ssid: "test"
  password: "testtesttest"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "testpassword"

captive_portal:
```

This is happening to me since I've updated to ESPHome version 2024.5.5. Is that just me? I have not changed anything on my configuration, I just updated the firmware when HA told me that there was a pending update for my device.

EverythingSmartHome, 2024-06-20:
Can you try keeping the original and pressing "clean build files" on the config before installing? Make sure to do it over USB initially.

Also you can't really compare those two YAMLs as the second one has almost no components loaded

StayPirate, 2024-06-20:
done, but same result. Here the logs:

```
[13:03:22]ets Jul 29 2019 12:21:46
[13:03:22]
[13:03:22]rst:0x1 (POWERON_RESET),boot:0x13 (Sets Jul 29 2019 12:21:46
[13:03:22]
[13:03:22]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:03:22]configsip: 0, SPIWP:0xee
[13:03:22]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:03:22]mode:DIO, clock div:2
[13:03:22]load:0x3fff0030,len:1184
[13:03:22]load:0x40078000,len:13132
[13:03:22]load:0x40080400,len:3036
[13:03:22]entry 0x400805e4
[13:03:23]ets Jul 29 2019 12:21:46
[13:03:23]
[13:03:23]rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:03:23]configsip: 0, SPIWP:0xee
[13:03:23]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:03:23]mode:DIO, clock div:2
[13:03:23]load:0x3fff0030,len:1184
[13:03:23]load:0x40078000,len:13132
[13:03:23]load:0x40080400,len:3036
[13:03:23]entry 0x400805e4
[13:03:24]ets Jul 29 2019 12:21:46
```

EverythingSmartHome, 2024-06-20:
Try flashing from the user guide, this seems like an issue with your local build possibly

StayPirate, 2024-06-20:
done, same issue. Here's a screenshot of the console after flashing from the user guide:

[image]

StayPirate, 2024-06-20:
I tried again with the No-Bluetooth firmware and now it worked! So the problem should be related to the Bluetooth component? BTW I need the Bluetooth firmware because I have some rules that are triggered by Bluetooth devices (toothbrush, scale). Are you able to reproduce it?

EverythingSmartHome, 2024-06-20:
Just flashed the bluetooth version myself from the user guide and no issues. It doesn't look like your unit has flashed properly as it seems to have some checksum issues relating to the partition structure. Here is the log from the one I just did:

[image]

Can you perhaps try flashing again now that you have a working firmware on there?

StayPirate, 2024-06-20:
> Can you perhaps try flashing again now that you have a working firmware on there?

Good point, I tried but I faced the same issue. I then re-flashed back to _no-ble_ firmware and it started booting fine again.

EverythingSmartHome, 2024-07-10:
Does this work now on ESPHome 2024.6?

closed 2024-10-08 by EverythingSmartHome