EverythingSmartHome/everything-presence-lite #91: Trying to install and I get this error:

Starf0x, 2024-06-20:
`INFO ESPHome 2024.6.1
INFO Reading configuration /config/esphome/everything-presence-lite-2230e8.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-lite-2230e8 (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- Improv @ 1.2.3
Linking .pioenvs/everything-presence-lite-2230e8/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/everything-presence-lite-2230e8/src/esphome/components/wifi/wifi_component.o:(.literal._ZN7esphome4wifi13WiFiComponent23is_esp32_improv_active_Ev+0x0): undefined reference to `esphome::esp32_improv::global_improv_component'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/everything-presence-lite-2230e8/src/main.o:(.literal._Z5setupv+0x3f8): undefined reference to `vtable for esphome::esp32_ble_server::BLEServer'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/everything-presence-lite-2230e8/src/main.o:(.literal._Z5setupv+0x3fc): undefined reference to `vtable for esphome::esp32_ble_server::BLEServer'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/everything-presence-lite-2230e8/src/main.o:(.literal._Z5setupv+0x400): undefined reference to `vtable for esphome::esp32_ble_server::BLEServer'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/everything-presence-lite-2230e8/src/main.o:(.literal._Z5setupv+0x49c): undefined reference to `esphome::esp32_improv::ESP32ImprovComponent::ESP32ImprovComponent()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/everything-presence-lite-2230e8/src/main.o: in function `setup()':
/data/packages/0c1ca49c/common/ld2450-base.yaml:52: undefined reference to `esphome::esp32_improv::ESP32ImprovComponent::ESP32ImprovComponent()'
collect2: error: ld returned 1 exit status
*** [.pioenvs/everything-presence-lite-2230e8/firmware.elf] Error 1
========================= [FAILED] Took 141.56 seconds =========================`

Any idea what this could be?

EverythingSmartHome, 2024-06-20:
Press the Clean build files button on the config then try again

Starf0x, 2024-06-21:
I did that, and also I connect the EPL to my computer.
Thank you, that was the solution.

btw, I saw the new EPL video, with new case. awesome improvements.