EverythingSmartHome/everything-presence-lite #92: Pets

Flat-Badger-1971, 2024-06-25:
First off, so glad I got one of these before they sold out again - sooo much better than another one I tried.

I suspect I already know the answer to this, but is there anyway to tune this thing to ignore pets (specifically two mental kittens)?

I've set up my automations to ignore when there is more than 1 target, but not sure how quickly two targets will register. Also doesn't help if only one of them is detected.

sfdye, 2024-07-15:
What worked for me (5 month puppy) is to adjust the angel of the EPL so that it does not detect objects closer to the floor.

EverythingSmartHome, 2024-10-08:
Yes unfortunately there isn't a direct way to ignore pets, but as mentioned above you can try to play around with the angles to help

closed 2024-10-08 by EverythingSmartHome