EverythingSmartHome/everything-presence-lite #94: Unable to update firmware and configure wifi

keithellis74, 2024-06-30:
Following the instructions, using Google Chrome on Linux Mint and MacOS I do not get a "Connect" Button. There is no link for a manual install so I am unable to continue to setup my device.

EverythingSmartHome, 2024-06-30:
Can you show a screenshot of what you are seeing?

keithellis74, 2024-06-30:
Screen shot attached from mobile, but same on all devices. I’ve not seen any setup videos and have never setup ESP devices like this before, I didn’t realise the HA logo was a button. Was not obvious to me.Keith Ellis

EverythingSmartHome, 2024-07-01:
There is no screenshot attached.

Yes you have to select which platform to use, it even says on the webpage to "select platform". 

So not making things hard, you just have to follow on screen instructions.

Glad you figured it out