EverythingSmartHome/everything-presence-lite #95: EP Lite shortly reports unoccupied always after occupancy off delay

thexperiments, 2024-06-30:
After the occupancy delay the sensor shortly reports unoccupied and directly goes back to occupied in <1s

[image]

Ignore the short times it shows occupied for a longer period, might rise another ticket for that. I'm sitting 1m away 50cm off axis from the sensor at upper body hight and it looses my presence even with 40s occupancy timeout.

thexperiments, 2024-06-30:
I just checked code and see you are using the delayed off from esp home also checked code there and looks pretty straight forward.

Thus I checked again and now seem to have longer detection times >5-10 mins so it doesn't seem to be the case that it resets every time after the timeout. 

Even with a flaky detection it really is wired that the interruptions 99% of the time are <1s before it detects again.
It would be  a too big coincidence if for all these detections it internally detected nothing for 40s and then it in 90+% of cases it detects occupancy again in <1s

EverythingSmartHome, 2024-07-08:
Sorry for the delay - do you know if its doing this because it thinks the target has moved out of the zone or if the entire sensor (check the non zone occupancy sensor) has no detection?

lure8225, 2024-07-08:
Looks like it was "just" detected outside of the zone
[image]
Example from yesterday when I was at my desk (Schreibtisch (Zone1)) 100% of the time

thexperiments, 2024-07-08:
Looks like it was "just" detected outside of the zone
[image]

Example from yesterday when I was at my desk (Schreibtisch (Zone1)) 100% of the time

EverythingSmartHome, 2024-07-08:
Helpful! Could it be that your sitting at the edge of a zone and occasionally detects you as outside if for a second? If you are near the edge, perhaps expand it a little?

thexperiments, 2024-07-11:
Yes, the zone is not huge but with the delay it would mean that it detected me outside of the zone for 40 seconds Just to detect me inside of the zone a few milliseconds after each time it switches over to not occupied

thexperiments, 2024-07-15:
[image]

wired, I (roughly) painted in the zone borders, it goes out of the zones occasionally but as I have set 40s of delay it should never go to unoccupied. A division here is about 15s so it would be out for one or both of the axis for two divisions which I can't se but its constantly toggling...

thexperiments, 2024-07-15:
the indication of number of targets within the zone seems to be fine
[image]
so it is some problem with the delay, you can see that the count is only 0 for very short durations

thexperiments, 2024-07-15:
It might be that this here interferes with the delayed off
https://github.com/EverythingSmartHome/everything-presence-lite/blob/main/common/ld2450-base.yaml#L772-L773
So if the state returns on until the timeout applies even after setting it to off before the time out will set it to off after 40s and back to on as soon as the next loop goes through as it now is not fenced anymore

thexperiments, 2024-07-15:
I think the fencing there might be not needed as there seems to be dedup code in the binary sensor already
https://github.com/esphome/esphome/blob/07b78fea760e78e5be5c0dbbf506f2fca490cff0/esphome/components/binary_sensor/binary_sensor.cpp#L15

thexperiments, 2024-07-15:
looks like that is the problem, left intentionally to check if turning off works properly
[image]
this is with the removed if check, will create a PR

thexperiments, 2024-07-15:
#117

EverythingSmartHome, 2024-07-19:
Thanks, I'm going to test it.

Just for clarities sake, can you let me know how to re-produce this error? I just want to be able to test before and after to make sure there is no unintended side effects or edge cases.

thexperiments, 2024-07-20:
Sure, should be easy to reproduce. 
1. set up a zone with a turn off delay (e.g. 30s)
2. move within the zone and observe presence and person count (occupied and 1)
3. Move out of the zone for less than the turn off delay (e.g. 5s) observe person count go to 0, presence will stay occupied (for now)
4. Re enter the zone and observe presence to toggle to off shortly and back immediately after. (Will happen exactly 30s after person count went to 0 for the zone)

Draghmar, 2024-07-24:
I think I had this too. Currently I'm testing EPL to get an idea of how it behaves in different configurations. And I did see situations when it was loosing count to 0 and occupancy even though I was in the zone and the main occupancy didn't go off.

EverythingSmartHome, 2024-08-03:
Thanks, https://github.com/EverythingSmartHome/everything-presence-lite/pull/122 merged!

closed 2024-08-03 by EverythingSmartHome