EverythingSmartHome/everything-presence-lite #96: "Zone 1 Occupancy" not working. How to properly determine Zone coordinates?

7bps, 2024-07-02:
I just got an EPL from the latest batch.

No matter what I do, **Zone 1 Occupancy** is always "Off" and **Zone 1 Target Count** is always "0".

Maybe I'm not entering the zone coordinates correctly?

The room is roughly 5m x 6m.

The EPL is wall mounted, pretty much right in the middle of one of the 6m sides.

I calculated the following values by walking around the room looking at the sensor values on my phone. Do these coordinates make sense for this room?

- **Zone 1 Begin X**: 2200.0 mm
- **Zone 1 Begin Y**: 300.0 mm
- **Zone 1 End X**: -2500.0 mm
- **Zone 1 End Y**: 4200.0 mm

Or is there anything else that would prevent **Zone 1 Occupancy** and **Zone 1 Target Count** from working?

[image]

7bps, 2024-07-02:
I solved this by making **Begin X** a negative number and **End X** a positive number. There's some logic to the numbers being lower to the left and higher to the right from the perspective of the EPL. However, the readings it gives for your X position sensor are the opposite of this.

closed 2024-07-02 by 7bps

EverythingSmartHome, 2024-07-02:
Hello, yes that is correct that lower numbers are the starting values. There is certainly more we can add to the documentation however to make this clearer.

I didn't quite follow your last sentence:

> However, the readings it gives for your X position sensor are the opposite of this.

Could you elaborate?

7bps, 2024-07-03:
Let's see if I can articulate this properly. I think it depends on which side of the sensor is considered "left". Is it the left from the sensor's perspective, or the left of a person facing the sensor? And I guess it kind of doesn't matter if the main rule is that the negative number is always Begin X. I might have been overthinking it.

So the first time I calculated the zone coordinates, I looked at the room from the perspective of the sensor.

I stood in the room to to the sensor's left. It gave me a positive X value for Target 1 X, which I recorded as the Begin X value for the zone.

I walked to the other side of the room over to the sensor's right. It gave me a negative value for Target 1 X, which recorded as the End X value of the zone.

Zone 1 Begin X: 2200.0 mm
Zone 1 Begin Y: 300.0 mm
Zone 1 End X: -2500.0 mm
Zone 1 End Y: 4200.0 mm

But the sensor wouldn't recognize me being in the zone until I made my Begin X number negative and End X number positive.

Zone 1 Begin X: -2200.0 mm
Zone 1 Begin Y: 300.0 mm
Zone 1 End X: 2500.0 mm
Zone 1 End Y: 4200.0 mm