EverythingSmartHome/everything-presence-lite #97: Cannot setup new Everything Presence Light Sensor - no Connect button

hajar97, 2024-07-02:
Just bought and received a brand new Everything Presence Light sensor. Go to Chrome (and tried Firefox as well). And connect button simply doesn't appear. Tried restarting browser. Tried reconnecting sensor to different USB ports. Previously setup this sensor without a problem a few months back. ESPHome works perfectly and connects via USB to my ESPHome devices. But this new sensor refuses to make Connect button to appear in the firmware update and wifi setup guide. I definitely see red led inside it blinking, so it is working. I also see in Chrome it appears in the list of devices as soon as I connect it to USB (have to use another ESPHome project to be able to do that). But on the official EPL setup page there is no Connect button. 

I use latest home assistant and latest ESPHome version and my previous EPL is updated to the latest ESPHome firmware and working well. Can you help solve the issue? 

I guess I could go down the route of setting it up manually in ESPHome and then copy paste the configuration yaml from my other EPL device into this one. Please confirm if that's the way I should solve it. Though this isn't ideally the way it should be...

EverythingSmartHome, 2024-07-02:
Can you show a screenshot what you are seeing on screen?

hajar97, 2024-07-02:
[image]

here you go

EverythingSmartHome, 2024-07-02:
You have to click the big button for Home Assistant where it says select smart home platform.

hajar97, 2024-07-02:
Whoah. I did use this intaller few months back and all went well. This time it totally didn’t cross my mind that I had to press that. Gosh, after pressing it I could finally see the connect button. I guess I was confused by the red warning that said if you don’t see connect then something is not right. And since I didn’t see connect and the other big button didn’t look like a button at all I got totally lost. 

Sorry for false alarm. Though, I’d definitely recommend improving the guide to make it more obvious, less confusing. Thanks for the quick response. Feel free to close this.

closed 2024-07-02 by EverythingSmartHome