EverythingSmartHome/everything-presence-lite #105: Delay Occupancy and the size of movement

Draghmar, 2024-07-09:
Hi.
I just got Lite and hook it up to test it for the first time. So far it looks like it work as it should but I wonder - if there's a way to make the Occupancy be delayed? For similar reasons like we have Off delay. mmWave is very sensitive, as we all know, which means that anything can trigger it. I'd like to deal with some things by making sure that it's detecting presence for some defined time before setting Occupancy to detected.
As a alternative (or maybe a additional sensor?), is there a way to know how big is the thing that moved? I don't know as much about mmWave to answer this myself but I imagine that if it would be possible, one could set Occupancy to detected only if the size would go above some threshold.
Thanks!

EverythingSmartHome, 2024-07-13:
Hello, unfortunately the LD2450 sensor does not expose those settings in it's firmware and so there is no way to set off latency like you can with SEN0609/SEN0395 for example

closed 2024-07-13 by EverythingSmartHome

Draghmar, 2024-07-13:
Oh, But does that mean if I'd get one of those and replace it, I'd get those additional settings?

EverythingSmartHome, 2024-07-14:
You can adjust on latency yes, size of target isn't a sensor I've seen on any sensors on the market currently