EverythingSmartHome/everything-presence-lite #114: Lite keeps bootlooping

martijn-brant, 2024-07-12:
Just received my batch of Lite's, thanks for that. However, I'm having some issues with it getting stable into Hass. My Hass doesnt have bluetooth so I used the manual installation guide on the wiki. Used the website to load a non-BLE fw on the device and have it join to my wifi. It works, I can join it into Hass.

Later I tried adopting it into the ESPHome dashboard, causing it to reinstall the fw. Have been getting really weird results so I wiped the device again using the wiki page, starting fresh. However, when I reboot the device after flashing, it seems to go into bootloop. I was able to get the logs via USB:

`ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6652
ho 0 tail 12 room 4
load:0x40078000,len:15052
load:0x40080400,len:3836
entry 0x4008069c
I (29) boot: ESP-IDF 4.4.7 2nd stage bootloader
I (29) boot: compile time 23:03:48
I (29) boot: Multicore bootloader
I (32) boot: chip revision: v3.1
I (36) boot.esp32: SPI Speed      : 40MHz
I (41) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 4MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (73) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (81) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (88) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (96) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (103) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=32880h (206976) map
I (191) esp_image: segment 1: paddr=000428a8 vaddr=3ffb0000 size=03350h ( 13136) load
I (196) esp_image: segment 2: paddr=00045c00 vaddr=40080000 size=0a418h ( 42008) load
I (214) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=a41c0h (672192) map
I (458) esp_image: segment 4: paddr=000f41e8 vaddr=4008a418 size=0ab18h ( 43800) load
I (486) boot: Loaded app from partition at offset 0x10000
I (486) boot: Disabling RNG early entropy source...
I (498) cpu_start: Multicore app
I (498) cpu_start: Pro cpu up.
I (498) cpu_start: Starting app cpu, entry point is 0x40081f94
I (486) cpu_start: App cpu up.
I (516) cpu_start: Pro cpu start user code
I (516) cpu_start: cpu freq: 160000000
I (516) cpu_start: Application information:
I (521) cpu_start: Project name:     everything-presence-lite
I (527) cpu_start: App version:      2024.6.6
I (532) cpu_start: Compile time:     Jul 10 2024 23:03:04
I (538) cpu_start: ELF file SHA256:  2f27a2236b36d4f3...
I (544) cpu_start: ESP-IDF:          4.4.7
I (549) cpu_start: Min chip rev:     v0.0
I (554) cpu_start: Max chip rev:     v3.99 
I (558) cpu_start: Chip rev:         v3.1
I (563) heap_init: Initializing. RAM available for dynamic allocation:
I (570) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (576) heap_init: At 3FFB7C08 len 000283F8 (160 KiB): DRAM
I (583) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (589) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (595) heap_init: At 40094F30 len 0000B0D0 (44 KiB): IRAM
I (603) spi_flash: detected chip: gd
I (606) spi_flash: flash io: dio
W (610) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (625) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[I][logger:156]: Log initialized
[C][safe_mode:079]: There have been 9 suspected unsuccessful boot attempts
[D][esp32.preferences:114]: Saving 1 preferences to flash...
[D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[C][i2c.idf:017]: Setting up I2C bus...
[... 310 lines cut ...]
E (10873) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10873) task_wdt:  - loopTask (CPU 0)
E (10873) task_wdt: Tasks currently running:
E (10873) task_wdt: CPU 0: IDLE0
E (10873) task_wdt: CPU 1: IDLE1
E (10873) task_wdt: Aborting.

abort() was called at PC 0x400fd1e8 on core 0

Backtrace: 0x40082966:0x3ffb0860 0x400896e5:0x3ffb0880 0x4008f89a:0x3ffb08a0 0x400fd1e8:0x3ffb0910 0x400832e9:0x3ffb0930 0x40170c83:0x3ffbace0 0x400fd939:0x3ffbad00 0x4008aabc:0x3ffbad20

ELF file SHA256: 2f27a2236b36d4f3

Rebooting...`

The most important being:

```
E (10873) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10873) task_wdt:  - loopTask (CPU 0)
```

So again, when flashing fresh FW via the wiki, the sensor works and I can add it to Hass. When opening the logs via USB, it still works. The moment you reboot the device (via the reboot button in the UI), it goes into this bootloop.

Also having a lot of issues trying to get this natively into esphome dash on my hass install but I'm guessing this bootloop is more important and could be the cause of all this.

martijn-brant, 2024-07-12:
Another log. Flashed it via wiki, joined wifi, added to hass, seeing values in hass (sensor working), load up logs via usb via wiki page, press reboot .. get stuck in bootloop?

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/16197109/esp-web-tools-logs.txt)

EverythingSmartHome, 2024-07-12:
Thanks for the details, I'm not able to replicate this issue currently. Do you know if its happening on more than one unit?

martijn-brant, 2024-07-12:
Just tried another. My exact steps:

- Open new unit from sealed box
- Open https://everythingsmarthome.github.io/everything-presence-lite/updating.html
- HA --> LD2450 --> No Bluetooth (not that it matters) --> Connect
- Connect ESP with supplied cable (tried multiple ports)
- Select device in Edge Serial connection box
- Install Everything...
- (have to hold BOOT button on ESP unit for it to be succesfull
- Flash is succesful
- Join to Wifi, works.
- Add unit to HA with IP from DHCP logs
- Sensor works in HA
- Open Logs screen on Wiki page
- Press Restart Device button
- Device goes into bootloop of 10 ~ 15 sec

Exact same result

EverythingSmartHome, 2024-07-12:
Thanks for confirming! Does it only happen with the logs open out of interest?

martijn-brant, 2024-07-12:
When I just power the units, the LED sometimes blinks (which it only does during boot i'm guessing?) and it pops up and drops from wifi, HA can't find it.

martijn-brant, 2024-07-12:
I've also tried this ESPHome code rather than using the wiki, same results

```
substitutions:
  name: "ep-lite-f1b3e8"
  friendly_name: "ep-lite-f1b3e8"
  illuminance_update_interval: "2s"
  hidden_ssid: "false"
  log_level: VERBOSE

packages:
  # Git repo examples
  everything-presence-lite-ha:
    url: https://github.com/EverythingSmartHome/everything-presence-lite/
    ref: main # optional
    files: [everything-presence-lite-ha.yaml]
    refresh: 1d # optional

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

```

martijn-brant, 2024-07-12:
To add to this: ESPHome 2024.6.6

martijn-brant, 2024-07-12:
OK, so I think Wifi is part of the problem. When Googling into that exact error, I found some references to Wifi. I was connecting this to an isolated vlan, with no access to internet or LAN.. only to HA itself. When I supply my main wifi creds (without any restrictions), it seems to work just fine. This is very weird to me.

martijn-brant, 2024-07-13:
The reboot of the device seems to be related to wifi not being able to find clients/HA(?) and thus it throws a reboot it seems. Searching for <ba-add>idx:1 (ifx:0, 02:ec:da:17:b6:db), tid:0, ssn:0, winSize:64 and esphome seems to find a few GH issues with similar issues but no solution. I also tried an internet-connected VLAN wifi that can't talk to LAN but can talk to HA and internet and that seems to work as well. The non-working wifi network has identical settings and other esphome devices have no issues with this network at all.

martijn-brant, 2024-07-13:
Hmm I don't think it's wifi.. it seems to be firewall related. When I change my isolation rule for the "Network of Things" wifi network (aka no internet, no LAN, only access to HA) to Allow instead of Drop, the boot loop goes away and the sensor works just fine. However when I block it again, it goes into bootloop. I saw in the firewall logging it's trying to access 185.199.110.153 which resolves to GitHub Pages. Could it be trying to parse an update manifest and when it can't access that, it fails?

Just tested with this ESPHome template (basically taking the external URLs out if it) and now it works just fine even with the firewall blocking everything.

```
substitutions:
  name: "ep-lite-f1b3e8"
  friendly_name: "ep-lite-f1b3e8"
  illuminance_update_interval: "2s"
  hidden_ssid: "false"
  log_level: DEBUG

# packages:
#   # Git repo examples
#   everything-presence-lite-ha:
#     url: https://github.com/EverythingSmartHome/everything-presence-lite/
#     ref: main # optional
#     files: [everything-presence-lite-ha-no-ble.yaml]
#     refresh: 1d # optional

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  #power_save_mode: NONE
  #fast_connect: TRUE
  

esphome:
  name: ${name}
  comment: Everything Presence Lite
  friendly_name: ${friendly_name}
  name_add_mac_suffix: True
  project: 
    name: EverythingSmartTechnology.Everything Presence Lite
    version: "1.0.5"

  on_boot: 
    priority: -100
    then:
      - button.press: get_mmwave_firmware
      - binary_sensor.template.publish:
          id: zone1_occupancy
          state: false 
      - binary_sensor.template.publish:
          id: zone2_occupancy
          state: false 
      - binary_sensor.template.publish:
          id: zone3_occupancy
          state: false 
      - binary_sensor.template.publish:
          id: zone4_occupancy
          state: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

# Enable logging
logger:
  level: ${log_level}

# Enable Home Assistant API
api:

ota:
  - platform: esphome
    id: ota_esphome
  - platform: http_request
    id: ota_http_request
[... 878 lines cut ...]
              if ((id(target2_x).state >= id(zone4_begin_x).state && id(target2_x).state <= id(zone4_end_x).state) &&
                  (id(target2_y).state >= id(zone4_begin_y).state && id(target2_y).state <= id(zone4_end_y).state)) {
                  zone4_count++;
              }
          }
          if (id(target3_active).state == true ) {
              if ((id(target3_x).state >= id(zone4_begin_x).state && id(target3_x).state <= id(zone4_end_x).state) &&
                  (id(target3_y).state >= id(zone4_begin_y).state && id(target3_y).state <= id(zone4_end_y).state)) {
                  zone4_count++;
              }
          }
          if (zone4_count > 0) {
            if (id(zone4_occupancy).state != true)
              id(zone4_occupancy).publish_state(true);
            if (id(zone4_target_count).state != zone4_count)
              id(zone4_target_count).publish_state(zone4_count);
          } else {
            if (id(zone4_occupancy).state != false)
              id(zone4_occupancy).publish_state(false);
            if (id(zone4_target_count).state != 0)
              id(zone4_target_count).publish_state(0);
          }
```

Is this an ESPHome bug for update: or dashboard_import: components?

martijn-brant, 2024-07-13:
The dashboard_import sections seems the have no effect on anything. The update: section triggers the bootloops. I've been looking into the esphome docs on update: but can't find anything on debug logging for it or for it to fail gracefully.

EverythingSmartHome, 2024-07-13:
Thanks for the updates and investigation. I would suspect from what you describe it's possibly a bug in relation to the new update mechanism in ESPHome, the http_request component.

This would really need to be reported upstream to ESPHome as this isn't related to hardware or anything specific with custom EP1 code (from what you described)

MrMDavidson, 2024-07-16:
I've the same issue and have a similarly configured network. An interim option people can do, so they don't have to copy/paste the entire source YAML from @EverythingSmartHome into their own projects is to use the `!remove` functionality of ESPHome. 

1. Adopt the device as per normal
2. Edit the YAML of the adopted device. Add the following somewhere in the YAML as a top-level-item (ie. `update` has no spaces before it on the line)
```yaml
update:
  - platform: http_request
    id: !remove update_http_request
```
3. Do a `Clean Build Files`
4. Install - **Note:** If your device is bootlooping you'll probably have to install this over USB.

Disclaimer: I've not played around with the `http_request` update component so I'm not sure what impact this will have. I presume it'll still grab the latest on a rebuild but you'll lose update notifications of any upstream EverythingSmartHome changes.

vhsdream, 2024-07-16:
Confirming I also have the same issue, with an EPL flashed with the non-BLE firmware. For me it really only happened when I attempted to update. I did get a notification to update, attempted it, and then the bootloop occurred. Had to open the case and reinstall fresh while pressing the boot button as it was continued to bootloop via USB - the serial connection would appear and then disappear over and over.

EverythingSmartHome, 2024-07-16:
When you did the update, was that from your own ESPHome addon? 

Also, this issue is related to when internet is blocked, is internet blocked on your device?

vhsdream, 2024-07-17:
> When you did the update, was that from your own ESPHome addon?
> 
> Also, this issue is related to when internet is blocked, is internet blocked on your device?

Hi there,

Yes it was via the update notification:
[image]

It appears the update goes through, but then connection drops. I then am unable to even connect via USB unless I crack the case and hold the boot button.

The device is on a separate VLAN but is not restricted from the internet.

EverythingSmartHome, 2024-07-19:
Can you please check the logs over USB when this occurs? You should be able to do this without opening the case.

shresth45, 2024-07-19:
> Confirming I also have the same issue, with an EPL flashed with the non-BLE firmware. For me it really only happened when I attempted to update. I did get a notification to update, attempted it, and [...]

Exactly the same situation here

EverythingSmartHome, 2024-07-23:
This should now be resolved in https://github.com/EverythingSmartHome/everything-presence-lite/pull/122, please update to version 1.0.7. A fix is coming in ESPHome 2024.7.2 but this workaround does the same thing for now.

closed 2024-07-23 by EverythingSmartHome

grigi, 2024-07-24:
I am experiencing the same issue as described here, I installed your version 1.0.7 and it doesn't resolve it.
It still bootloops all the time.

I'll try martjin's suggested workaround.

dsander, 2024-07-30:
I just hit the same problem, even with ESPHome 2024.7.2 the device does not work without giving it internet access. This is very confusing to me, are there only a few people denying internet access to their ESPHome devices? I could not find anything in their issue tracker. 

Didn't spend too much time on the issue because it seems obvious, but maybe I am missing something https://github.com/esphome/issues/issues/6104

cptquad, 2024-08-21:
I just got a pair of these today, it worked ok out of the box but when I imported the config into esphome and flashed it the boot loop started.

I am also running an isolated network with internet access blocked.

martijn's workaround works for me too :)

mlauweri, 2024-08-22:
I just received a pair of these as well and I'm also suffering from the bootloop issue, in case I connect them to my NoT (Network of Things, i.e. only local access/no internet access) network .
The bootloop also persists on my IoT network
The device only works on my main network and my guest network.
Disabling my firewall rules for my NoT and IoT networks on my router don't seem to help, so I'm a bit puzzled, especially since the device is running the latest firmware: 1.0.8 (ESPHome 2024.7.3). 

Anybody has any clue how to get this device working on a NoT network?

martijn-brant, 2024-08-23:
@mlauweri Take a look at my post here to have a short code snippet that takes out the update check and dashboard referrence. This makes it 100% local, no issues: https://github.com/EverythingSmartHome/everything-presence-lite/issues/114#issuecomment-2226585094

mlauweri, 2024-08-23:
> @mlauweri Take a look at my post here to have a short code snippet that takes out the update check and dashboard referrence. This makes it 100% local, no issues: [#114 (comment)](https://github.com/EverythingSmartHome/everything-presence-lite/issues/114#issuecomment-2226585094)

Thanks Martijn, somehow I was expecting the new firmware to be fixing the issue. I can confirm that applying your template as a workaround fixed the issue.

wneessen, 2024-09-01:
I have occasional boot loops too. Contrary to the others I do not have a limited network for that device. I checked my DNS logs and the device is fetching the https://everythingsmarthome.github.io URL just fine every 5-10 minutes. 

The sensor worked fine all day yesterday but just 30 minutes ago it went into the same boot loop like described here: https://github.com/EverythingSmartHome/everything-presence-lite/issues/114#issue-2405974129

After like 15 minutes of looping it finally connected again and is providing data to HASS.

I received the devices two Friday. Only one seems to be having the issue so far, the 2nd device hasn't run into a boot loop yet.

wneessen, 2024-09-01:
Apparently tonight at around 1pm it was also unavailable for like 10 minutes. 
[image]

Nelson994, 2024-12-19:
I do have a similar issue. The first fresh install works in HA by following the wizard on the EPL update page. But the moment I unplug the device and plug it back in, it becomes unavailable. I've tried my sensor in a friend's house with the same steps, and there, it works fine on his setup (different router and installation method of HA). Checking the logs, it seems to go in a boot loop as well when it restarts. It only runs for a few hours if I leave it plugged in before it goes unavailable again. I would really like to know how to fix this!

MrMDavidson, 2024-12-19:
> I do have a similar issue. The first fresh install works in HA by following the wizard on the EPL update page. But the moment I unplug the device and plug it back in, it becomes unavailable. I've [...]

@Nelson994 I've been running it with the removed `http_request` component and it's been flawless. See https://github.com/EverythingSmartHome/everything-presence-lite/issues/114#issuecomment-2229991519

Nelson994, 2024-12-23:
Thanks, @MrMDavidson! I had trouble uploading a new config, so I went on troubleshooting. Thanks to what was earlier said here (that the EPL has trouble updating), I removed a DNS rule I had on my router. And that worked for me! I had my Raspberry set as the first DNS server (from a tutorial on fixing the external URL errors from inside my network). Now I removed that rule and set the first and secondary DNS servers to default and it works! My external linked works still as well, so apparently I didn't even need that.

SLG, 2024-12-30:
I've had the same issue as @Nelson994 . Somehow my DNS settings were preventing the EP-Lite to be connecting to the wifi. When I removed my local DNS server it started working.