EverythingSmartHome/everything-presence-lite #115: Zone detection based on Distance

Asherslab, 2024-07-13:
Hi, love the new lite sensor

Was wondering if the zones could be upgraded to include a minimum and maximum distance in mm as well? (Specifically referring to the Target X distance variable)

Thanks!

EverythingSmartHome, 2024-07-13:
Glad to hear it!

I'm not really sure I follow, since a zone defines a specific area which by nature is already a minimum and maximum distance?

Asherslab, 2024-07-13:
Sorry, I might be misunderstanding myself possibly, I'm referring to the three variables on a target, of:
Target 1 X
Target 1 Y
Target 1 Distance

From looking at distance, it doesn't appear to be an addition or subtraction of X and Y, so I presumed you were able to measure the sensor's distance to the target

From what I can tell, the Zones currently only take into account the X and Y positions, not the distance. But I could be misunderstanding (this is mostly understood by trial and error)

kostasw, 2024-07-31:
My 2 cents (well, my understanding):
**Target X and Y** refer to the coordinates of the subject/target on the 'plane' which starts at 0,0 (the sensor itself).
**Distance** is the straight-line distance between sensor and subject/arget.

scotttimmins, 2024-10-22:
Distance is calculated from the x & y coordinates (using pythagoras for those interested).

The zones are defined using min/max distances in the x and y directions.