EverythingSmartHome/everything-presence-lite #118: Switching off Bluetooth does not work

bikeymouse, 2024-07-16:
With firmware 1.05 a new configuration switch is available to enable/disable bluetooth on the EPL.

This however doesn't seem to work. BlueTooth remains enabled after pressing the button.
See logging below, which suggests it works. However I still see bluetooth messages after this.

Also when restarting the log in the ESP Home panel, it shows that the Bluetooth state is on, even if it seems off in the Home Assistant device page.

[```
23:03:34][D][switch:016]: 'mmWave Bluetooth' Turning OFF.
[23:03:34][D][switch:012]: 'mmWave Configuration Mode' Turning ON.
[23:03:34][D][switch:055]: 'mmWave Configuration Mode': Sending state ON
[23:03:34][D][switch:055]: 'mmWave Bluetooth': Sending state OFF
[23:03:36][D][sensor:094]: 'Target 1 X': Sending state -4.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 1 Y': Sending state -164.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 1 Resolution': Sending state 772.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 1 Distance': Sending state 164.04877 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 1 Angle': Sending state -181.39719 ° with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 2 X': Sending state -258.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 2 Y': Sending state -772.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 2 Speed': Sending state -163.03999 m/s with 2 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 2 Resolution': Sending state 16383.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 2 Distance': Sending state 813.97052 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 2 Angle': Sending state -198.47946 ° with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 3 X': Sending state -416.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 3 Speed': Sending state -4.16000 m/s with 2 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 3 Distance': Sending state 416.00000 mm with 0 decimals of accuracy
[23:03:36][D][sensor:094]: 'Target 3 Angle': Sending state 90.00000 ° with 0 decimals of accuracy
[23:03:36][W][component:237]: Component uart took a long time for an operation (108 ms).
[23:03:36][W][component:238]: Components should block for at most 30 ms.
[23:03:36][D][binary_sensor:036]: 'Target 1 Active': Sending state ON
[23:03:36][D][binary_sensor:036]: 'Target 2 Active': Sending state ON
[23:03:36][D][binary_sensor:036]: 'Target 3 Active': Sending state ON
[23:03:37][D][binary_sensor:036]: 'Zone 2 Occupancy': Sending state OFF
[23:03:38][D][button:010]: 'Reboot mmWave Sensor' Pressed.
[23:03:38][D][sensor:094]: 'Target 1 X': Sending state -2.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 1 Y': Sending state -163.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 1 Speed': Sending state -7.72000 m/s with 2 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 1 Resolution': Sending state 258.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 1 Distance': Sending state 163.01227 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 1 Angle': Sending state -180.70297 ° with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 2 X': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 2 Y': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 2 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 2 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 2 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 2 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 3 X': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 3 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 3 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:38][D][sensor:094]: 'Target 3 Angle': Sending state 0.00000 ° with 0 decimals of accuracy
[23:03:38][D][binary_sensor:036]: 'Zone 2 Occupancy': Sending state ON
[23:03:38][D][sensor:094]: 'Zone 2 Target Count': Sending state 2.00000  with 0 decimals of accuracy
[23:03:38][W][component:237]: Component uart took a long time for an operation (138 ms).
[23:03:38][W][component:238]: Components should block for at most 30 ms.
[23:03:38][D][binary_sensor:036]: 'Target 2 Active': Sending state OFF
[23:03:38][D][binary_sensor:036]: 'Target 3 Active': Sending state OFF
[23:03:39][D][sensor:094]: 'Target 1 X': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 1 Y': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 1 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 1 Resolution': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 1 Distance': Sending state 0.00000 mm with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 1 Angle': Sending state -90.00000 ° with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 2 Angle': Sending state -90.00000 ° with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Target 3 Angle': Sending state -90.00000 ° with 0 decimals of accuracy
[23:03:39][D][sensor:094]: 'Zone 2 Target Count': Sending state 1.00000  with 0 decimals of accuracy
[23:03:39][W][component:237]: Component uart took a long time for an operation (55 ms).
[23:03:39][W][component:238]: Components should block for at most 30 ms.
[23:03:39][D][binary_sensor:036]: 'Target 1 Active': Sending state OFF
[23:03:40][D][sensor:094]: 'Zone 2 Target Count': Sending state 0.00000  with 0 decimals of accuracy
[23:03:40][D][switch:016]: 'mmWave Configuration Mode' Turning OFF.
[23:03:40][D][switch:055]: 'mmWave Configuration Mode': Sending state OFF
[23:03:54][D][binary_sensor:036]: 'Occupancy': Sending state OFF
[23:03:55][D][binary_sensor:036]: 'Zone 2 Occupancy': Sending state OFF
```

EverythingSmartHome, 2024-07-16:
The switch is named "mmWave Bluetooth" meaning it disables or enables Bluetooth on the mmWave module itself which has its own Bluetooth for updating firmware on it directly. This does not control Bluetooth on the EPL

Draghmar, 2024-07-17:
> The switch is named "mmWave Bluetooth" meaning it disables or enables Bluetooth on the mmWave module itself which has its own Bluetooth for updating firmware on it directly. This does not control Bluetooth on the EPL

Sorry for hijacking - does it mean there's no need for it to be turned on once everything is set up? I assume, having additional radio could mean some hypothetical interference to show up, so if disabling doesn't have any downsides then I'd do that.

EverythingSmartHome, 2024-07-17:
Yes it's just if you want to use with the HLKRadarTool app

bikeymouse, 2024-07-17:
Ok, thanks for explaining!
Will there also be an option to switch off the Bluetooth of the ESP32 (besides not compiling this)?

EverythingSmartHome, 2024-07-17:
I don't believe you can currently, it still is on in some way always

closed 2024-07-17 by EverythingSmartHome

gthbusrr, 2024-09-22:
@EverythingSmartHome Sorry to hijack this thread but I have 2 EPLs and both of their `mmWave Bluetooth` toggle doesn't work. I can connect to the module via Bluetooth (w/ the HLKRadarTool App) without turning on their `mmWave Bluetooth`