EverythingSmartHome/everything-presence-lite #120: Presense sensor stop updating radar

shresth45, 2024-07-19:
All radar sensor data just stops populating altogether in HA. In the Esphome logs too I cannot see the sensor data peing updated.
Tried this on 2024.6.6, 2024.7.0 ESPhome. Works fine after a firmware flash, then stops after a few hours.

```
INFO ESPHome 2024.6.6
INFO Reading configuration /config/esphome/everything-presence-lite-dc3658.yaml...
INFO Starting log output from 192.168.31.27 using esphome API
INFO Successfully connected to everything-presence-lite-dc3658 @ 192.168.31.27 in 0.005s
INFO Successful handshake with everything-presence-lite-dc3658 @ 192.168.31.27 in 0.092s
[19:01:59][I][app:100]: ESPHome version 2024.6.6 compiled on Jul 19 2024, 08:01:23
[19:01:59][I][app:102]: Project EverythingSmartTechnology.Everything_Presence_Lite version 1.0.6
[19:01:59][C][status_led:065]: Status Led Light:
[19:01:59][C][status_led:066]:   Pin: GPIO14
[19:01:59][C][wifi:599]: WiFi:
[19:01:59][C][wifi:427]:   Local MAC: FC:B4:67:DC:36:58
[19:01:59][C][wifi:432]:   SSID: 'Ghosh_2'[redacted]
[19:01:59][C][wifi:435]:   IP Address: 192.168.31.27
[19:01:59][C][wifi:439]:   BSSID: 54:AF:97:81:CC:44[redacted]
[19:01:59][C][wifi:440]:   Hostname: 'everything-presence-lite-dc3658'
[19:01:59][C][wifi:442]:   Signal strength: -51 dB ▂▄▆█
[19:01:59][C][wifi:446]:   Channel: 6
[19:01:59][C][wifi:447]:   Subnet: 255.255.255.0
[19:01:59][C][wifi:448]:   Gateway: 192.168.31.1
[19:01:59][C][wifi:449]:   DNS1: 0.0.0.0
[19:01:59][C][wifi:450]:   DNS2: 0.0.0.0
[19:01:59][C][logger:185]: Logger:
[19:01:59][C][logger:186]:   Level: DEBUG
[19:01:59][C][logger:188]:   Log Baud Rate: 115200
[19:01:59][C][logger:189]:   Hardware UART: UART0
[19:01:59][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:01:59][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:01:59][C][i2c.idf:075]: I2C Bus:
[19:01:59][C][i2c.idf:076]:   SDA Pin: GPIO21
[19:01:59][C][i2c.idf:077]:   SCL Pin: GPIO22
[19:01:59][C][i2c.idf:078]:   Frequency: 50000 Hz
[19:01:59][C][i2c.idf:084]:   Recovery: bus successfully recovered
[19:01:59][I][i2c.idf:094]: Results from i2c bus scan:
[19:01:59][I][i2c.idf:100]: Found i2c device at address 0x23
[19:01:59][C][uart.idf:139]: UART Bus 1:
[19:01:59][C][uart.idf:140]:   TX Pin: GPIO17
[19:01:59][C][uart.idf:141]:   RX Pin: GPIO16
[19:01:59][C][uart.idf:143]:   RX Buffer Size: 256
[19:01:59][C][uart.idf:145]:   Baud Rate: 256000 baud
[19:01:59][C][uart.idf:146]:   Data Bits: 8
[19:01:59][C][uart.idf:147]:   Parity: NONE
[19:01:59][C][uart.idf:148]:   Stop bits: 1
[19:01:59][C][template.number:050]: Template Number 'Illuminance Offset'
[19:01:59][C][template.number:050]:   Icon: 'mdi:brightness-5'
[19:01:59][C][template.number:050]:   Unit of Measurement: 'lx'
[19:01:59][C][template.number:051]:   Optimistic: YES
[19:01:59][C][template.number:052]:   Update Interval: never
[19:01:59][C][template.number:050]: Template Number 'Occupancy Off Delay'
[19:01:59][C][template.number:050]:   Unit of Measurement: 's'
[19:01:59][C][template.number:051]:   Optimistic: YES
[19:01:59][C][template.number:052]:   Update Interval: 60.0s
[19:01:59][C][template.number:050]: Template Number 'Max Distance'
[19:01:59][C][template.number:050]:   Unit of Measurement: 'cm'
[19:01:59][C][template.number:051]:   Optimistic: YES
[19:01:59][C][template.number:052]:   Update Interval: 60.0s
[19:01:59][C][template.number:050]: Template Number 'Zone 1 Begin X'
[19:01:59][C][template.number:050]:   Unit of Measurement: 'mm'
[19:01:59][C][template.number:051]:   Optimistic: YES
[19:01:59][C][template.number:052]:   Update Interval: 60.0s
[19:01:59][C][template.number:050]: Template Number 'Zone 1 End X'
[19:01:59][C][template.number:050]:   Unit of Measurement: 'mm'
[19:01:59][C][template.number:051]:   Optimistic: YES
[19:01:59][C][template.number:052]:   Update Interval: 60.0s
[19:01:59][C][template.number:050]: Template Number 'Zone 1 Begin Y'
[19:01:59][C][template.number:050]:   Unit of Measurement: 'mm'
[19:01:59][C][template.number:051]:   Optimistic: YES
[... 254 lines cut ...]
[19:02:11][D][binary_sensor:036]: 'Zone 3 Occupancy': Sending state OFF
[19:02:13][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:13][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:15][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:15][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:17][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:17][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:19][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.8lx
[19:02:19][D][sensor:094]: 'Illuminance': Sending state 373.06573 lx with 1 decimals of accuracy
[19:02:21][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:21][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:23][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:23][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:25][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:25][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:27][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:27][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:29][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:29][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:31][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:31][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy
[19:02:33][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.9lx
[19:02:33][D][sensor:094]: 'Illuminance': Sending state 421.67990 lx with 1 decimals of accuracy

```

shresth45, 2024-07-19:
My network is typically firewalled off to prevent WAN access. I allowed all traffic temporarily to triage the issue. 
There seems to be no impact base don it. The issue keeps occuring with unrestricted access as well. And the only way to restore the radar sensor is to reflash the EPL

EverythingSmartHome, 2024-07-19:
Can you share the settings when this happens? Also, does things like the LED control work? Just no radar data?

shresth45, 2024-07-20:
Do you mean settings as shown in HA? The buttons for LED and Reboot work as expected. But everything other than the Illuminance value is not updating.
[image]

[image]

shresth45, 2024-07-20:
I see the only way to get this working again is to connect via USB and reflash the firmware. Over Wireless does not resolve the issue. And the issue reoccurs at least twice a day. No changes on the network side.
This started occuring only after I installed latest update on ESPHome 2024.07, but now rolling back to the previous version is not resolving it either

jsk-sct, 2024-07-23:
I had the same kind of problem (sensor stuck at minus Y-values for example and didn't recognize when I walked in front of it) but tried the solution in https://github.com/EverythingSmartHome/everything-presence-lite/issues/98#issuecomment-2209154036
It seems to be working the last 24h so maybe give it a try

MikeDembek, 2024-07-26:
I'm seeing very similar behavior following the 1.0.7 update - I'm using imperial units but the values seem to be identical to @shresth45's:
[image]

EverythingSmartHome, 2024-07-27:
No change in 1.0.7 to cause this but definitely a bug with those values, will look into it

shresth45, 2024-07-29:
> I had the same kind of problem (sensor stuck at minus Y-values for example and didn't recognize when I walked in front of it) but tried the solution in [#98 (comment)](https://github.com/EverythingSmartHome/everything-presence-lite/issues/98#issuecomment-2209154036) It seems to be working the last 24h so maybe give it a try

@jsk-sct Does this fix work long term? I went on the route to downgrade the entire EPL release to 1.0.3 using the manual yaml method (working stable for 3 days now)

jsk-sct, 2024-07-30:
> > I had the same kind of problem (sensor stuck at minus Y-values for example and didn't recognize when I walked in front of it) but tried the solution in [#98 (comment)](https://github.com/EverythingSmartHome/everything-presence-lite/issues/98#issuecomment-2209154036) It seems to be working the last 24h so maybe give it a try
> 
> @jsk-sct Does this fix work long term? I went on the route to downgrade the entire EPL release to 1.0.3 (before http_request was introduced) using the manual yaml method (working stable for 3 days now)

Yeah, it has been working for a week now and hasn't frozen on minus-values one time. I upgraded to 1.0.7 and it didn't change anything.
I have noticed that the EPL has become a little more sensitive since I upgraded the sensor to 2.0.4 but that could be because had to loosen it before I upgraded to 1.0.3.

tkadauke, 2024-08-01:
Any workaround? I have a whole lot of EP1s around the house, and one by one they seem to be falling into over into this mode where nothing updates.

I managed to get one back to work for a few hours by disassembling it and restarting it a few times. Not sure exactly what fixed it, but it's broken again after ~5 hours.

alexdw369, 2024-08-02:
I have a bunch as well, all with the same 1.0.7, and all but one was stuck with the exact values previously described.  I tried updating mmWave sensor firmware to 2.0.4 on my one working EP Lite and on a non-working...the only change is that now ALL of my devices are stuck with stale values.

EverythingSmartHome, 2024-08-03:
Would you mind verifying if 1.0.8 changed anything for you?

alexdw369, 2024-08-03:
> ID: <EverythingSmartHome/everything-presence-lite/issues/120/2266641939@
> github.com>

Thanks for the reply, unfortunately I won't be much help here ...  I left
town for a few days, so my access is going to be limited to what I can see
in the HA front end since I won't have physical access to the EPLs.

Before I left, the only thing I noticed was that after allowing HA to
update device firmware to 1.0.8 on two of the EPLs, neither would connect
to HA due to a mismatch in encryption keys or something. I can grab a
screenshot of the logs when I get to my destination and can hop on my
computer.

Also (and this is probably a "me" problem), I'm not super sure what the
workflow is supposed to look like when flashing firmware using the
everythingsmarthome web utility, and then connecting to HA.  Am I NOT
supposed to allow adoption through the ESPHome Addon?  Because when I do
that, it'll want to re-install a different configuration and the firmware
will get reverted back to 1.0.7.  Otherwise, I'm not sure how to obtain the
API encryption key that HA asks for when adding the new EPL as an
integration.

I'll grab a screenshot of the logs and get them to you as soon as I am
able.  Thanks again for working on this.

Alex

tkadauke, 2024-08-03:
I just updated to 1.0.8, and still have the same issue. I haven't updated the mmWave firmware yet since the HKLRadarTool app keeps crashing as soon as I want to do anything with it.

Any suggestions what else I could try?

shresth45, 2024-08-04:
Just updating to 1.0.8 does not fix it completely. 
Currently I see 2 cases where this issue is fixed (EPL is working at least?) 
- Downgrade to EPL 1.0.3 with stock HLKRadar firmware (stable for 5 days)
- Upgrade to latest trial firmware v2.04.23101915 using HLKRadar, upgrade EPL to 1.0.7 firmware (stable for 5 days)

> Thanks for the reply, unfortunately I won't be much help here ... I left town for a few days, so my access is going to be limited to what I can see in the HA front end since I won't have physical [...]

I faced this exact issue with EPL in earlier releases as well, so I avoid using the HA GUI altogether. Updating from the ESPHome Addon GUI (tab or webui port, however you access) works properly. In the addon GUI, so long your yaml is referencing the master branch it will pull in the latest EPL release

MikeDembek, 2024-08-05:
I bumped to 1.0.8 yesterday, then had to re-push ESPHome firmware before the EPL would connect to HA again, but so far I'm seeing it work as expected running HA 2024.7.4 and ESPHome 2024.7.3.

biodrone, 2024-08-07:
This has been happening to my office sensor for a week or so now (and not to my living room or bedroom EPL) but it's just started happening to a customer since I upgraded them to 1.0.8, which now means it's a problem I need to fix... Sensor screenshots are below, I seem to be the only person in this thread with all 3 targets exposed but 1 & 3 are stuck at those values. I also tried downgrading to 1.0.4 but the build fails, and anything older than that doesn't seem to resolve and just builds 1.0.8.

Oddly when I reboot or reset the mmWave sensor the values change and get stuck as per the second image, then upon rebooting the whole EPL they revert back to the first image.

- Downgraded to 1.0.6
- Factory reset mmWave using HA button
- Downgraded to 1.0.5
- Rebooted mmWave using HA button
- Factory reset mmWave using HA button

Initial: [image]
Post mmWave reboot/reset: 
[image]

I'm remote from the customer site so it's not simple to go and update the mmWave firmware physically. I know that's the only thing I haven't yet tried, is there a way to do this remotely?

biodrone, 2024-08-07:
Another bit of info I've just realised, all of the devices of mine that are having this issue are from the original batch of EPLs, I have 2 that are working fine and they're both from the newest batch... Not sure if that makes a difference but figured it was worth mentioning.

EverythingSmartHome, 2024-10-08:
I believe this should be resolved now by updating the EPL firmware, and also ensuring that the mmWave firmware is on 2.04 by using the HLKRadarTool app on Android or iOS. Please let me know if still an issue

closed 2024-10-08 by EverythingSmartHome