EverythingSmartHome/everything-presence-lite #123: What is Target X Resolution?

Draghmar, 2024-07-24:
I don't understand this sensor. It gets different values although I've seen it mostly being at 360mm.

detroitgoose, 2024-08-28:
@Draghmar I went searching all around for an explanation. Best I could come up with is that it seems to be a measure of how close two targets can be to each other before they are recognized as one single target.

I would appreciate corrections or clarification on this subject from a more knowledgeable source.

Draghmar, 2024-08-29:
Thanks! I thought it could be that as that makes sense. I thought that maybe Lewis Barclay, seller and designer of this device, would know. ;)

EverythingSmartHome, 2024-10-08:
Hello, sorry for the delay! Yes that is my understanding of it too, though I've very very rarely ever seen it change on this sensor so it's not really useful information. Sometimes you can configure the resolution at the sacrifice of distance but not on this sensor

closed 2024-10-08 by EverythingSmartHome