EverythingSmartHome/everything-presence-lite #124: Lack of install app

chaoscode, 2024-07-28:
Relying on ESPHome to setup this device if you do not have BLE on your home assistant is a bad user experience. The device should broadcast a ssid and you should have a web config app. I have been messing with this device for a while. I find it very short sided of you all to not have a basic wifi configuration app built into the device.  I need to know where I can send this back to and how I can get a refund.

EverythingSmartHome, 2024-07-29:
Not sure what you mean? There is zero need to have ESPHome installed to set it up, even if you don't have Bluetooth. The instructions are very clear about this.

Can you elaborate?

chaoscode, 2024-07-29:
So here is the link from the box.

https://everythingsmarthome.github.io/everything-presence-lite/

You click here for wifi connection directions
[image]

You are presented with two options.

1.  Automatic (BLE)
[image]
2. Manual (ESPHome)
[image]

Step 2 is the "Manual" configuration and here is a quote from your documentation.

"Here you can install the latest [ESPHome](https://esphome.io/) firmware on the Everything Presence Lite board for direct integration with [Home Assistant](https://home-assistant.io/)"
[image]

This is the only option provided to configure the wifi without BLE.

So yes, the directions are very clear that you need ESPHome to manually configure the devices for wifi connection. 

This is confusing to me that you would then tell me "There is zero need to have ESPHome installed to set it up, even if you don't have Bluetooth" when your documentation says very clearly that you do.

chaoscode, 2024-07-29:
Thank you for your help with this. Both my wife and I did not understand the documentation as intended. May I offer some feedback on your documentation. It's too busy. Simplify the interface. The link should just drop you into the config of the device with options. It isn't Stright forward what you need to do to configure the device.

closed 2024-07-29 by EverythingSmartHome

chaoscode, 2024-07-29:
Well, good luck to you.