EverythingSmartHome/everything-presence-lite #126: What could be the reason behind flat X/Y coordinates?

Draghmar, 2024-07-30:
I'm testing EPL in bathroom, where I also have FP1. I've noticed that there some flat values for some uknown reason:
[image]
This one is relatively short but I've seen longer one. And I can't pinpoint the culprit that could be behind it...Any suggestions?

Draghmar, 2024-08-04:
I have even better example: those are two different entries to the bathroom. After the first one, there was this "flat issue" and Target 1 remained On for the whole time, until the second time occurred and it finally changed to Off at leave.

[image]

V4ler1an, 2024-08-14:
Thank you for your comment on [my recent issue](https://github.com/EverythingSmartHome/everything-presence-lite/issues/136) in which you directed me to your history filter here. Perhaps read my latest comment on my issue. If you consider that my issue looks as though it's some sort of EPL hang. Might those flat lines be something similar?

EverythingSmartHome, 2024-08-14:
Do both of you have the Bluetooth proxy firmware installed?

Draghmar, 2024-08-14:
> Do both of you have the Bluetooth proxy firmware installed?

Well...I don't remember which one I choose when doing first update through the webpage. :) Is there a way to confirm that right now? All I can tell is that I have:
```
Firmware: 1.0.8 (ESPHome 2024.7.3)
Hardware: 1.0.8 
```
And I have `mmWave Bluetooth` toggle available in the configuration.

V4ler1an, 2024-08-14:
> Do both of you have the Bluetooth proxy firmware installed?

I definitely chose firmwares with it disabled for all my EPLs.

EverythingSmartHome, 2024-08-15:
Can you go to the device page in home assistant for the Lite, then click the three dots (top left) and download ESPHome diagnostics then attach here?

Draghmar, 2024-08-15:
Here you go:
[config_entry-esphome-01J2BGVHE97ZYKR0KNQ986TGVX.json](https://github.com/user-attachments/files/16623369/config_entry-esphome-01J2BGVHE97ZYKR0KNQ986TGVX.json)

V4ler1an, 2024-08-15:
Here is mine: 
[config_entry-esphome-01J22E41GN6DNEKZPNFP5KFGKP.json](https://github.com/user-attachments/files/16623397/config_entry-esphome-01J22E41GN6DNEKZPNFP5KFGKP.json)

Draghmar, 2024-09-25:
That issue is still present and I still have no idea what's causing it. Here is the very short period when it happened. You can see that target 1 distance was going down and then stayed for some time at the same value even if the target was actually physically gone. Maybe there's an issue with edge detection? EPL is hanging on the wall, near where entrance is, so someone leaving could make sudden disappearance near the sensor. It's not something that makes too much sense but it's only thing I could think about.
As a side note: I have also Aqara FP1 there (the `Occupancy` value) but I think it works on different frequency. 
 
[image]

[image]

EverythingSmartHome, 2024-10-08:
Can you try the latest version of the firmware (1.1.1) and see if this resolves? There was  a bug fix related to this which may resolve it

EverythingSmartHome, 2024-10-08:
What version are you on here?

Draghmar, 2024-10-08:
The latest I think (those two last screenshot was taken with that version):
```
Firmware: 1.1.1 (ESPHome 2024.9.2)
Hardware: 1.0.8 
```

Well, I don't agree. :) I said about cool down being low just to let you know that the 35s it needed was way more than what it should be (it was 9s at that moment I think). Also it doesn't matter in this case because the issue is not about getting Occupancy to stay on (which is what cool down helps with) but quite the opposite. Not to mention strange flat behaviour. PIRs actually has the similar thing built-in in 99% of devices I've seen. That's why I choose one from Philips, where cool downs are relatively low (MotionSensor03 is the one). ;)

EverythingSmartHome, 2024-10-08:
Cool just checking version.

Uh yeah, I'm of course aware that PIRs have cool downs ;) that's not what I was referring to. PIRs are pretty simple in that they see changes in heat, mmWave sensors use algorithms to target tracking which is a lot more complex. The cooldown on a mmWave comes after the mmWave judges that a target is no longer present based on it's internal algorithm, however they can sometimes "hang on" to a target a little longer, thus delaying the cool down period starting.

The target lingering for 25s (plus 9s of cooldown) isn't really an issue I don't think personally, but the flat coordinates may or may not be depending on if it's the mmWave sensor itself reporting those.

What's your use case for a very short cooldown period?

Draghmar, 2024-10-08:
Yeah, I know the basics. I'm trying to find the best sensor for quite some time. :D

I'm using it to turn on/off lights in the bathroom. Which ideally should happen at the moment of entering/leaving. Any delay is annoying. Especially when I can see light lingering after closing door (I have small apartment - I can see almost everything, everywhere ;)). With FP1 and Philips I was able to get down to about 11-13s after door was closed (Philips cool down). FP1 alone is not enough because both warm up and cool down is really huge (you can check it on my screens as Occupancy). But it's good to preventing lights going out when PIR wasn't able to detect micro movements.

With EPL, at first, I was able to get to 3-5s after closing the door and that was really amazing. But then some anomalies started to happen, like entering bathroom and closing door but not getting in the EPL range and things like that, which forced me to add some failovers in the automation. I have limited places where I can put EPL so it's not in the ideal spot - right beside door, "lookin" inside, when it would be much better looking at the door.

Basically I'm experimenting with settings all the time, both on device and in automations. Those flat lines are really weird for me though. I don't understand why sensor could be detecting something that isn't there. Is this simply some downside of mmWave itself? Or mmWave sensors? Would changing sensor type helped here?

scotttimmins, 2024-10-15:
I've been looking more into the "stuck sensor" issues and I've run into the LD2450 itself sending the values (latest version of the firmware for the mmwave sensor but my EPL firmware is a bit custom atm (hence I was able to get debug logs).

```
[22:27:00][D][uart:833]: p2_x: 1099, p2_y: 2600, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2822.729248, p2_angle: -22.913403
[22:27:00][D][sensor:094]: 'Target 2 X': Sending state 1099.00000 mm with 0 decimals of accuracy
[22:27:00][D][sensor:094]: 'Target 2 Y': Sending state 2600.00000 mm with 0 decimals of accuracy
[22:27:00][D][sensor:094]: 'Target 2 Speed': Sending state 0.00000 m/s with 2 decimals of accuracy
[22:27:00][D][sensor:094]: 'Target 2 Distance': Sending state 2822.72925 mm with 0 decimals of accuracy
[22:27:00][D][sensor:094]: 'Target 2 Angle': Sending state -22.91340 ° with 0 decimals of accuracy
[22:27:01][D][uart:833]: p2_x: 1164, p2_y: 2584, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.069824, p2_angle: -24.249866
[22:27:01][D][sensor:094]: 'Target 2 X': Sending state 1164.00000 mm with 0 decimals of accuracy
[22:27:01][D][sensor:094]: 'Target 2 Y': Sending state 2584.00000 mm with 0 decimals of accuracy
[22:27:01][D][sensor:094]: 'Target 2 Distance': Sending state 2834.06982 mm with 0 decimals of accuracy
[22:27:01][D][sensor:094]: 'Target 2 Angle': Sending state -24.24987 ° with 0 decimals of accuracy
[22:27:02][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:02][D][sensor:094]: 'Target 2 X': Sending state 1168.00000 mm with 0 decimals of accuracy
[22:27:02][D][sensor:094]: 'Target 2 Y': Sending state 2583.00000 mm with 0 decimals of accuracy
[22:27:02][D][sensor:094]: 'Target 2 Distance': Sending state 2834.80396 mm with 0 decimals of accuracy
[22:27:02][D][sensor:094]: 'Target 2 Angle': Sending state -24.33188 ° with 0 decimals of accuracy
[22:27:03][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:04][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:05][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:06][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:07][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:08][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:09][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:11][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:12][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:13][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:14][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:15][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:16][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:17][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:18][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
[22:27:19][D][uart:833]: p2_x: 1168, p2_y: 2583, p2_speed: 0.000000, p2_distance_resolution: 360, p2_distance: 2834.803955, p2_angle: -24.331882
```

I'm wondering if resetting the x & y values to 0 if they don't change for a while would be a reasonable way to resolve the symptom.

(Code I'm running is [here](https://github.com/scotttimmins/everything-presence-lite/blob/chore/prevent-stuck-sensors-v0.1/common/ld2450-base.yaml) for sanity checking, though it was running with a 100ms update time when I got the logs above. The code hasn't been battle tested so I don't recommend using it just yet)

scotttimmins, 2024-10-19:
I implemented the change I suggested above using the same readings 8 times in a row and did some testing .  Whilst I did see some improvements I still had stuck sensors (I'm running a bluetooth proxy so that may factor in).

The change I made did have a (potential) flaw in that if a sensor got stuck with a presence in the room it may report that there was no-one there.