EverythingSmartHome/everything-presence-lite #127: Firmware upgrade via HA front end results in blinking led, no boot

kostasw, 2024-07-31:
I encountered this issue twice now, across both my Lite devices. The only way to revive them is to connect via USB and flash them from scratch from the user guide.
The issue then is that you need to re-onboard them in HA, go through the device and entity renaming anf reprogramming. Or am I doing something wrong?

AlesAlitis, 2024-08-03:
> I encountered this issue twice now, across both my Lite devices. The only way to revive them is to connect via USB and flash them from scratch from the user guide. The issue then is that you need to [...]

I have the same issue. See https://github.com/EverythingSmartHome/everything-presence-lite/issues/128

m4v3r1ckNl, 2024-08-06:
Same issue. See https://github.com/EverythingSmartHome/everything-presence-lite/issues/128#issuecomment-2268688481

m4v3r1ckNl, 2024-08-16:
Resolved my issue by booting into macOS Monterey iso macOS Sonoma. The CH34xVSPDriver seems to be not working A-oke in Sonoma. Connected the EPL with USB on ESPHome (beta) page in HA, selected the "USB Serial (cu.wchusbserialfa410) - Paired" and reflashed the EPL with the previous .yaml file I downloaded previous.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Hope this may help others.

EverythingSmartHome, 2024-10-08:
Will close this in favour of https://github.com/EverythingSmartHome/everything-presence-lite/issues/128 which contains more details.

closed 2024-10-08 by EverythingSmartHome