EverythingSmartHome/everything-presence-lite #128: Notification on update in setting page after flashing most updated firmware

Motya-HA, 2024-08-03:
After updating all EPL sensors, in setting page I still can see "update available"
[image]

dodgydd, 2024-08-03:
Latest firmware has taken my EPL out of action. I can see it in ESPhome, but all the sensors are showing blank. Unable to load firmware onto it manually either, effectively it's bricked until I can figure this out.

EverythingSmartHome, 2024-08-03:
> After updating all EPL sensors, in setting page I still can see "update available"
> 
> ![Screenshot 2024-08-03 at 3 09 48 AM](https://private-user-images.githubusercontent.com/143926968/354809741-99d36805-b8a5-4c8f-977d-b90c1cd5fbe6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2ODM5NTYsIm5iZiI6MTcyMjY4MzY1NiwicGF0aCI6Ii8xNDM5MjY5NjgvMzU0ODA5NzQxLTk5ZDM2ODA1LWI4YTUtNGM4Zi05NzdkLWI5MGMxY2Q1ZmJlNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QxMTE0MTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMjgxNjFjZmE4N2U1ZWY1MDQ3ZDdmODU5ZjE1OTgyMDhmZmI3MzVjZTQzNjBmMGVmZDdmZTNkZTlkNTJlMTZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FA8yUxzy_Q-AT3gIML8NmNQSxzPLxFjfXt5_P570dOQ)
> 
> ![Screenshot 2024-08-03 at 3 07 56 AM](https://private-user-images.githubusercontent.com/143926968/354809624-22ee8a04-4b5b-457c-a1b1-28dd0cf9bd8d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI2ODM5NTYsIm5iZiI6MTcyMjY4MzY1NiwicGF0aCI6Ii8xNDM5MjY5NjgvMzU0ODA5NjI0LTIyZWU4YTA0LTRiNWItNDU3Yy1hMWIxLTI4ZGQwY2Y5YmQ4ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODAzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgwM1QxMTE0MTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00ODFlNTM1NDIyMjJhNTRjNjJjZDQwNzM5MDM0NTgzOTVkNjJhNGJjNzJkNzBiNGE1YjM5YjQxMGU0OTVjZmFmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.GWp_1FEl-jxw7JQ48N6PXsQVnpmUyANgcAZ9J8wbE54) 88dbe)
> 
> Further updates are not removing this note

Did you update from your own ESPHome addon or from the update message there?

EverythingSmartHome, 2024-08-03:
> Latest firmware has taken my EPL out of action. I can see it in ESPhome, but all the sensors are showing blank. Unable to load firmware onto it manually either, effectively it's bricked until I can figure this out.

This is not an issue with the update. It's an issue with applying the update directly from GitHub when you have already configured it in the ESPHome add-on. When you add it to the ESPHome addon, it automatically adds an encryption key. Home Assistant uses this encryption key to communicate with the Lite.

When you install the update directly from GitHub instead of via the ESPHome addon, it will remove the encryption key. So home assistant is effectively trying to communicate with it using a password, while the device has no password.

Either remove and re-add to Home Assistant without the password, or flash again from ESPHome to add the password back

dodgydd, 2024-08-03:
Thanks, fixed now. I did only use the ESP home add on to install the update, so I'm not sure what went wrong. In the end I deleted the device in ESP home which immediately showed up a new device with the ability to adopt it. I adopted it and some time later ESPhome reported an API key error with a dialogue to re-enter it. Thanks for the help by email.

AlesAlitis, 2024-08-03:
I get the notification (and the issue) _after_ installing the update via ESP home. I have to install the HA update then install via EPS home again, otherwise the notification doesn't go away and re-appears even after skipping them.

Motya-HA, 2024-08-03:
> Did you update from your own ESPHome addon or from the update message there?

I think I tried to do both, but after updating from update message things broke and I needed to flash sensors connecting them directly with USB C to computer and using flasher from your site. Now I was trying to updated from add-on it is not removing message.
In the process, 2 EFL sensors can't connect to WIFI, no matter what I do :-(
What sequence you woulds suggest?
Another observation, I use 2 sensors with SEN0395, and both are having the same issue - red LED in SEN0395 turns on (without me doing anything) and "occupancy" became "clear" even if I am next to the sensor. I can turn LED off by toggling   "mmWave LED" switch in dashboard. 

EPO sensors are not having any issues.

I am using many EFL sensors with SEN0395 and custom "ceiling case" mount and it is a cornerstone of my smart house automatons. 
Any suggestions will be highly appreciated

Motya-HA, 2024-08-03:
This is the process that worked for me (almost entire day, becuase I have many sensors):
1. Updated via HA
2. Adopted sensor (need to change a name, but not the same it was, can't be same) - takes time for flashing
3. Had a prompt in "notifications" to reconfigure sensors (need to re-enter API key from the "adopted" sensor
4. Delete in ESPHome old sensor (which suppose to be off-line)
5. Change the name to old one (if desire) - takes time for flashing

Red "LED on" for SEN0395 Sensor after any flashing or power cycle.

AlesAlitis, 2024-08-04:
That's my workaround:

1. Install ESP Home plugin update
2. Update all devices in the plugin
3. Some time later (usually next day) update my EPLs via the HA notification
4. Install each sensor again in ESP home. They appear offline, but they are connected to the WiFi and the install works fine
5. The sensors are back online in HA

This works for all but 1 of my EPLs. It's connected to the WiFi, but the OTA update fails

janchlebek, 2024-08-04:
In my case the behaviour is different: did the following:

- update ESP add-on
- update the device
- updated firmware (device disappeared as expected)
- reinstalled via ESPHome (device is back)
- illuminance is working but mmWave sensor does not work (no object detected, entities show some old values constantly)

Any ideas?

Motya-HA, 2024-08-04:
> * illuminance is working but mmWave sensor does not work (no object detected, entities show some old values constantly)
What sensor do you have? I am having issues with SEN0395. Illuminance comes from the separate sensor (on main board)  and has nothing to do with mmWave sensor itself

Motya-HA, 2024-08-04:
> > * illuminance is working but mmWave sensor does not work (no object detected, entities show some old values constantly)
> >   
What sensor do you have? I am having issues with SEN0395. Illuminance comes from the separate sensor (on main board)  and has nothing to do with mmWave sensor itself

janchlebek, 2024-08-04:
I do not know exactly - I did buy the lite bundle from EverythingSmartHome. How can I tell which one it is?

Motya-HA, 2024-08-04:
> How can I tell which one it is?
By default it comes with LD2410 sensor. I did not have any issue with it.

ndalbjergs, 2024-08-04:
Sooooo

Just to be sure.. I should never update via the promted message from HA in settings?
But only from the ESPhome, where I should select update all?

That is a pretty important message if true, because I have had a lot of issues with mine the last months, and mostly when selecting update via HA..
Afterwards I then choose install wireless in ESPhome, and they would work again.. But then a new update (which seems to come one to two times a week?) would fuck it up again..

Motya-HA, 2024-08-04:
> I should never update via the promted message from HA in settings?
>>

Updating from HA settings is the only thing that worked for me. Updates from ESPHome did not remove prompt whatsoever.

m4v3r1ckNl, 2024-08-05:
Same issues with the update with the add-on and/or with HA Settings updates. 

Reinstalling with the ESPHome webpage:

> Failed to initialize. Try resetting your device or holding the BOOT button while selecting your serial port until it starts preparing the installation.

kostasw, 2024-08-06:
> Same issues with the update with the add-on and/or with HA Settings updates.

same experience for me

biodrone, 2024-08-07:
> In my case the behaviour is different: did the following:
>
>     * update ESP add-on
>
>     * update the device
>
>     * updated firmware (device disappeared as expected)
>
>     * reinstalled via ESPHome [...]

@janchlebek Looks like you might be having an issue similar to a few of us over here: https://github.com/EverythingSmartHome/everything-presence-lite/issues/120

m4v3r1ckNl, 2024-08-16:
> Same issues with the update with the add-on and/or with HA Settings updates.
> 
> Reinstalling with the ESPHome webpage:
> 
> > Failed to initialize. Try resetting your device or holding the BOOT button while selecting your serial port until it starts preparing the installation.

Resolved my issue see

Motya-HA, 2024-08-19:
Interesting, today got notifications for firmware update for EPO - updated them via ESPHome. No problem whatsoever

EverythingSmartHome, 2024-10-08:
The GitHub update entity was disabled in a previous version to avoid users running into this problem. I will close for now but do let me know if its still an issue