EverythingSmartHome/everything-presence-lite #129: Everything Presence lite targets a object, but shows no occupancy

ndalbjergs, 2024-08-04:
I have a EP lite in a bathroom.

Sometimes when standing in front of it, it will show distance to target 1.. So it can see me..

But occupancy is set to no registration (or what it is called in English).

This mean the automation that should turn on the light (controlled by a shelly) does nothing..

How can a EP lite show a target, but says "no" in occupancy? 

[image]

ndalbjergs, 2024-08-07:
Now one of my other EPL did the same..

Occupancy is a "not registered", it there is a target (me)..

I fixed my cutting the power, and then on again, and then it worked again.

Is this a known bug, and/or should I change my light automation to look at "target" instead of "occupancy"? 
[image]

ndalbjergs, 2024-08-15:
No one with an idea here?

I have now tried to connect the two EPL to my windows machine, and update via. the guide with the firmware without bluetooth.

I works for some days, but then all of a sudden, the issue is there (it tracks me as a target and changes the value in accordance with my movement, but occupancy is listed as "not registered" in HA), and I need to cut the power to the EPL for 30 sec, and then it works again.

@EverythingSmartHome ?

Draghmar, 2024-08-16:
Not sure if it's directly related but I see something like that with zones and targets.  I can have target in zone 1 and yet occupancy being off. There was a moment when I was also considering using targets instead of occupancy but that is not reliable as well I'm afraid. At least for me it got some weird behaviour...but I'm still testing it.
At least I'm not forced to use EPL as sensor because I have others already working. ;)

EverythingSmartHome, 2024-08-16:
Do you notice any other surrounding events when that issue occurs? Such as other entities going unavailable?

ndalbjergs, 2024-08-19:
> Do you notice any other surrounding events when that issue occurs? Such as other entities going unavailable?

No, not really..

I have found out, that it fixes it self after a while.. But the quick fix is power off and on.

There is also problems sometimes with occupancy is stuck on "registered", and target 1 is showing the same distance no matter what..
And that has also happened on different EPL.. And the distance is gets stuck on is the same digits everytime..

Here a screen save of that situstion:
[image]

ndalbjergs, 2024-08-19:
And now one of the other EPL also got stuck with registered occupancy, and taget 1 with the exact same values as the other EPL I just shared a screen shot of.. As you can see it is not the same EPL, but same exact values. 

[image]

EverythingSmartHome, 2024-08-22:
Can you download the HLKRadarTool app on iOS or Android and update the firmware of the mmWave sensor to 2.04 please?

ndalbjergs, 2024-09-08:
> Can you download the HLKRadarTool app on iOS or Android and update the firmware of the mmWave sensor to 2.04 please?

Tried it now, but it is not helping the isue.. The issue comes back again and again.. Sometimes fixing it self, sometimes I have to cut power to the sensor, and then it works again..

EverythingSmartHome, 2024-09-08:
Can you show the HLKRadarTool firmware? The values and behaviour you are getting are a symptom of old firmware, but the HLKRadarTool app is very buggy and doesn't always actually update the firmware. Another user reported RadarTools works better sometimes: https://apps.apple.com/gb/app/radartools/id6502672452

ndalbjergs, 2024-09-08:
RedarTools seems to work, so will have a look the next week or so..

EverythingSmartHome, 2024-10-08:
Will close as assume resolved now

closed 2024-10-08 by EverythingSmartHome