EverythingSmartHome/everything-presence-lite #130: EPL with SEN0609 sensor - can't turn LED off

Motya-HA, 2024-08-06:
Hello

I installed SEN0609 DFRobot sensor in one of my EPL devices. 

I used the following YAML code:

EverythingSmartTechnology.Everything_Presence_Lite: github://everythingsmarthome/everything-presence-lite/everything-presence-lite-ha-sen0609.yaml@main

Everything look ok, but I have blue LED which is constantly "on" and yellow LED which is blinking all the time. I can't turn them off. "mmWave LED" entity does nothing (Blue LED  turns off for 3 sec and turns on again). Yellow LED continue to blink

Thanks

EverythingSmartHome, 2024-08-06:
The SEN0609 has 2 LEDs on it, one is controllable and one is is not. The SEN0609 that we sell has the non cotrollable LED removed, but the consumer version doesn't. You would need to desolder or cover this LED to turn it off unfortunately as it's tied directly to the output pin

closed 2024-08-06 by EverythingSmartHome

Motya-HA, 2024-08-06:
Thanks
Which one I should remove? Yellow blinking LED?
Blue one is on as well if presence is detected. I can't control blue LED. I want to disable that - how?

Thanks