EverythingSmartHome/everything-presence-lite #131: Install Failed - Internal error (Segmentation fault)

TheManoeuvre, 2024-08-08:
The device was previously set up on my HA instance but this was corrupt and is being rebuild from new.
I tried to install, this initially failed but I suspect this was due to insufficient RAM configured on my VM. Updated to 4GB and re-ran. I'm receiving the below in my logs  

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/everything-presence-lite-28510c.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing everything-presence-lite-28510c (board: esp32dev; framework: espidf; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Dependency Graph
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- Improv @ 1.2.3
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/bluetooth_proxy/bluetooth_connection.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_client/ble_client_base.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_client/ble_service.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_server/ble_2901.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_server/ble_2902.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_server/ble_characteristic.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_server/ble_descriptor.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_server/ble_server.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_server/ble_service.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esp32_improv/esp32_improv_component.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/esphome/ota/ota_esphome.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/http_request/http_request.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/http_request/http_request_arduino.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/http_request/http_request_idf.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/http_request/ota/ota_http_request.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/http_request/update/http_request_update.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/http_request/watchdog.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/i2c/i2c.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/i2c/i2c_bus_arduino.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/i2c/i2c_bus_esp_idf.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/improv_base/improv_base.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/improv_serial/improv_serial_component.o
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/json/json_util.o
/tmp/ccSAOLoB.s: Assembler messages:
/tmp/ccSAOLoB.s:326: Internal error (Segmentation fault).
Please report this bug.
Compiling .pioenvs/everything-presence-lite-28510c/src/esphome/components/light/addressable_light.o
*** [.pioenvs/everything-presence-lite-28510c/src/esphome/components/improv_base/improv_base.o] Error 1
========================= [FAILED] Took 34.72 seconds =========================
```

EverythingSmartHome, 2024-08-08:
Hello! This would be an ESPHome issue unfortunately as the Lite config compiles fine (I know your one isn't but it's specific to your setup) - I wouldn't know where to start with that error since its a compiler issue.

It looks like you should be able to press the "Clean build files" button then hit install to fix, but perhaps you already tried this

TheManoeuvre, 2024-08-09:
Yeah I got it working eventually - thanks for the response :)

closed 2024-08-09 by TheManoeuvre