EverythingSmartHome/everything-presence-lite #134: EPL sensors with SEN0395 or SEN0609 mmWave are stuck with last status

Motya-HA, 2024-08-11:
Hi 
I am experiencing an issue with EPL with both SEN0395 or SEN0609 mmWave radars. Sometimes they are stuck with recent status (clear or detected) and become not responsive. LED of the mmWvae sensor is turning on. 
Any change of settings (distance, sensitivity, LED on/off) brings mmWave sensor back to life.
It is not happening with LD2410 mmWave radar. 
The issue started after last update which was pretty challenging (see #128) . 

This is current firmware: 
Everything_Presence_Lite
by EverythingSmartTechnology
Firmware: 1.0.8 (ESPHome 2024.7.3)
Hardware: 1.0.8
Flashing through OTA is not resolving the issue

EverythingSmartHome, 2024-08-11:
I've seen similar issues before where users have automations that turn off all switches and lights in HA. If you check under hidden entities you'll find a switch for the mmWave sensor. Is it possible you have an automation changing this? That's why the LED goes constantly on is when that switch gets toggled to off (please note the UI may not always reflect the true position of the mmWave switch due to it being serial)

Motya-HA, 2024-08-11:
Thanks for following up
Actually, the sensors under question are not used by any automation. I checked all entities (and hidden ones) in the YAML code for scripts, automations and scene, just to be sure I am not missing anything.
I enabled "mmWave sensor" entity in one of the problematic devises.

Any other debugging can be suggested. 
I am running test of all these devices before place them in the custom ceiling case (for ceiling mount) and install in the rooms ceiling area.

[image]
[image]

Motya-HA, 2024-08-12:
It happened again. 
[image]
tree sensors mmWave stoped responding: EPL with SEN0395, EPL with SEN0609 and even EPL with LD2410 (which doesn't have mmWave LED)
[image]
Snapshot shows that mmWave entity in the sensor where it was enabled, became unavailable 
[image]

Motya-HA, 2024-08-12:
Wow...in order to make 2 (two) sensor work again, I needed to delete them from Home Assistant, delete from ESPHome and adopted them again (ESPHome) and include to Home Assistant  

I just saw that another sensor in the office, which is already in the ceiling mount was non-responsive as well.  But in this case just change one of the settings make it responsive. I did not have any of those issues before last ESPHome update and issue #128

EverythingSmartHome, 2024-08-12:
From what you describe it seems like there is something in your setup that is causing the mmWave sensors to be stopped, since you have 3 units all with different mmWave sensors exhibiting the same behaviour at the same time.

When the issue occurs, can you see if toggling the mmwave sensor switch to off, waiting 10s then back to on resolves the issue?

I know it may seem like it was the only thing that fixed it, but flashing the firmware doesn't touch any of the mmwave sensor settings and won't be the fix. The mmwave sensor has it's own firmware that is completely separate to the EP1/Lite firmware and isn't affected by any firmware done to the main ESP32.

Don't worry, I'm sure we can figure out what the cause is!

Motya-HA, 2024-08-13:
>When the issue occurs, can you see if toggling the mmwave sensor switch to off, waiting 10s then back to on resolves the issue?

> 
currently "mmwave sensor" entity is hidden. I'll enable them first. BTW, in almost 100% of cases when mmwave sensor stuck in "clear" or "detecting" state, any control brings it back to responding. But I'll follow you suggestion

EverythingSmartHome, 2024-08-13:
Yes, which sounds like the mmWave sensor toggle is getting switched off because that's the same behaviour. Then changing any setting will cause the mmWave sensor to turn on again

Motya-HA, 2024-08-14:
Observations:
1. Only SEN0609 and SEN0395 in EPL board are becoming stacked (in the current state). 
2. Non of LD2410 in EPL exhibits this issue.
3. "mmWave sensor" switch entity is in "on" state even when issue occurs. Toggling of "mmWave sensor switch" makes the mmWave sensor responsive again. Same as any of the setting entities, like sensitivity, distance, latency etc
4. One impotent observation - I caught it when I was  present in the area - ESP LED started "blinking" for maybe 15-20 sec and mmWave sensor LED turned on (and sensor became stack). Even when ESP LED stoped blinking, mmWave LED remained "on" and sensor remained stacked

EverythingSmartHome, 2024-08-14:
So it occurs directly after the esp starts blinking? If so, the esp LED will start blinking if WiFi becomes disconnected, or Home Assistant drops it's connection to the Lite.

If it's happening directly after one of those events, then it could be a software bug.

The toggle for the mmWave is not reliable because it's serial. It is possible for the toggle to show "on" the UI while the sensor is very much turned off (in the backend)

Motya-HA, 2024-08-14:
Yes, it is happening immediately after  ESP LED is blinking.
this is the log snap shots
[image]
[image]
[image]
[image]

I have other ESP devices and they do not show this type of issues

Thankls

Vladi

EverythingSmartHome, 2024-08-14:
> I have other ESP devices and they do not show this type of issues

It's not really relevant in this instance.

I think I can see what's happening. To confirm, can you please attach the debug logs for devices? You can do this by going to the device page in Home Assistant, then click the three dots in the left hand corner and download ESPHome diagnostics. Then attach those here

EverythingSmartHome, 2024-08-15:
I wonder if the issue is because you have Bluetooth proxy enabled which can cause disconnects in some environments.

What I suspect is happening is that it is working fine, it disconnects from WiFi, mmWave stops, it reconnects to WiFi but mmWave doesn't enable.

If so, the mmWave not turning back on is a bug, but also you probably don't want it disconnecting.from WiFi either.

To confirm, can you flash the non Bluetooth version of the firmware and test to see if it solves the issue? I'll take a look at why it's not re-enabling too

Motya-HA, 2024-08-15:
Ok
I flashed on sensor with no Bluetooth proxy enabled. Will see tomorrow
I am just wondering why would is disconnect from WIFI recently. I did not change any of my AP (UniFi)

Motya-HA, 2024-08-15:
Looks like no Bluetooth proxy enabled, sensor is not restarting and survived night and morning

However:
1. After power cycle (unfortunately power outage is happening in our area - Bay Area CA) mmWave sensor is off and I need to toggle any setting entity (everything presence one doesn't have that issue)
2. Why  Bluetooth proxy enabled causes disconnects?

I can leave with no Bluetooth proxy enabled (not ideal)
But I can't control our electric company, so #1 is an issue

Thanks for support

Motya-HA, 2024-08-15:
After HA restart all sensors are stack until some of the setting or mmWave sensor entity is pushed.

xpenno255, 2024-08-18:
I have the same issue as this. I'm running the non ble version of the firmware and my eplites all have sen0395 sensors. They were all working fine previously. This started happening about 4 or 5 weeks ago but I'm not sure if it was following a HA update or ESPHome.

xpenno255, 2024-08-18:
Also, to add, this doesn't happen on my EPOne units with the same sensor only the EPLites.

EverythingSmartHome, 2024-08-19:
Can you give firmware 1.0.9 a try? I added a condition to SEN0609/SEN0395 to switch the mmwave on after wifi connect which will hopefully resolve this issue

xpenno255, 2024-08-19:
> Can you give firmware 1.0.9 a try? I added a condition to SEN0609/SEN0395 to switch the mmwave on after wifi connect which will hopefully resolve this issue

Will give 1.0.9 a go shortly.

For info I just cross referenced the code vs the EPOne version and edited the restore_mode from RESTORE_DEFAULT_ON to DISABLED (as per the EPOne code) and now I don't see this issue anymore. I've not read up on what this means/does so no idea why it might have fixed it.

Previous

`switch:
  - platform: template
    name: mmWave sensor
    id: mmwave_sensor
    disabled_by_default: True
    entity_category: config
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
    turn_on_action:
      - uart.write: "sensorStart"
      - delay: 1s
    turn_off_action:
      - uart.write: "sensorStop"
      - delay: 1s`

New

`switch:
  - platform: template
    name: mmWave sensor
    id: mmwave_sensor
    disabled_by_default: True
    entity_category: config
    optimistic: true
    restore_mode: DISABLED
    turn_on_action:
      - uart.write: "sensorStart"
      - delay: 1s
    turn_off_action:
      - uart.write: "sensorStop"
      - delay: 1s`

xpenno255, 2024-08-19:
Urgh, my ESPHome addon has cached 1.0.8 and I can't make it update......

Motya-HA, 2024-08-19:
Can you give firmware 1.0.9 a try?

> 
With  ble or without ble?

Is there any idea why  with  Bluetooth proxy enabled ESP board looses connection to WiFi (or HA)?

xpenno255, 2024-08-19:
Ok, I forked and renamed the repo and 1.0.9 has fixed the issue for me. I'll just wait for EPSHome cache to expire and update the rest of them.

Motya-HA, 2024-08-19:
Checked firmware 1.0.9. Works great. Restart, power cycling - no problem.

Can I get back to ble? :-)
I have several aqura temperature.humidity ble devices

Thanks

EverythingSmartHome, 2024-08-19:
> > Can you give firmware 1.0.9 a try?
> 
> > 
> 
> With ble or without ble?
> 
> Is there any idea why with Bluetooth proxy enabled ESP board looses connection to WiFi (or HA)?

It's a very well documented issue, I'm no developer so the understanding could be wrong, but I believe that since Bluetooth uses the same radio as WiFi, it has to allow airspace for both to operate at the same time which can cause disconnects. 

Take a look at some of the warnings on the Bluetooth proxy docs for example: https://esphome.io/components/bluetooth_proxy.html#improving-reception-performance

Glad the issue is resolved, I will close for now since it seems to be working.

closed 2024-08-19 by EverythingSmartHome