EverythingSmartHome/everything-presence-lite #135: Zone occupancy doesn't work and angle is reversed?

V4ler1an, 2024-08-11:
**The good:** The main 'Occupancy' entity works and I can use all 3 of my EPLs for just room presence ok. I've had lighting and presence automations working fine for weeks.

**The bad:** I cannot get the specific Zone 1, 2 etc occupancies to register anything, ever. They've never changed. I read through some guides and have setup one of my EPLs with a simple two zone setup the same as [Smart Home Junkie's guide](https://www.smarthomejunkie.net/everything-presence-lite-blueprints-manual/) but I get nothing. The EPL tracks me around the room, I watch the X and Y coordinates change on my companion app and it appears to match the layout in SHJ's diagram below. If I walk around the left side (zone 1), the tracking coordinates are all positives as expected for zone 1.

**The strange:** There are two very weird things I've noticed. First of all, would I be wrong to assume that an X axis should go to negative on the left side? That would match every X and Y diagram I've ever seen. If so, then why when i'm being tracked and similarly in SHJ's diagram below, does the X axis go to positive on the left and negative on the right? Am I missing something? That seems counter intuitive.
 
[image]

The second thing I don't understand... in the next image below from Smart Home Junkie's guide, you can see that he is being tracked in zone 1 (both coordinates show positives). And his 'Angle' is registered as +44 degrees. When I walk around the left side of my identical setup (in zone 1), the coordinates track me the same as his (all positive coordinates), but my angle is all negatives! Until I cross the centre and walk over to the right side where the angles then go negative. Now, again, correct me if i'm wrong, but wouldn't the left side of the angles logically be negative?? Matching the way an X axis should also be negative on the left? It seems like my 'Angle' sensor is reversed to SHJ. Something is very odd here. Can anyone suggest the reasoning behind what i'm seeing?

SHJ's imge from his guide:
[image]

My image when I walk around to the left of my EPL in zone 1:
[image]

EverythingSmartHome, 2024-08-11:
Can you share the config you have for the zone first of all?

V4ler1an, 2024-08-11:
Sure my version for this EPL:
[image]

This is the current config. With the exception of the strange (reversed?) angle readings, the coordinates I see when walking around seem to be as expected:
[image]

EverythingSmartHome, 2024-08-11:
Yeah so this version is quite a few behind, please update to the latest which fixes the negative angle and reversed coordinates.

The issue with the zone is because you have the start and and end X values the wrong way around. The start value should be the lower number and the end the higher number. Switch the start and end X values around and should start to work

V4ler1an, 2024-08-11:
Ok i'll update tomorrow. One of the biggest questions I had when first playing with these zones was exactly that. Which corner do you start with. Even SHJs guide doesn't mention that. Is there an official guide that covers these details?

V4ler1an, 2024-08-11:
Also, is there a way to update the firmware over Wi-Fi / via HA? Or do I need to USB direct connect them?

V4ler1an, 2024-08-12:
Reversing the co-ordinates, (starting from the bottom right corner of each zone) has got the zone occupancy activating now. Thank you for that. I haven't updated the firmware yet but will do soon. As per my previous message, is the only option direct USB currently? (my bluetooth radios are turned off at the moment).

EverythingSmartHome, 2024-08-12:
Zones setup is definitely on the cards to add to documentation soon yes.

You can update the firmware over wifi if you have the ESPHome add-on. You can also update without the ESPHome add-on and do it directly from Home Assistant, but unfortunately that feature was added in a later release (1.0.7, you are on 1.0.3)

EverythingSmartHome, 2024-10-08:
I will close this as resolved for now, please see the new zone documentation and visual configurator tool for creating zones:

[image]

https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/creating-zones.html

closed 2024-10-08 by EverythingSmartHome

V4ler1an, 2024-10-08:
Cool! Thank you.