EverythingSmartHome/everything-presence-lite #136: Phantom target registers only once after leaving a room

V4ler1an, 2024-08-13:
I think I may have a fault with one of my EPLs. It's setup in a room with 2 zones. The zones activate and deactivate normally as I move around the room, turning their respective lights on and off as expected. Everything seems ok until I leave the room however. For some strange reason, a few moments after leaving, one of the zones registers activity briefly (always the same zone) and turns the light on again, then straight off. I've been sat with my laptop in the next room and watched as the zone activity spike pops up in Home Assistant's logbook for the device. It briefly registers a target. 

What I can't understand is that once it's had its little delayed phantom activity spike, it never does nothing further. On occasion it's been so delayed (30 seconds or so), that i've heard it switch on while I was going up my stairs to bed. The stairs are two rooms away. Not a chance of something triggering it. I've been really careful to ensure it's not me, and the fact that it doesn't ever repeat it indicates there's nothing in the room causing it.

The EPL is on the latest firmware. Just updated it today to see if that resolves the issue. No change. Any ideas what I could try? Have you ever come across something like this?

Draghmar, 2024-08-13:
Do you have records of X/Ys for targets and other stuff? I've seen similar things going on but I wasn't able to pin point the culprit...outside of the faulty/unreliable device that is. I'm testing my unit to get better understanding of it's limitations so I've created history filter [like so](https://github.com/EverythingSmartHome/everything-presence-lite/issues/126). I have few PIRs there and FP1 to compare. If you didn't install it, try the `plotly-graph`. Maybe it will show you something interesting.

V4ler1an, 2024-08-13:
OK. Here it is... i've been monitoring the EPL's sensors very closely on the companion app as I walk out of the room. Here's what happens...

- When I walk out I leave from zone 2. Zone 1 has no activity and its light is off (this is correct, all good). 
- I watch the EPL actively tracking my target as I approach the door way, just a few feet away. The target 1 numbers are all changing, all looks normal. 
- As I get close to the door way and leave the zone, all the Target 1 numbers freeze. They don't zero out, they just freeze. (I don't know if this is normal or they should all reset to zeros?)
- At around the same time or soon after the numbers freeze as I leave Zone 2, its occupancy clears (all good). Now the occupancy for both zones are clear with zero target counts.
- I go and sit down in my chair in the next room. All the previous tracking numbers are still frozen. Zone 1 and 2 are still clear. 
- Around 15 second after I left the room (when the tracking numbers froze) the following happens all at the same instant:
1. The target trackng numbers all go to zero.
2. The 'Zone 1 Occupancy' registers as 'Detected' out of nowhere. Despite the 'Zone 1 Target Count' underneath it remaining on '0'.
3. The light for Zone 1 turns back on (My automation is set to turn the light on based upon Zone 1 becoming occupied so you would expect this).

So what do you make of that? This behaviour is repeatable, every time. The EPL has all its occupancy delays set to 15 seconds. So it appears as though once the occupancy delay expires for the room (or zone 1 perhap? (They both expire at the same time obviously), it causes Zone 2 to 'blip' its occupancy out of nowhere, hence the Zone 1 light coming back on. The automation and my Zone 1 light are just following orders. Why is the EPL throwing out this spurious short detection without activity when the occupancies expire? Should the tracking numbers all freeze, even though all occupancy has cleared? It the EPL hanging? Then causing this occupancy spike glitch when it recovers?

V4ler1an, 2024-08-14:
> Do you have records of X/Ys for targets and other stuff? I've seen similar things going on but I wasn't able to pin point the culprit...outside of the faulty/unreliable device that is. I'm testing [...]

Your chart looks like the sensors are hanging for periods. This also seems similar to my issue? Only when mine recovers it triggers an unregistered occupancy in one of my zones.

Taomyn, 2024-08-14:
I think I'm seeing this same behaviour today, I had left home and from the office noticed it said occupancy was detected. Checked my cameras and the room was empty. Checked and did see that it cleared when I left but then around an hour later changed to detected when nothing was present. Hit reboot in HA and after a short period of unavailable it came back on-line but was still showing detected. - all the target 1 values looked to be the same. So I remotely cut power to the EPL, waited a few minutes, re-enabled power and this time it went back to clear. Haven't seen it change again yet, and won't be home for a few hours to check it still actually works.

V4ler1an, 2024-08-14:
> I think I'm seeing this same behaviour today, I had left home and from the office noticed it said occupancy was detected. Checked my cameras and the room was empty. Checked and did see that it [...]

Did the occupancy register along with a target, or did the target count remain on 0? Mine only does it fairly soon after leaving the room where yours sounds a lot later. Also, did you notice any frozen tracking sensors? Mine froze and when they cleared, that's the exact moment it briefly registered an occupancy. I would be interested to know if you see ny of these characteristics when the unexpected occupancy is registered.

EverythingSmartHome, 2024-08-14:
Can you let me know which version of the software you are on please? It is likely to be a bug in the logic rather than anything hardware related

V4ler1an, 2024-08-14:
Some further info on the behviour i'm seeing:
- The action of entering and leaving only zone 2 causes zone 1 to register occupancy without any target count or activity. I can exit the room via two doors, both avoiding zone 1 completely. It triggers zone 1 whenever I step away from zone 2. 
- The second trigger for zone 1 to register a  phantom occupancy is the 'Zone Occupancy Off' delay. It's currently set to 15 seconds. When that expires for zone 2, it also trigges zone 1 to register phantom occupancy and turn the light on.
- If I go into then out of only zone 1. It doen't do the delayed switch back on upon exit or 15 sec delay.

Kummallinen, 2024-08-14:
I'm seeing the same issue with Zone 1 always triggering whenever a target stops being detected. Setting the "Zone 1 Start Y" to 10mm stops it for me

V4ler1an, 2024-08-14:
> I'm seeing the same issue with Zone 1 always triggering whenever a target stops being detected. Setting the "Zone 1 Start Y" to 10mm stops it for me

Interesting! Thank you I will give that a try tomorrow. Just reverted everything to a single zone for this evening.

Kummallinen, 2024-08-14:
> > I'm seeing the same issue with Zone 1 always triggering whenever a target stops being detected. Setting the "Zone 1 Start Y" to 10mm stops it for me
> 
> Interesting! Thank you I will give that a try tomorrow. Just reverted everything to a single zone for this evening.

Hopefully it works. I was working on the assumption that the reset to "0,0" for the target is triggering that zone momentarily. So if there is no zone covering "0,0" nothing can trigger

V4ler1an, 2024-08-15:
> > > I'm seeing the same issue with Zone 1 always triggering whenever a target stops being detected. Setting the "Zone 1 Start Y" to 10mm stops it for me
> > 
> > 
> > Interesting! Thank you I will give that a try tomorrow. Just reverted everything to a single zone for this evening.
> 
> Hopefully it works. I was working on the assumption that the reset to "0,0" for the target is triggering that zone momentarily. So if there is no zone covering "0,0" nothing can trigger. 10mm is the lowest start Y it would accept other than 0

My friend you were absolutely correct. Have a cyber-beer on me! I changed the Y of zone 1 (the one nearest the EPL) to +30 and I can now walk in and out of both zones and into the next room all day without issues. Big thanks! Now i'm going to really push my like and try a 3rd zone :)

V4ler1an, 2024-08-15:
> Can you let me know which version of the software you are on please? It is likely to be a bug in the logic rather than anything hardware related

I'm not sure of the exact reasoning behind the cause, calibration perhaps? But https://github.com/Kummallinen was absolutely correct. Y to +30 has worked around the problem for now.

V4ler1an, 2024-08-15:
3rd zone working perfectly. Zones are stacked in line, one behind the other.

EverythingSmartHome, 2024-10-08:
Please try and update to the latest 1.1.1 version and let me know if this is still an issue

V4ler1an, 2024-10-08:
Already updated my EPLs to 1.0.11. Have just reverted my kitchen/diner zone 1 to Y start of 0.0 instead of +20. I'll let you know results in a few days.

EverythingSmartHome, 2024-10-08:
1.1.1 is the latest which has the fixes related to this. 1.0.11 does not

Taomyn, 2024-10-08:
I updated mine even though I think upgrading the BT firmware may have helped anyway, and both my units are working.

[image]

However, the firmware updates are temporarily breaking my units because it disables API encryption, it happened to the one that got released last week - when the firmware is done they connect to WiFi just fine but HA/ESPHome can't see them. When I try to view the logs it shows:

```
INFO ESPHome 2024.9.2
INFO Reading configuration /config/esphome/epl-22dd80.yaml...
INFO Starting log output from 192.168.1.33 using esphome API
INFO Successfully connected to epl-22dd80 @ 192.168.1.33 in 0.049s
WARNING epl-22dd80 @ 192.168.1.33: Connection error occurred: epl-22dd80 @ 192.168.1.33: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
WARNING Can't connect to ESPHome API for epl-22dd80 @ 192.168.1.33: epl-22dd80 @ 192.168.1.33: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2). (HandshakeAPIError)
INFO Trying to connect to epl-22dd80 @ 192.168.1.33 in the background
INFO Successfully connected to epl-22dd80 @ 192.168.1.33 in 0.037s
WARNING epl-22dd80 @ 192.168.1.33: Connection error occurred: epl-22dd80 @ 192.168.1.33: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
INFO Successfully connected to epl-22dd80 @ 192.168.1.33 in 0.056s
WARNING epl-22dd80 @ 192.168.1.33: Connection error occurred: epl-22dd80 @ 192.168.1.33: The connection dropped immediately after encrypted hello; Try enabling encryption on the device or turning off encryption on the client (ESPHome Logs 2024.9.2).
```

To fix this I have to perform the following through the main ESPHome menu in HA:

- Edit the yaml and remark the API lines
- Press save (not install) then close - they come back online
- Edit the yaml again and unremark the API lines
- Press install (not save) - it re-installs the code and they are back to normal

No idea what is going on, but this has happened twice to me now.

[image]

EverythingSmartHome, 2024-10-08:
@Taomyn Yes that will happen, that entity you showed in your screenshot "Everything Presence Lite Firmware" that shows the update from 1.0.11 to 1.1.0 is disabled by default for this reason. I'm guessing you enabled it again?

It's disabled because if you have it added to the ESPHome addon then that behaviour will occur because the ESPHome add-on automatically adds an encryption key where as updating from GitHub does not

Taomyn, 2024-10-08:
@EverythingSmartHome so how do I upgrade the firmware with ESPHome? I've only ever seen HA tell me there are updates since the new ESPHome add-on was added, but I don't recall ever enabling the entity though that's always possible at some point.

V4ler1an, 2024-10-08:
> 1.1.1 is the latest which has the fixes related to this. 1.0.11 does not

Updated!

V4ler1an, 2024-10-10:
After several days of usage I've experienced no further issues running my kitchen EPL with three stacked zones and zone 1 beginning at Y 0.0. If anything further comes up i'll let you know. Thanks for all your help!