EverythingSmartHome/everything-presence-lite #137: LX value wildly different after ESPHome update 2024.7.3

markwebsterUK, 2024-08-15:
Since the last firmware update the lx value in HomeAssistant is wildly different than before, messing up automations.

In the logs I see the following when the lx value is updated:
```
[08:53:32][D][bh1750.sensor:159]: 'Illuminance': Got illuminance=19.6lx
[08:53:32][D][sensor:094]: 'Illuminance': Sending state 275.83578 lx with 1 decimals of accuracy
```
Why the discrepancy? The offset is set to 0.

EverythingSmartHome, 2024-08-16:
Did you read the release notes? It's specifically and clearly mentioned as a breaking change in 1.0.6:

https://github.com/EverythingSmartHome/everything-presence-lite/releases/tag/1.0.6

The reason for the "discrepancy" in the logs is because calibration data was added to improve light level readings with cover on Vs cover off

closed 2024-08-16 by EverythingSmartHome

markwebsterUK, 2024-08-16:
I made no mention of updating to 1.0.6, I updated the esphome firmware itself and I'm guessing it grabbed the new version of the lite code when it did that. So no, I didn't see the release notes because I didn't know I needed to look. But thanks for the "friendly" pointer and sorry for apparently causing so much anger... The values on mine also seem to be all over the place, sometimes it reads ~80, sometimes ~200 when the light level hasn't changed (when it's dark outside and the lights are on). I'll experiment to see what's going on, but don't worry, I won't trouble you with any other "discrepancies". Probably won't give you any more of my money, either.

EverythingSmartHome, 2024-08-16:
There is certainly no anger, far from it - please don't read any tone from text on the internet, there was intended to be any and sorry if you took it that way.

Yes that's correct, when you update firmware it will also download the latest. Those changes were made quite a while ago though and had quite a few ESPHome releases since then so I assumed you would have applied those, though maybe you only just updated the ESPHome addon in which case that would explain it.

The calibration values were just initial values and there is definitely some more work to do on those calibration values to make sure the behaviour your describing isn't occuring.

Again, there was no anger or frustration from me and I'm certainly happy for you to add issues!

markwebsterUK, 2024-08-16:
Yeah I've been a bit lax in updating stuff recently, been busy and was of the mindset "it works, so don't touch right now". All good man, I probably assumed attitude where there was none (spending too much time on reddit :laughing: ) I'll have a play with a calibration values locally and see if I can find something that works for me. :v:

markwebsterUK, 2024-08-16:
Hey, I'm pretty sure I've found the issue. For a filter of type `calibrate_linear` with `method: exact` the values on the left of the datapoint are replaced with the values on the right. So if you're calibrating for the light lost by passing through the cover, the values on the right side of the datapoint should increase in a fairly linear fashion, however, the values currently are a bit wild :)

```
    filters:
      - calibrate_linear:
          method: exact
          datapoints:
            - 0.6 -> 1.7
            - 9 -> 25
            - 10 -> 29
            - 14 -> 367
            - 19 -> 26
            - 21 -> 885
            - 22 -> 88
            - 29 -> 52
            - 85 -> 166 
            - 283 -> 394
            - 807 -> 993
            - 917 -> 1124
            - 1132 -> 1170
```

The replacement values for 14, 19, 21, and 29 stand out as possibly being incorrect. It might be worth removing those values and changing the `method` to `least_squares`. I _think_ that will result in a more predictable output. (I'll try it out later when I'm at my desk)

EverythingSmartHome, 2024-08-16:
You are correct that those values may seem to be off in comparison to the others, but it's actually because I tested units in different scenarios - light source from the top, side, 45 degree, underneath etc then also repeated same tests with multiple different strengths of light source

I've played around with the exact method vs linear (because we use calibrated values on the EP1) and I can't remember the reason I found exact to work better there. Definitely could be wrong though

EverythingSmartHome, 2024-08-16:
Ideally it would be great to get soke calibration values from other environments instead of just my own, as there is only so many ways I can test it!

markwebsterUK, 2024-08-17:
I can see what you're getting at, but as the sensor doesn't know if the light it's receiving is direct, angled, or ambient, the extra values will just end up confusing things. The filters will apply no matter what. I think the best thing to do is either calibrate it with indirect, ambient light to get a nice even curve that compensates for the light lost by having the cover on. Or not calibrate at all and have people use the offset if they want to. Most of these things will go into position and stay there, automations can be set up by users depending on their circumstances.Trying to accommodate every use case is a path to losing your sanity 😆

markwebsterUK, 2024-08-19:
I can't find values that work. I don't know if I'm doing it wrong or what, but I just end up with lx values that are way higher than they should be at certain levels which is making my office lights flash on and off. I've reverted back to 1.0.5 and it's working fine again.