EverythingSmartHome/everything-presence-lite #138: Various occupancy detection issues

clintebbesen, 2024-08-16:
Hi,

I've been having a lot of issues with detection. I am having constant issues across 4 sensors where occupancy is both on when it shouldn't be, and off when it should be on. When occupancy is on, I can usually see a conflict of entities and see the issue.

1. As per the image below, Occupancy is showing as Detected, however there is no one in the room, no motion. Zone 2 shows occupancy on, count of 1, however targets 1, 2, 3 active are all off.
2. Sometimes occupancy will be detected falsely, and when doing so the Target Y is 0. I set the zone 1 start Y to 10,  which has fixed this issue, but thought id report it.
3. if I leave the detection zone without the sensor seeing me move out of range, this fairly reliably causes the sensor to say presence is detected for anywhere from about 30 seconds to a few minutes or sometimes even longer. In some cases I could fix this by decreasing the max distance, but not all. my bathroom and kitchen are two examples. In the bathroom, the sensor is in the corner of the bottom left of the room, the doorway is in the bottom right. see the second image for a rough sketch of the room, sensor location, door, and my path walking. When I walk from the lounge into the bathroom the sensor triggers outside the bathroom in the hallway when im in line of sight to the sensor, when I walk back to the lounge, at this point im still within the 6m range of the sensor (sensor was set to about 3.5m), but as I walk out to the lounge I disappear behind the wall. same issue with the sensor in the kitchen when I walk to the right of the sensor past its range of motion, not past its max distance. The issue seems to be, if you leave line of sight of the sensor while your distance to the sensor is less than the configured max distance, then the sensor stays occupied for some time, anywhere from 30s to minutes or more in rare cases. To fix the bathroom sensor, I changed the max distance down to 3m, the width of the room so that as I walk out the door into the hallway, before I turn the corner and get out of line of sight of the sensor, I first cross the max distance, turning off occupancy, then I turn the corner to the lounge leaving line of sight and everything works. This however isnt a solution as it doesnt fix the kitchen sensor when I leave the room to the side past the max angle the sensor can see to the side, and with the bathroom sensor, I do want the sensor to see into the hallway so when im in the hall looking in the hallway cupboard, the bathroom light will be on providing light to the hallway.
4. Occupancy on, zone 1 occupancy on, but zone 1 targets 0, when there is no one in the room and occupancy should be off. the targets of 0 is correct, but occupancy is on for some reason, walking in and out of the room generally fixes it.
5. I cant remember the other cases, but there has been many cases like the last one above where its not working as expected, I look at the sensor page and see conflicts between the entities, I was thinking for a short time I could create new occupancy template sensors in home assistant which check for occupancy of each of the zones, while the target count of the zones is greater than 0, and either target 1, 2, or 3 active is on. but thought id report the issues here instead and hope someone else has had the issues and can help fix it rather than me making tacky fixes in HA for it.

Thank you for your help! I love the idea, but so far have been spending way too many hours trying to get everything working (probably 30+ hours tweaking them and working out work arounds to get them being more reliable), but tonight its run into significant issues and the house is in darkness with only 1 of the 4 working at all. im working through whats going on and hoping to get them fixed soon. 

image for point 1:
[image]

image for point 3. Sensor is in corner of bathroom at bottom left, doorway is bottom right of room. 
[image]

clintebbesen, 2024-08-17:
I found an example of point 2 I mentioned in my original post. Currently there is no one in my bathroom, however zone 2 occupancy is showing as Detected. Overall occupancy is clear, which contradict each other. Target 1 Active is on, and the target 1 distance is 0 mm. I realised now I haven't set this sensor to have a Y start of 10 like my other sensors which mitigates this issue.

However it is a bit odd that Target 1 is on, but has 0's for all values, Target 2 is off, but does have values for angle, distance, and resolution.

Also another issue I spotted is that Zone 2 occupancy is Detected, while target count is 0, and Zone 3 target count is 1, but occupancy is clear.

This is an example of the odd things ive been seeing which makes me think of creating helpers that check everything and only show presence if target count is greater than 0, while zone occupancy is detected and overall occupancy is detected and the target that is active must have values for angle, distance and resolution.

[image]
[image]

clintebbesen, 2024-08-17:
After writing this and getting the screenshots I pressed the ESP reboot button on the sensor page in home assistant to reboot it fixing the issue, and that gave me the idea to maybe add an automation so that is values are conflicting, then reboot the sensor, that could be another temp fix until the root cause is found

clintebbesen, 2024-08-17:
20 mins later I noticed the Bathroom lights on again. This time when checking the sensor, occupancy is off so I am unsure why the lights are on, but I did notice some issues in the entities.

occupancy is clear overall and in each of the 3 zones, however target 1 active is on, and zone 3 target count is 1. 

None of the values for target 1 are changing, including the speed which is at 0.08m/s. I clicked on speed, and it shows it was last updated 9 minutes ago, and I can see in the history, there was a time this morning where the speed was the same value for 1 hour, and earlier this evening the same value for 2 hours. looking at the history the day before there was also two times where the speed has a stuck value for an hour each time. Id imagine if the distance, x or y values aren't changing the speed cant be anything other than 0 m/s 

[image]
[image]

clintebbesen, 2024-08-17:
This is from my kitchen sensor, no one is in the kitchen, occupancy is off as expected for overall and each zone, however target 2 active is on with values populated, including speed which has been hanging with the same value for the last 14 minutes, and this afternoon had a value for 5 hours straight

[image]
[image]

bazcurtis, 2024-08-23:
I am finding the EPL very sensitive. Walking up the stairs which are next to the room with the EPL will trigger presence in the room. I have had to revert back to the Sonoff as that is more stable. I have to be honest, I am having trouble mapping the room. The x and y values jump about a bit when I try and map them. I am not sure which value to use.

clintebbesen, 2024-08-27:
Hi,

I am still having continual issues to the point I only have 2 of the 7 presence sensors with automations enabled, the other 5 are sitting idle doing nothing. the 2 that are on work about 60-80% of the time, enough that I have kept them on.

Below is another issue, All occupancies are off, target 1, 2, 3 Active are off, but its clearly tracking me as I step left and right the target 1 angle changes following me, and the distance, x, y, etc all change as usual. When I stepped into a different zone in that room, then the occupancy came on, but soon after going back to that first zone the occupancy went back to off, while still tracking me. I turned the sensor off and on at the wall, but it still had an issue. 

[image]

EverythingSmartHome, 2024-08-27:
Thanks for all the details, I will take a close look soon.

As a first step, can you try the non Bluetooth proxy version of the firmware? I wonder if some of the issues your seeing correspond with a WiFi drop which can be cause by Bluetooth proxy. Would be good to rule that out

EverythingSmartHome, 2024-08-27:
@clintebbesen I also wonder if on top of changing to the non bluetooth version of the firmware, if you can add a 10mm gap in between the zone definitions? So for example, you have zone 2 starting at -4000 to -600 on X and 1900 to 3600 on Y. Then zone 3 starts at -600 to 4000 on X and 1900 to 3600 on Y. Could you try change the X on zone 3 to -590 to 4000 therefor creating a 10mm gap between zones?

Do the same with any zones that technically "overlap" - perhaps there is a slight edgecase there.

bazcurtis, 2024-08-27:
> > I am finding the EPL very sensitive. Walking up the stairs which are next to the room with the EPL will trigger presence in the room. I have had to revert back to the Sonoff as that is more stable. I have to be honest, I am having trouble mapping the room. The x and y values jump about a bit when I try and map them. I am not sure which value to use.
> > Does the resolution change the sensors sensitivity?
> 
> This is expected since sensors can see through walls. In this instance, you would set the max distance of the device to limit it to the size of the room so that it does not see the stairs.
> 
> X and Y values will natural change, roughly +/- 30cm. That's for two main reasons:
> 
> 1. It's impossible to stand perfectly still
> 2. Mostly because the sensor does it's best to latch on to the middle of a targets movement which can change continuously even when trying to stand very still. For example, if you stand still breathing, it may see your chest moving with a breath, but then after it sees your hand move a little bit or a head tilt. There is distance between all of these moving parts and the sensor is doing it's best to centre the source of movement. That's an extremely rudimentary example but hope it makes sense.
> 
> To use the coordinates, they don't need to be exact to the mm - remember that if you see for example a change of 200mm, that may look huge but in reality its only 20cm which when talking about the size of a room is really very small. Just use an approximation and then dial it in from there

Thank you, that is very helpful. The stairs are to the right of the EPL. If I measure from the EPL to the wall, which value would I change?

bazcurtis, 2024-09-09:
Yesterday I managed to catch the stairs being triggered. This is what I am seeing. I thought I had cracked it, but it is still triggering.

The EPL is where the floor map says 25c and the door is in the top right hand corner. This is about 250cm away. The stairs are to the right of the sensor.

The stairs are at least 80cm to the right. I assume -500mm would exclude that? Any ideas would be greatly appreciated.

[image]
[image]
[image]
[image]
[image]

EverythingSmartHome, 2024-09-09:
It might not necessarily be -500mm because that would assume that the target is picked up in an exact straight line to the stairs. The best way would be to stand on the stairs and use your phone to check the coordinates of the target it is detecting, then use those coordinates to check your zone config.

bazcurtis, 2024-09-09:
This is what I saw when I did that. The above sensors were also from the stairs.

[image]

EverythingSmartHome, 2024-10-08:
Can you please try the new 1.1.1 firmware and see if this is still an issue?

bazcurtis, 2024-10-08:
I am more than happy to do that. How do I install that?

tuomashamalainen, 2024-10-12:
> 3\. if I leave the detection zone without the sensor seeing me move out of range, this fairly reliably causes the sensor to say presence is detected for anywhere from about 30 seconds to a few minutes or sometimes even longer.

Currently seeing this in our hallway that's about 2.7 m long and 1 m wide, and the sensor is installed on one of the short walls. Kitchen connects to the opposite short wall, but the doors to bathroom, living room and bedroom are all along the long walls. I've set the maximum distance to 270 which is right at the entrance to kitchen.

Going into the kitchen, the sensor 100 % reliably sees that the target has left because you cross the max distance boundary, but going into any of the other rooms to the sides which are closer, the values simply freeze and can stay that way for a long time – I would like to say this happens about half the time. The sensor has been updated to 1.1.1.

I don't really need zones in this room, but what I'm trying right now is setting up a zone that is slightly smaller than the room itself and use that for the overall presence detection. For now this should, hypothetically, be more reliable than simply relying on the sensor figuring out when it can no longer see the target.

bazcurtis, 2024-10-12:
> The sensor has been updated to 1.1.1.

I have tried to upgrade to 1.1.1, but I am not sure how to. I tried to do it via the instructions page, but I don't get a connect button. I have installed the driver the CH340 on my MacBook Air, but still don't see it.

EverythingSmartHome, 2024-10-12:
> > The sensor has been updated to 1.1.1.
> 
> I have tried to upgrade to 1.1.1, but I am not sure how to. I tried to do it via the instructions page, but I don't get a connect button. I have installed the driver the CH340 on my MacBook Air, but still don't see it.

Are you making sure to click the Home Assistant button to select a platform and then selecting the options?

tuomashamalainen, 2024-10-13:
> Yes, the only option I see are the other boards to buy. I will try it again later.

Also, just in case you missed it, make sure to use Chrome or Edge to do this, it won't work in Safari (it does not support serial port connectivity). I updated my sensors on a MacBook Pro via Edge and did not have to install any drivers to do it.

bazcurtis, 2024-10-13:
> Also, just in case you missed it, make sure to use Chrome or Edge to do this, it won't work in Safari (it does not support serial port connectivity). I updated my sensors on a MacBook Pro via Edge and did not have to install any drivers to do it.

I have been using Arc to update my M5Stack Atom Echo and that didn't need any drivers either. In that case I used Arc. That is what I was using this time. I will try native Chrome too.

bazcurtis, 2024-10-13:
Unfortunately that didn't work either. This is what I see.
[image]

tuomashamalainen, 2024-10-13:
You have to go through the flow first:

1. Choose Home Assistant as the platform, as you have already done.
2. Select the mmWave sensor you have (probably LD2450).
3. Select if you have the CO2 add-on module installed or not.
4. Select if you want to use this EPL as a Bluetooth proxy or not.

Then you should be greeted with the Connect button.

EverythingSmartHome, 2024-10-13:
Yes as above you have to click to select the options..

Just as an FYI, drivers will not prevent the connect button from showing up, the connect button will always display even if nothing is plugged in, assuming you use a compatible browser (and you actually select the options)

htims1989, 2024-10-16:
Found this after coming looking for help, I was having issues with false detections happening outside of the zone I've set. It's only been 30 mins or so since I flashed 1.1.1 but so far I haven't had a single false detection.

bazcurtis, 2024-10-16:
I have just check mine and it has firmware 1.1.1. I thought I had 1.0.8. Did this come via my ESPHome 2024.10.0 update I did this morning? Regarding the above, thanks for the information, that fixed my connection issue. I assume most use the none Bluetooth option? I will see if 1.1.1 fixes my zone issue.

bazcurtis, 2024-10-17:
I just checked my new EPL. It came with 1.0.6, after installing, it did an update on its own and installed ESPHome 2024.10.0 and firmware 1.1.1. I think I am still seeing the issue above, but I need to take a closer look.

bazcurtis, 2024-10-19:
Now I have found the [Zone Creation Tool ](https://everythingsmarthome.github.io/everything-presence-lite/Home%20Assistant/creating-zones.html)and that shows what the EPL sees, it will be easier to work out what is going on.

bazcurtis, 2024-10-21:
I swapped from the general occupancy entity to zone 1 occupancy to decide if anyone was in the room and that fixed. If I used the general occupancy then the light still came on with 1.1.1

Linux685, 2025-01-20:
I have similar problems, I wonder, can two zone overlap each other? I would need it to works that way, but my problems started when I overlapped two zones and now I'm wondering if it's the cause of it...